openthread/openthread #5928: expects RCP mode test fail intermittently

simonlingoogle, 2020-12-09:
**Describe the bug** 

expects RCP mode test fail intermittently.

It happens quite often recently.

**Console/log output**

[logs_44259.zip](https://github.com/openthread/openthread/files/5663306/logs_44259.zip)

Moandor-y, 2020-12-09:
Looks like the node crashed. Do we have core dumps for this?

Irving-cl, 2020-12-09:
@Moandor-y 
I haven't added dumps for that yet. We can add dump uploading as: https://github.com/openthread/openthread/pull/5861.

Irving-cl, 2020-12-09:
I'll add dumps uploading for expect tests.

Moandor-y, 2020-12-14:
Happened again after core dump uploading added: <https://github.com/openthread/openthread/pull/5839/checks?check_run_id=1547881295>, but there are no core dumps in the uploaded artifacts. Maybe this is not a crash?

jwhui, 2020-12-14:
Here is one recent failure:
- [logs_45459.zip](https://github.com/openthread/openthread/files/5690858/logs_45459.zip)
- [core-expect-rcp.zip](https://github.com/openthread/openthread/files/5690860/core-expect-rcp.zip)
- [core-expect-tun.zip](https://github.com/openthread/openthread/files/5690861/core-expect-tun.zip)

jwhui, 2020-12-14:
Here is another recent failure:
- [logs_45435.zip](https://github.com/openthread/openthread/files/5690867/logs_45435.zip)
- [core-expect-rcp.zip](https://github.com/openthread/openthread/files/5690868/core-expect-rcp.zip)
- [core-expect-tun.zip](https://github.com/openthread/openthread/files/5690869/core-expect-tun.zip)

closed 2020-12-17 by jwhui