openthread/openthread #5944: [github-actions] thread-1-2-backbone fails intermittently

jwhui, 2020-12-11:
**Describe the bug**

thread-1-2-backbone fails intermittently

**Console/log output**

[logs_44970.zip](https://github.com/openthread/openthread/files/5681711/logs_44970.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/5681714/thread-1-2-backbone-results.zip)

jwhui, 2020-12-23:
Another instance:
- [logs_47181.zip](https://github.com/openthread/openthread/files/5737487/logs_47181.zip)
- [thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/5737488/thread-1-2-backbone-results.zip)

jwhui, 2020-12-23:
Another instance:
- [logs_47189.zip](https://github.com/openthread/openthread/files/5737492/logs_47189.zip)
- [thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/5737493/thread-1-2-backbone-results.zip)

simonlingoogle, 2020-12-24:
This is a random fail of 
 [test_mlr_multicast_routing_across_thread_pans.py](https://github.com/openthread/openthread/blob/e0bff3641fa325e021d91412203ef7be70aa0e92/tests/scripts/thread-cert/backbone/test_mlr_multicast_routing_across_thread_pans.py)

The PCAPs show that some ping request or response packets are missing from PCAP, but the packets seem to be properly sent and received by nodes. 

For example, this PCAP shows that PBBR1 forwards the `Ping Reply` sent by `ROUTER1`, but we can't find the intial `Ping Reply` packet sent by `ROUTER1` with hoplimit of 64 from the PCAP. 
[image]

This may imply that the packet is indeed sent and processed but the Real Time Simulation sniffer failed to capture this packet. 

The only cause I am aware that could cause such issue is the UDP receive buffer overflow.

I think we can increase the UDP write buffer for simulation sniffers and see if this issue happen again. 

BTW, the UDP write buffer has been increased to 25MB in [OTNS](https://github.com/openthread/ot-ns) because the default buffer size is far from enough.

closed 2021-01-05 by jwhui