openthread/openthread #5950: Support for LAUNCHXL-CC26X2R1

Goran58, 2020-12-13:
**Describe the bug**
Unable to run the CC2652 output file on LAUNCHXL-CC26X2R1
This issue was reported first time in Aug 2018 but it seams still be a problem in latest version.

**To Reproduce**
build with;
make -f examples/Makefile-cc2652
Use TI UniFlash tool and download the output file ot-cli-ftd to the launchpad (Binary unchecked)

The download is successfull but no response on the cli connection (minicom to device /dev/ttyACM0)

srickardti, 2020-12-15:
Hello @Goran58 ,

I am unsure what the issue is here. I am able to build and flash with Ubuntu 20.04 And UniFlash 6.1.0.2829. I was able to setup a basic two node network between two `CC26X2R1_LAUNCHXL` LaunchPads connected to my laptop.

One gotcha I've seen some people get caught on is the UART CLI does not prompt the user when the terminal is connected. This is a basic interface and doesn't handle special terminal control characters.

Goran58, 2020-12-15:
Hello @srickardti 
Thank you for your reply. 
You are probably correct that my issue only is that I can't see any UART response when I run the ot-cli-ftd image. 
But do you mean I can't use minicom to enter the commands to be able to configure the node?

srickardti, 2020-12-15:
You can use minicom, I did for this verification. `minicom -b 115200 -D /dev/ttyACM0` and `minicom -b 115200 -D /dev/ttyACM2` on another terminal. You just won't see the prompt (`>`) right away, start typing a command like `status` or hit `<Enter>`.

Keep in mind that each LaunchPad will show up as two `tty` devices, one is the "User UART" terminal, the other is a data pipe from the XDS110 debugger.

Goran58, 2020-12-15:
No response whatsoever. Have tried <enter>, help and state on both ACM0 and ACM1 (device names verified). Have also tried to connect with putty on Win10. 

Could it be any settings on the launchpad I have missed? I think all switches are in default positions.

Goran58, 2020-12-15:
[image]

Goran58, 2020-12-15:
Only the GREEN LED above the TM4C129 chip is turned ON.

srickardti, 2020-12-15:
The XDS110 green LED will light up with power, and the red LED will light up when the JTAG interface is being used.

The position of the jumpers looks correct. However, based on the passive component layout for the EnergyTrace part of the XDS110, this is a Rev C. About two years ago, this repository was updated to support Rev E. You can check compatibility [here](https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/767777). Sorry for the inconvenience.

If you were to attach to the target with CCS or OpenOCD, it is likely the device would be trapped in [this trap](https://github.com/openthread/openthread/blob/b9d20b590e51b410c6f6faa0b3c886aa303cf248/third_party/ti/devices/cc13x2_cc26x2/driverlib/chipinfo.h#L632). Which ensures the RF and hardware patches aren't applied to older silicon, which may damage them.

Goran58, 2020-12-15:
Yes, this is an old HW (Labeled with Rev 1.0.1, SDK.1.60RC1) so it is probably useless. 

Which boards do you suggest today, for some basic experimenting in Thread? Don't need to be TI.

srickardti, 2020-12-15:
Any of the supported boards in this repository will work fine. I'm a bit biased, but all the boards [here](https://www.ti.com/wireless-connectivity/thread/design-development.html) will work with OpenThread. We ensure that they run the examples from our SDK and this repository regularly.

The CC26X2 and CC13X2 are both in production now, so only Rev E is available.

Goran58, 2020-12-15:
@srickardti
Thank you very much for your help!

vgolovanov, 2020-12-16:
@Goran58 

You can download older SIMPLELINK-CC13X2-26X2-SDK from TI's website and compile CLI or NCP example. I little bit outdated, but works just fine.

Goran58, 2020-12-16:
@vgolovanov 
Thank you for the tip. Actually, I have since the beginning, two years ago, used TI's older Thread examples on the boards. But now, my intention was to try latest version of openthread.

closed 2020-12-28 by jwhui