openthread/openthread #5963: [NCP] Power state modes for nrf52840

vgolovanov, 2020-12-16:
**Is your feature request related to a problem? Please describe.**After trying to put nrf52840 based NCP module into LowPower or DeepSleep mode, in response getting "Unimplemented" error. 

**Describe the solution you'd like** Implemented lowpower and deepsleep modes for NCP because i it is important for Sleepy devices. 

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.

**Additional context** Add any other context or screenshots about the feature request here.

Irving-cl, 2020-12-16:
@vgolovanov 

Thanks for raising the issue. Could you provide the steps how you get the error? BTW, the power state is not related to the radio state as mentioned in [doc](https://github.com/openthread/openthread/blob/master/include/openthread/platform/misc.h#L132). (Just a reminder, since you mentioned Sleepy device).

Irving-cl, 2020-12-16:
@vgolovanov 

Upd:
I think you found the `spinel_power_state`, which has been deprecated:
[image]

vgolovanov, 2020-12-16:
@Irving-cl 

Thanks for you reply. 

Sleepy device i meant for battery powered devices and for reducing power consumption of NCP device    I get this information from Spinel specs. https://tools.ietf.org/html/draft-rquattle-spinel-unified-00

5.5.8.  PROP 7: PROP_POWER_STATE

Is there any other way to power off or put into low power mode of coprocessor device by NCP?

Thank you.

vgolovanov, 2020-12-16:
@Irving-cl  I presume for updated version of NCP should be used SPINEL_PROP_MCU_POWER_STATE property. But with that property getting error - "Prop_not_found"

abtink, 2020-12-16:
Thanks @vgolovanov for raising the issue. Let me try add some (hopefully useful) info about this:

- Yes, the `SPINEL_PROP_MCU_POWER_STATE` should be used (`PROP_POWER_STATE` is deprecated).
-  Note that the `SPINEL_PROP_MCU_POWER_STATE` is controlling the NCP's MCU power state *and not the radio component* (radio state is controlled through `otPlatRadio` APIs (for example by the OT stack itself (mac layer) or through raw-link APIs).
-  Here is the description of the `MCU_POWER_STATE`:
https://github.com/openthread/openthread/blob/4caecf44fb86336778f62b0e77b83c467e79442e/src/lib/spinel/spinel.h#L1368-L1410
- In order for this property to be supported, OpenThread config `OPENTHREAD_CONFIG_NCP_ENABLE_MCU_POWER_STATE_CONTROL` must be enabled which then requires the platform layer to provide `otPlatSetMcuPowerState()` and `otPlatGetMcuPowerState()` functions (please see `openthread/platform/misc.h`).
- This is an optional feature:
   -  A platform may choose to have its own way of MCU power control (e.g., go to lowest power level when idle) and not require host to be involved in setting this.
   - Or have the host control the NCP's MCU power state through the spinel property and/or the platform APIs.
      - Note that this is mainly intended for situations/platforms where host needs an external trigger (a "poke") to NCP to awaken it before it can communicate with the NCP (in certain low power states).
- I am not certain, but I think nordic platform may be following the first model. @LuDuda may be able to add more details on this. 

The spinel specification/documentation (encoding formats, capabilities, properties, etc.) for NCP/RCP used by OpenThread is included in `spinel.h` header file in OT repo. (The IETF draft is old (has not been supported for years with no plan to update it)).

vgolovanov, 2020-12-18:
@abtink 

Thank you for detailed explanation.

I briefly looked for the code on nrf5284x and CC2652 targets and didn't find anything related to the low power modes. 
It's looks really strange if described in specs low power modes as optional. Because that limits to use NCP devices only with power supply. 

If property SPINEL_PROP_MCU_POWER_STATE is accessible in Spinel protocol, but I'm not sure that otPlatRadio APIs you can access from host MCU. 

Any way thank you for your help. Need to think how to solve that problem.

jwhui, 2021-01-12:
Closing stale issue.

closed 2021-01-12 by jwhui