openthread/openthread #5977: Openthread example on EFR32 doesn't work with current Flex SDK version

MarcelloDavis, 2020-12-18:
Hello,

I tried to build an openthread example on my Thunderboard Sense 2 (EFR32MG12) device, according to this tutorial ([https://github.com/openthread/openthread/tree/master/examples/platforms/efr32/efr32mg12#verification]()).
Unfortunately, I don't own a Linux Computer, only a Raspberry Pi 3B (which works as my OT-Borderrouter), so I did all the programming on the RPi. Since Simplicity Studio doesn't run on the RPi, I installed it on my Windows device and used an USB stick to transfer the gecko_sdk_suite folder (_Flex (Gecko) SDK including RAIL Library_) to the RPi.
I also noticed that the Flex SDK version currently is v3.1, whereas the tutorial uses version v3.0. I thought that I might give it a try, so continued the tutorial with a Flex SDK v3.1.

So I followed the tutorial and came to the point, where you make the example file:
`make -f examples/Makefile-efr32mg12 BOARD=BRD4166A`

But when I do this, I get the following error:
`make[4]: *** no rule available to create the target "gecko_sdk_suite/v3.0/util/third_party/mbedtls/library/aes.c" from "gecko_sdk_suite/v3.0/util/third_party/mbedtls/library/libsilabs_mbedtls_a-aes.o".`
and some other following errors based on the previous.

So I figured out that Flex SDK version 3.1 is incompatible for this tutorial. As I am quite new to the topic, I couldn't find a way to install older SDK versions from Simplicity Studio (and I don't think that you can access older Versions from there), I thought I might ask you for help. Do you know a way to get older Flex SDK versions, so I can continue with the tutorial?

Thank you very much in advance!
Marcel

silabs-Simran, 2020-12-18:
@MarcelloDavis if you need to download the older version of Flex SDK 3.0.x, you can download it through Simplicity Studio 5 using the package manager.  Please check the screenshot.
[image]

[image]

I would also recommend you to use Simplicity Studio 5 to build the Open thread example for your board using GSDK.

MarcelloDavis, 2020-12-19:
@silabs-Simran thank you so much for your help! I now managed to get openthread working on my Thunderboard Sense 2 :)

After building and compiling the binaries, I flashed them to the MCU using Simplicity Commander like shown below:
[image]

In the 'browse' menu I just selected the compiled ot-cli-ftd.hex file. That's all I did, and it worked. But I am now wondering: Don't I need a bootloader? All around the internet I read about .gbl or .s37 files. What about these?

Thanks for your help!

MarcelloDavis, 2020-12-20:
So I found this explaination of the .hex-file format:
".hex files are Ascii files written using the IntexHex file format and is used for putting code into memory of microcontrollers. This file type contains metadata about where to load the file, the length of the file, the type of stuff being loaded, etc. Also when putting a .hex file onto a microcontroller you do not have to specify where to load the code, because the address is specified as part of the file."

So I guess I don't need to select a bootloader on my Thunderboard when flashing it with my ot-cli-ftd.hex-file, as the .hex-file contains the information about the loading process, right?

silabs-Simran, 2020-12-21:
@MarcelloDavis  we have KBA to explain the difference between hex and s37 format: 
https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2012/06/28/what_is_the_differen-X835

You still need the bootloader to be flashed on your device to properly boot the application. Simply erasing the MCU before flashing using Simplicity Commander does not delete the bootloader from the device as it resides at a different flash address.

Therefore, in your case you already have the bootloader installed before flashing, that is why the application is working fine.

In the image below, you can see after erasing MCU and then clicking on the "Blank Check" button. It will show that the bootloader is not blank. 

[image]

closed 2020-12-28 by jwhui