openthread/openthread #5979: [github-actions] Cert_5_1_13_RouterReset fails intermittently

jwhui, 2020-12-19:
**Describe the bug**

tests/scripts/thread-cert/Cert_5_1_13_RouterReset.py fails intermittently.

**Console/log output**

- [logs_46426.zip](https://github.com/openthread/openthread/files/5719864/logs_46426.zip)
- [packet-verification-pcaps.zip](https://github.com/openthread/openthread/files/5719865/packet-verification-pcaps.zip)

FiveDimensions, 2020-12-20:
According to the testplan, responses to multicast Link Requests MUST be delayed by a random time of up to MLE_MAX_RESPONSE_DELAY (1 second). But in this run:
[image]
The time difference is about 1.002 second, should we change the judgement to < 1.1 or INT(timedifference) < 1?

simonlingoogle, 2020-12-21:
@FiveDimensions 

I think 1.002 second should be a **PASS**. I think we should change the judgement to `< 1.1` .

PR #5987 (2020-12-21): [thread-cert] relax time difference to less than 1.1 for case 5.1.13 to fix issue #5979
Fixes #5979
closed 2020-12-21 by jwhui