openthread/openthread #5982: [github-actions] Cert_9_2_05_ActiveDataset fails intermittently

jwhui, 2020-12-20:
**Describe the bug**

tests/scripts/thread-cert/Cert_9_2_05_ActiveDataset.py fails intermittently.

**Console/log output**

- [logs_46436.zip](https://github.com/openthread/openthread/files/5720468/logs_46436.zip)
- [packet-verification-1.1-on-1.2-pcaps.zip](https://github.com/openthread/openthread/files/5720469/packet-verification-1.1-on-1.2-pcaps.zip)

FiveDimensions, 2021-01-05:
MGMT_ACTIVE_SET.rsp got LEAD_PET.rsp ack,  similar with #6005
[image]

simonlingoogle, 2021-01-06:
If it happens only with `/c/lp` and `/c/as`, maybe there is something wrong with our CoAP Token generation. :(

librasungirl, 2021-01-29:
Confirmed that it has the same cause as #6005 
[image]

Let's close this one and track in #6005

closed 2021-01-29 by librasungirl