openthread/openthread #5991: [nrf52xxx] add `rssi_get` and `lqi_get`method

Irving-cl, 2020-12-21:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
We have merged the PR for enhanced-ACK based probing. I'm now working to implement it on nRF platform and test it. As we discussed in https://github.com/openthread/openthread/pull/5780#issuecomment-732797975, we may need to add two api to get RSSI value and LQI value in `tx_ack_started`.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
Would you add these two methods in your official release? Or I can simply submit a PR to add it.

edmont, 2020-12-23:
@Irving-cl please feel free to add those new methods since we don't expect a new release on the 1.6.x Radio Driver any time soon.

Irving-cl, 2021-01-13:
Fixed by #6021

closed 2021-01-13 by Irving-cl