openthread/openthread #5992: [github-actions] Cert_9_2_09_PendingPartition fails intermittently

jwhui, 2020-12-21:
**Describe the bug**

tests/scripts/thread-cert/Cert_9_2_09_PendingPartition.py fails intermittently.

**Console/log output**

- [logs_46731.zip](https://github.com/openthread/openthread/files/5725575/logs_46731.zip)
- [packet-verification-pcaps.zip](https://github.com/openthread/openthread/files/5725578/packet-verification-pcaps.zip)

jwhui, 2020-12-28:
Another instance:
- [logs_47602.zip](https://github.com/openthread/openthread/files/5748490/logs_47602.zip)
- [packet-verification-pcaps.zip](https://github.com/openthread/openthread/files/5748491/packet-verification-pcaps.zip)

FiveDimensions, 2021-01-06:
This issue happened after opening the rf-shield for Router_1 and Router_2 for 250s, Router_2 may not merge into the previous network. Changing network id timeout setting from the Router_2 initial bring-up to the beginning of shielding them, it can make sure merge them to original network

jwhui, 2021-05-10:
[logs_75541.zip](https://github.com/openthread/openthread/files/6453883/logs_75541.zip)
[cli-time-sync-thread-cert.zip](https://github.com/openthread/openthread/files/6453884/cli-time-sync-thread-cert.zip)

jwhui, 2021-05-17:
[logs_77505.zip](https://github.com/openthread/openthread/files/6496570/logs_77505.zip)
[multiple-instance-thread-cert.zip](https://github.com/openthread/openthread/files/6496571/multiple-instance-thread-cert.zip)

jwhui, 2021-06-17:
[logs_82386.zip](https://github.com/openthread/openthread/files/6672173/logs_82386.zip)
[packet-verification-1.1-on-1.2-pcaps.zip](https://github.com/openthread/openthread/files/6672174/packet-verification-1.1-on-1.2-pcaps.zip)

jwhui, 2021-07-12:
[logs_85602.zip](https://github.com/openthread/openthread/files/6804073/logs_85602.zip)
[packet-verification-pcaps.zip](https://github.com/openthread/openthread/files/6804074/packet-verification-pcaps.zip)

PR #6833 (2021-07-21): [tests] fix intermittent failure in Cert_9_2_09_PendingPartition.py
Keeping network id timeout at 70 can result in ROUTER2 occasionally
creating its own partition. Reset back to 120 to avoid occasional test
failures.

Fixes #5992
closed 2021-07-21 by jwhui