openthread/openthread #5994: Cannot start joiner role on EFR32

MarcelloDavis, 2020-12-21:
Hello,

I just set up my first openthread network with 2 nodes: A borderrouter via a RaspberryPi 3B and a Thunderboard Sense 2 acting as a full thread device, which takes the role as a connected router. When I call `sudo ot-ctl router table` on my Borderrouter, I see 2 listed devices. Also, if I call `ping fe80:...` on the Thunderboard, I get an output saying `16 bytes from xx:xxxx:x:x:x:x:xxx:xxx: icmp_seq=3 hlim=45 time=72ms`. So I guess setup worked fine and everything's working. For the setup of the Joiner-device (Thunderboard) and connecting it to the borderrouter I followed this tutorial right here (https://github.com/openthread/openthread/tree/master/examples/platforms/efr32/efr32mg12). 

However, in the official openthread documentary they use the command `joiner start J01NME` to join an existing network. When I tried to call this command on my Thunderboard, I got the output `joiner start J01NME Error 35: InvalidCommand`. Also, when I type in "help" to see all openthread-related commands, there is no such command as `joiner`. For building the ot-binaries I set the switch to `JOINER=1`, as you can see in the screenshot below:
[image]

I have successfully connected my Thunderboard to the RPi-hosted thread network, so I guess I should be happy :D but still I wanted to learn why I don't have the command `joiner start`, although I did set the switch to the Joiner role..

_Additionally, here's a list of the `help`-output on my Thunderboard Sense 2_:
bufferinfo
ccathreshold
channel
child
childip
childmax
childtimeout
contextreusedelay
counters
dataset
delaytimermin
discover
eidcache
eui64
extaddr
extpanid
factoryreset
fem
help
ifconfig
ipaddr
ipmaddr
joinerport
keysequence
leaderdata
leaderweight
log
mac
masterkey
mode
neighbor
netdata
netstat
networkdiagnostic
networkidtimeout
networkname
panid
parent
parentpriority
partitionid
ping
pollperiod
preferrouterid
promiscuous
pskc
rcp
releaserouterid
reset
rloc16
router
routerdowngradethreshold
routereligible
routerselectionjitter
routerupgradethreshold
scan
singleton
state
thread
txpower
udp
unsecureport
version

jwhui, 2020-12-21:
@MarcelloDavis , it appears you may have had a previous build configuration where the joiner feature was not enabled. Can you try building from a fresh tree? You can clean your existing OpenThread repo using `git clean -xfd` - note that will blow away any changes you have.

MarcelloDavis, 2020-12-21:
@jwhui thanks for your fast response! I checked the cloned repo version via this command:

[image]

According to this i cloned the repo 3 days ago. Is there a way to check whether I cloned an older version?
Thank you very much in advance!

jwhui, 2020-12-21:
@MarcelloDavis , it doesn't have anything to do with what git revision you have. If you've previously run `make -f examples/Makefile-<platform>`, then you have configured the build based on the existing flags. You will need to reconfigure the build if you want to change the build configuration. An easy and sure way to do that is by cleaning your repo of the existing build files.

MarcelloDavis, 2020-12-27:
@jwhui thank you very much for your help! I now have the joiner commands available on my Thunderboard and understood, how to change the build configuration.

jwhui, 2020-12-28:
> @jwhui thank you very much for your help! I now have the joiner commands available on my Thunderboard and understood, how to change the build configuration.

@MarcelloDavis , great to hear!

closed 2020-12-28 by jwhui