openthread/openthread #5995: [efr32] Is MBEDTLS_SSL_MAX_CONTENT_LEN correct?

dismirlian, 2020-12-21:
Hi all, browsing the code I saw that `MBEDTLS_SSL_MAX_CONTENT_LEN` is defined as 768 for the efr32 platform, here:

https://github.com/openthread/openthread/blob/968440da34c1f569035e9338dd4bef7fcb970adb/examples/platforms/efr32/src/mbedtls_config.h#L13-L17

However, in the generic OpenThread mbedTLS config file, `MBEDTLS_SSL_MAX_CONTENT_LEN` is defined conditionally as 768 or 900, depending on `OPENTHREAD_CONFIG_COAP_SECURE_API_ENABLE`, here:

https://github.com/openthread/openthread/blob/c00cb8a4c4f6e2a619770ffbb4abfd0f36a7fdc2/third_party/mbedtls/mbedtls-config.h#L127-L131

Is the efr32 config file correct?

romacdon, 2020-12-21:
@dismirlian - thanks for finding this.  This does indeed look to be a bug.  Feel free to submit a PR with the fix.  Should be just including the same logic from the generic OpenThread mbedTLS config file in the efr32 platform specific version,

dismirlian, 2020-12-21:
@romacdon, just a thought... can't we rely on the generic config file for all the common logic and required modules, and only use the efr32-specific file for the efr32-specific acceleration configs? I think this is how it was done on the efr32 platform before #5666 (see `MBEDTLS_USER_CONFIG_FILE`).

romacdon, 2020-12-21:
@dismirlian I believe there were some issues that we encountered when we initially tried that but I will have to defer to @lmnotran for the details. It could be that the issues we hit are no longer relevant and we could go back to the approach that use to be in place and that you are recommending we use again now.

lmnotran, 2020-12-22:
Hi @dismirlian - To give you some context behind the switch from `efr32-mbedtls-config.h` to `mbedtls_autogen_config.h`, one of the changes we made when updating to Gecko SDK v3.0 (GSDK) was switching from the built-in mbedtls distribution to the mbedtls distribution that's included in the GSDK.

Part of this change involved switching to the auto-generated `mbedtls_autogen_config.h` files we have today. These autogen'd files are taken directly from a project generated by using Simplicity Studio 5 and are unmodified to allow for newer versions to be generated and dropped in with ease. This is especially prudent because, eventually, we'd like to get rid of these manually generated files and have the config generation done as part of the build process.

As a consequence of taking the `mbedtls_autogen_config.h` files whole and unmodified, there are a lot of redefined macros between [efr32mg12/src/crypto/mbedtls_autogen_config.h](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32/efr32mg12/crypto/mbedtls_config_autogen.h), for example, and the [built-in generic mbedtls config file](https://github.com/openthread/openthread/blob/master/third_party/mbedtls/mbedtls-config.h).

Per your suggestion, we could create a new `efr32-mbedtls-config.h` that only contains efr32-specific acceleration configs, but that may make it more labor-intensive to pull in mbedtls config updates as opposed to importing the autogen'd files whole.

For now, I can just fix examples/platforms/efr32/src/mbedtls_config.h to have the same conditionally defined `MBEDTLS_SSL_MAX_CONTENT_LEN` as third_party/mbedtls/mbedtls-config.h

Any thoughts or concerns?

dismirlian, 2020-12-26:
Hi @lmnotran, thanks for the extra context; the reason for this is clear now. My only concern is that the efr32 platform will have to manually keep up with the changes in the default/generic OT mbedTLS config file.

closed 2020-12-28 by jwhui

lmnotran, 2020-12-28:
@dismirlian That's a completely valid concern. I don't expect this to be too much of a problem since it doesn't look like the default mbedTLS config file isn't changed very frequently https://github.com/openthread/openthread/commits/master/third_party/mbedtls/mbedtls-config.h

PR #6004 (2020-12-28): [efr32] conditional definition for MBEDTLS_SSL_MAX_CONTENT_LEN
Fixes #5995