openthread/openthread #6003: WPANTUND fails to initialize nrf52840-mdk

purvisb, 2020-12-23:
This is more of a question then a bug. If this is not the correct place for this question then please point me to the correct place to post this question. Also please let me know if I've formatted this correctly.

**Describe the bug** 
I'm following [these instructions](https://wiki.makerdiary.com/nrf52840-mdk/openthread/) to setup an NCP Joiner. wpantund fails to initialize the NCP.

**To Reproduce** 
See the instructions I linked to. I believe I followed each step exactly. The point of failure for me is the command under the "Configure wpantund" section.

**Expected behavior**
Set the NCP to an offline state.

**Console/log output** 
purvisb@Ubuntu-Research:~$ bash start-wpantund.sh 
[sudo] password for purvisb: 
wpantund[12682]: Starting wpantund 0.07.01 (Dec 22 2020 17:11:04) . . .
wpantund[12682]: 	SOURCE_VERSION = 0.07.01
wpantund[12682]: 	BUILD_VERSION = 0.07.01-2-g6993264
wpantund[12682]: Reading configuration from "/etc/wpantund.conf" . . .
wpantund[12682]: Configuration file "/etc/wpantund.conf" read.
wpantund[12682]: set-config-param: "Config:NCP:SocketPath" = "/dev/ttyACM0"
wpantund[12682]: set-config-param: "Config:TUN:InterfaceName" = "utun7"
wpantund[12682]: set-config-param: "Daemon:SyslogMask" = " -info"
wpantund[12682]: Ready. Using DBUS bus ":1.0"
wpantund[12682]: Running as root without dropping privileges!
wpantund[12682]: SpinelNCPInstance-Protothreads.cpp:436: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
wpantund[12682]: Resetting and trying again... (retry 1)
wpantund[12682]: SpinelNCPInstance-Protothreads.cpp:426: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[12682]: Resetting and trying again... (retry 2)
wpantund[12682]: SpinelNCPInstance-Protothreads.cpp:426: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[12682]: Resetting and trying again... (retry 3)
wpantund[12682]: SpinelNCPInstance-Protothreads.cpp:426: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[12682]: The NCP is misbehaving: Repeatedly unable to initialize NCP. Entering fault state.
wpantund[12682]: State change: "uninitialized" -> "uninitialized:fault"
^C
Caught SIGINT!
wpantund[12682]: select() errno="Interrupted system call" (4)
wpantund[12682]: Cleaning up. (gRet = 130)
wpantund[12682]: Stopped.

**Additional context** 
start-wpantund.sh:
`#!/bin/sh`
`sudo /usr/local/sbin/wpantund -o Config:NCP:SocketPath /dev/ttyACM0  \`
`        -o Config:TUN:InterfaceName utun7 \`
`        -o Daemon:SyslogMask " -info"`

The NCP [firmware ](https://github.com/makerdiary/nrf52840-mdk/tree/master/firmware/openthread) found here worked for me. By "work" I mean it managed to set the device to an offline state. I don't know why this firmware works.

abtink, 2020-12-23:
The error from `wpantund` message indicates that it cannot establish communication with NCP (it tries  3 times,  issuing/sending a RESET command to NCP and waiting for a response, after 3 failed attempts, it enters "fault" state).

Quick thing to check is if the `/dev/ttyACM0` is correct. Is the NCP connected to this or maybe some other?
Otherwise, there may be some issue with firmware on NCP.

purvisb, 2020-12-23:
I'll resume troubleshooting in Ubuntu 18.04.5 LTS running on VirtualBox.

I ran `dmesg | grep tty` before passing the usb to the vm, and after. `ttyACM0: USB ACM` device populated when checking dmesg after passing the device through.

jwhui, 2020-12-23:
@purvisb , the instructions you linked refer to a very old version of wpantund. When building NCP firmware from the latest OpenThread master, try running the latest wpantund with:

```bash
git clone https://github.com/openthread/wpantund.git
cd wpantund
./bootstrap.sh
./configure
sudo make
sudo make install
```

purvisb, 2020-12-23:
I've ran those commands. I then appended this text to /usr/local/etc/wpantund.conf:

`Config:TUN:InterfaceName      "wpan0"`
`Config:NCP:SocketPath         "/dev/ttyACM0"`
`Config:NCP:DriverName         "spinel"`
`Config:Daemon:PrivDropToUser  "nobody"`
`NCP:CCAThreshold              "-70"`

Then I ran wpantund and it failed:

purvisb@Ubuntu-Research:~/wpantund$ sudo wpantund
wpantund[23802]: Starting wpantund 0.08.00d (Dec 23 2020 14:23:11) . . .
wpantund[23802]: 	SOURCE_VERSION = 0.07.01-349-ge62ad0a
wpantund[23802]: 	BUILD_VERSION = 0.07.01-349-ge62ad0a
wpantund[23802]: Configuration file "/usr/local/etc/wpantund.conf" read.
wpantund[23802]: Ready. Using DBUS bus ":1.110"
wpantund[23802]: SpinelNCPTask.cpp:71: Requirement Failed (Failure while trying to send command)
wpantund[23802]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
wpantund[23802]: SendCommand task encountered an error: 7 (0x00000007)
wpantund[23802]: SpinelNCPInstance-Protothreads.cpp:412: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
wpantund[23802]: Resetting and trying again... (retry 1)
wpantund[23802]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[23802]: Resetting and trying again... (retry 2)
wpantund[23802]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[23802]: Resetting and trying again... (retry 3)
wpantund[23802]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[23802]: The NCP is misbehaving: Repeatedly unable to initialize NCP. Entering fault state.
wpantund[23802]: State change: "uninitialized" -> "uninitialized:fault"

jwhui, 2020-12-24:
@purvisb , it appears that serial communication is not occurring as expected. Are you able to test with `ot-cli-ftd` firmware and see if you can interact via the CLI?

purvisb, 2020-12-28:
on the VM I burned the `ot-cli-ftd` hex file into the chip and it did not work. Here's what I did:

I cloned the openthread repo:

`cd ~`
`git clone --recursive https://github.com/openthread/openthread.git`
`cd openthread`
`./bootstrap`

Then I appended this text to ~/openthread/examples/platforms/nrf52840/platform-config.h
`#ifndef UART_HWFC`
`#define UART_HWFC NRF_UART_HWFC_DISABLED`
`#endif`

`#ifndef UART_PIN_TX`
`#define UART_PIN_TX 20`
`#endif`

`#ifndef UART_PIN_RX`
`#define UART_PIN_RX 19`
`#endif`

Then I made the firmware and flashed it:
`make -f examples/Makefile-nrf52840 clean`
`make -f examples/Makefile-nrf52840 JOINER=1`
`cd ~/openthread/output/nrf52840/bin`
`arm-none-eabi-objcopy -O ihex ot-cli-ftd ot-cli-ftd.hex`
`sudo pyocd-flashtool -t nrf52 -ce ot-cli-ftd.hex`

Run wpantund and get this error:
purvisb@Ubuntu-Research:~/openthread/output/nrf52840/bin$ sudo wpantund
wpantund[18515]: Starting wpantund 0.08.00d (Dec 23 2020 14:23:11) . . .
wpantund[18515]: 	SOURCE_VERSION = 0.07.01-349-ge62ad0a
wpantund[18515]: 	BUILD_VERSION = 0.07.01-349-ge62ad0a
wpantund[18515]: Configuration file "/usr/local/etc/wpantund.conf" read.
wpantund[18515]: Ready. Using DBUS bus ":1.86"
wpantund[18515]: SpinelNCPTask.cpp:71: Requirement Failed (Failure while trying to send command)
wpantund[18515]: SpinelNCPTaskSendCommand.cpp:351: Requirement Failed ((mRetVal) == 0)
wpantund[18515]: SendCommand task encountered an error: 7 (0x00000007)
wpantund[18515]: SpinelNCPInstance-Protothreads.cpp:412: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
wpantund[18515]: Resetting and trying again... (retry 1)
wpantund[18515]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[18515]: Resetting and trying again... (retry 2)
wpantund[18515]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[18515]: Resetting and trying again... (retry 3)
wpantund[18515]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[18515]: The NCP is misbehaving: Repeatedly unable to initialize NCP. Entering fault state.
wpantund[18515]: State change: "uninitialized" -> "uninitialized:fault"
^C
Caught SIGINT!
wpantund[18515]: select() errno="Interrupted system call" (4)
wpantund[18515]: Cleaning up. (gRet = 130)
wpantund[18515]: Stopped.

jwhui, 2020-12-29:
@purvisb , the ot-cli-ftd firmware is intended for interacting with the command line interface using a terminal - see [examples/platforms/nrf528xx/nrf52840/README.md](https://github.com/openthread/openthread/blob/master/examples/platforms/nrf528xx/nrf52840/README.md#running-the-example). I suggested this approach to test if basic serial communication is working.

purvisb, 2020-12-29:
Switching to a laptop running the same version of Ubuntu natively to avoid complications with the VM.

I ran these commands:
`stty -F /dev/ttyACM0 115200 cs8 -cstopb -parenb crtscts`
`screen /dev/ttyACM0 115200`

A blank terminal window opens that won't display any characters when I type. Although the CDC LED on the Chip does flash whenever I type a character on the laptops keyboard.

jwhui, 2020-12-30:
@purvisb , it appears that basic UART communication is not occurring as expected.

Note that the UART configurations have moved to: [examples/platforms/nrf528xx/nrf52840/transport-config.h](https://github.com/openthread/openthread/blob/master/examples/platforms/nrf528xx/nrf52840/transport-config.h). Have you tried modifying `transport-config.h` (rather than `platform-config.h`)?

purvisb, 2020-12-30:
I appended the text ([in this comment](https://github.com/openthread/openthread/issues/6003#issuecomment-751848130)) I originally put in `platform-config.h` to `transport-config.h`. Then I remade the firmware with no parameters and burned `ot-cli-ftd.hex` to the chip. Attempting to open the CLI has the same results as last time.

jwhui, 2020-12-30:
@purvisb , have you tried reaching out to makerdiary to see if they have any suggestions?

purvisb, 2020-12-30:
I just emailed support@makerdiary.com. I'll post any relevant suggestions on this issue unless told otherwise.

purvisb, 2021-01-06:
I have not heard back from makerdiary. I spent several hours troubleshooting and I think I have a solution. But I don't know why this resolves the issue:

Make and burn the firmware with USB=1. I (luckily) observed that when you do this, ttyACM0 and ttyACM1 populate in /dev. This only seems to happen when the USB=1 parameter is supplied. If you try and connect to ttyACM0, it gives the same error as before. But if you try and connect to ttyACM1, it works. I think ttyACM0 is for "DAPLink CMSIS-DAP" and ttyACM1 is for "nrf528xx OpenThread Device". Regardless, I was able to interface with the chip via the command:
`sudo /usr/local/bin/wpanctl -I utun7`

Is this expected behavior?

jwhui, 2021-01-06:
> Make and burn the NCP firmware with USB=1. I (luckily) observed that when you do this, ttyACM0 and ttyACM1 populate in /dev. This only seems to happen when the USB=1 parameter is supplied. If you try and connect to ttyACM0, it gives the same error as before. But if you try and connect to ttyACM1, it works. I think ttyACM0 is for "DAPLink CMSIS-DAP" and ttyACM1 is for "nrf528xx OpenThread Device". Regardless, I was able to interface with the chip via the command:
> `sudo /usr/local/bin/wpanctl -I utun7`
> 
> Is this expected behavior?

@purvisb , great to hear that you were able to figure it out.

It appears that this is expected behavior. From their [documentation](https://wiki.makerdiary.com/nrf52840-mdk/):

> The kit contains a Microchip USB 2.0 Hi-Speed hub controller with two downstream ports: one for DAPLink interface and one for nRF52840 USB device controller.

The `USB=1` option enables the USB controller on the nRF52840.

purvisb, 2021-01-07:
That makes sense.

I also see that [your documentation](https://github.com/openthread/openthread/blob/master/examples/platforms/nrf528xx/nrf52840/README.md) includes USB=1 as a parameter when making the firmware. I did follow this documentation. But at the step where this is burned to the chip, I never though to re-confirm that I was using the right ttyACM file. Because I thought I already confirmed I was using the right one previously. I was mistaking ttyACM0 as the one for openthread, when really it was for the DAPLink firmware.

I don't know why the makerdiary documentation does not include USB=1 as a parameter. I will close the issue I opened with them with the suggestion to add this information.

Thank you for your assistance, this issue can probably be closed.

closed 2021-01-07 by jwhui

ChauhanPD, 2021-03-12:
I am facing the same issue with wpantund as well as NCP device. I have taken the latest commit of wpantund, but still the same error.
In my set-up, I have nrf52840 chip is connected via UART with the host processor (i.MX6ULL) on our custom board.
I have taken out the RX, TX, CTS, RTS and GND from the nrf52840 UART which is connected to the host processor and connected with the PC (ubuntu) to give command from PC terminal.

I have flashed nrf52840 Chip with cli-ftd.hex file which is generated with below command:
> make -f examples/Makefile-nrf52840 clean
> make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 UDP_PROXY=1 BORDER_AGENT=1 LINK_RAW=1 SERVICE=1

and flashed via j-link using below command :
nrfjprog -f nrf52 --chiperase --program output/nrf52840/bin/ot-cli-ftd.hex --reset

I have gone through many posts/forums, but no progress for last two days. I think my set-up is different from all as I am not using any mdk or dongle of nrf52840. Let me again clarify my set-up. I have custom board having i.MX6ULL as host processor and it is connected to nrf52840 via UART. I have taken the same 4 lines (TX, RX, CTS and RTS) and connected to the FTDI cable to access nrf52840 using cli application.

Kindly help / guide me.

purvisb, 2021-03-12:
@ChauhanPD I would recommend you open a [new issue](https://github.com/openthread/openthread/issues/new/choose). The OpenThread team is pretty responsive.

ChauhanPD, 2021-03-15:
Thanks purvisb for your comment. I have opened this as a new issue. I thought the observations for error are the same, so maybe I should continue here. The only difference is of the hardware setup.