openthread/openthread #6005: [github-actions] Cert_9_2_04_ActiveDataset.py fails intermittently

jwhui, 2020-12-23:
**Describe the bug**

tests/scripts/thread-cert/Cert_9_2_04_ActiveDataset.py fails intermittently.

**Console/log output**

- [logs_47184.zip](https://github.com/openthread/openthread/files/5737499/logs_47184.zip)
- [packet-verification-pcaps.zip](https://github.com/openthread/openthread/files/5737502/packet-verification-pcaps.zip)

FiveDimensions, 2020-12-31:
In Step 3, should filter MGMT_ACTIVE_SET.rsp with Accept, but there is a LEAD_PET.rsp sent out by Leader with the same Message ID(packet index# 36) in Step 2 @librasungirl @simonlingoogle 
[image]

simonlingoogle, 2020-12-31:
@FiveDimensions @librasungirl 

Seems wireshark matched PKT 36 as the ACK of PKT26 because they have same `TKN`.

It's strange that PKT 34 and PKT 26 use the same `TKN: 89 03`. CoAP should generate random token for each new request.

librasungirl, 2021-01-03:
> @FiveDimensions @librasungirl
> 
> Seems wireshark matched PKT 36 as the ACK of PKT26 because they have same `TKN`.
> 
> It's strange that PKT 34 and PKT 26 use the same `TKN: 89 03`. CoAP should generate random token for each new request.

There is no guarantee in existing implementation that the random token of 2 bytes would be unique as required by RFC7252.

Some possible options
1. increase the token length (valid token length [0, 8]) to reduce the probability of generating the same token - the disadvantage is the extra CoAP header cost
2. Update Wireshark to match request/response properly as Message ID should also be considered when matching the CON req and ACK for the piggybacked response, per RFC7252 Section 5.3.2

 > The exact rules for matching a response to a request are as follows:
 > 1. The source endpoint of the response MUST be the same as the destination endpoint of the original request.
> 2. In a piggybacked response, the Message ID of the Confirmable request and the Acknowledgement MUST match, and the tokens of the response and original request MUST match. In a separate response, just the tokens of the response and original request MUST match.

Thoughts? @simonlingoogle @jwhui @FiveDimensions

jwhui, 2021-01-04:
I think we should update the matching to conform with RFC 7252 Section 5.3.2.

jwhui, 2021-07-21:
Have not seen this failure for a while. Closing.

closed 2021-07-21 by jwhui