openthread/openthread #6010: [github-actions] v1_2-rcp.exp fails intermittently

jwhui, 2020-12-24:
**Describe the bug**

tests/scripts/expect/v1_2-rcp.exp fails intermittently.

**Console/log output**

[logs_47454.zip](https://github.com/openthread/openthread/files/5740384/logs_47454.zip)

brianljt, 2020-12-25:
Thanks for raising this up. This could be caused by task scheduling on both end on simulation platform. Possible solutions are,
1. Increase the sample window size and start receive earlier to tolerate scheduling.
2. Or try with more pings and consider the test case pass with ping lost ratio below certain threshold.

I will take a look at it and update with more info.

brianljt, 2021-01-06:
As suggested by @bukepo , I'm trying to add V1.2-* test cases into posix test and use virtual time with better simulation time accuracy for CSL tests.

PR #6070 (2021-02-18): [test] add posix verification for OT 1.2
This change,
1. Adds new github action to verify posix app for OT1.2
2. Only enable compile flag `OT_SIMULATION_VIRTUAL_TIME_UART` in script/test for ot-rcp build target
3. Allow mix use of posix node and ot-cli-mtd node by changing python test script to accept node type as parameter
4. Remove CSL test from expect as it is covered by thread-1-2-posix tests. Fix #6010.
closed 2021-02-18 by jwhui