openthread/openthread #6011: Possible different datasets in the same network

Moandor-y, 2020-12-25:
**Describe the bug**

If we form a network with some nodes, and change some fields in the dataset (e.g. mesh local prefix) of some nodes, while keeping master key, channel, pan id the same values, they will still form a single network, but can't talk to others because of the different network parameters.

**To Reproduce**

Setup 2 nodes. Initialize the first one.
```
> dataset init new
Done
> dataset commit active
Done
> ifconfig up
Done
> thread start
Done
> state
leader
Done
> dataset active
Active Timestamp: 1
Channel: 17
Channel Mask: 0x07fff800
Ext PAN ID: 735c87b49993c682
Mesh Local Prefix: fd2d:5f03:f5d8:c6::/64
Master Key: ea14fcfb39bda2a41e5ac21e3dcd6ac3
Network Name: OpenThread-1f3a
PAN ID: 0x1f3a
PSKc: de6bbe039c144259560c54e6c76288f5
Security Policy: 672, onrcb
Done
> dataset active -x
0e080000000000010000000300001135060004001fffe00208735c87b49993c6820708fd2d5f03f5d800c60510ea14fcfb39bda2a41e5ac21e3dcd6ac3030f4f70656e5468726561642d3166336101021f3a0410de6bbe039c144259560c54e6c76288f50c0302a0ff
Done
```

Then set the dataset in the second node to the same except mesh local prefix.
```
> dataset set active 0e080000000000010000000300001135060004001fffe00208735c87b49993c6820708fd2d5f03f5d800c60510ea14fcfb39bda2a41e5ac21e3dcd6ac3030f4f70656e5468726561642d3166336101021f3a0410de6bbe039c144259560c54e6c76288f50c0302a0ff
Done
> dataset init active
Done
> dataset meshlocalprefix fd1d:5f03:f5d8:c6::
Done
> dataset commit active
Done
> ifconfig up
Done
> thread start
Done
> state
child
Done
```

We can see they are attached, but they can't talk to each other. And they will keep their differences forever because none of their active timestamps are greater than the other.

Should we consider this an intended behavior? Or should we make changes to the dataset mechanics so that they can reach consensus eventually? @jwhui @abtink @librasungirl @bukepo @simonlingoogle 

**Additional context**

This was found in the development of <https://github.com/openthread/openthread/pull/5998>. When I tried to set the initial value of ext pan id and mesh local prefix to random values, the OTNS tests failed because they started 2 nodes at the same time, so both of them became leader, generated different datasets, then merged into one network, but still can't talk to each other.

jwhui, 2020-12-25:
This behavior is intentional. When forming an Operational Dataset, the Active Timestamp must be different to ensure that one Operational Dataset takes precedence.

taoyinzhou, 2020-12-25:
I am having the same problem.

I now need to implement a feature to control multiple OTBRs in the cloud to create a thread network.

I have also tried to create the network by setting  the same dataset to all BRs, and it results in some strange problems such as child cannot become router, etc. 

The specific problem can be found in this [post](https://groups.google.com/g/openthread-users/c/ywJgnIGoJ1c/m/bopAu3LWCQAJ?from=from_parent_mindnote).

Finally this problem was solved by @jwhui who let me use src/cli/README_DATASET.md to guide the creation of the network, and the first BR used the dataset to create the thread network, and the later ones used to set the maskerkey method Attach.

But now I am controlling in the cloud, the first BR can set the network with dataset, when I add BR connection later, there is no guarantee that there must be a previously created thread network around, at this time using maskerkey to attach will always be stuck in the deteched state. If I change to send all datasets to configure the network, it will lead to the same problem I encountered in my last [post](https://groups.google.com/g/openthread-users/c/ywJgnIGoJ1c/m/bopAu3LWCQAJ?from=from_parent_mindnote).

Now that I've seen this post, I've tried to make the timestamp of the two different when sending the dataset, but it doesn't seem to do anything, both threads are in different chanel, both are leaders.

Is there any way to solve this compatibility problem of having to create multiple thread networks with dataset?

Moandor-y, 2020-12-25:
> This behavior is intentional. When forming an Operational Dataset, the Active Timestamp must be different to ensure that one Operational Dataset takes precedence.

If this is intentional, I think we should keep default values of extended pan id and mesh local prefix in the code. If we randomize the initial values of these parameters, 2 nodes with the same master key, channel, pan id may not form a correct network. Also, randomizing only the master key should be secure enough. Is this OK? @bukepo

Moandor-y, 2020-12-25:
@taoyinzhou I think you can set the second BR's full dataset to the exact same one in the first BR. If you set only the master key, it will only try to attach to an existing network. But if you set all the fields, it will create a network and become a leader if there is no compatible network.

For the timestamps, you need to meet 2 conditions: 1. all nodes have the same channel, pan id, master key, 2. the leader's active timestamp is greater than the others'. Then the leader's dataset will take precedence.

simonlingoogle, 2020-12-25:
Should be fine if we only make the masterkey random.

jwhui, 2020-12-25:
> If this is intentional, I think we should keep default values of extended pan id and mesh local prefix in the code. If we randomize the initial values of these parameters, 2 nodes with the same [...]

The Extended PAN ID is intended to provide a unique identifier for a Thread network, so it should be randomly generated.

When joining a node to an existing network, it should be sufficient to set just the master key (and, optionally, channel and PAN ID) as long as the Active Timestamp is not configured - see [src/cli/README_DATASET.md](https://github.com/openthread/openthread/blob/master/src/cli/README_DATASET.md). When commissioning nodes out-of-band, then we need to provide the full Operational Dataset.

Can you clarity the use case that is not satisfied by the above?

jwhui, 2020-12-25:
> Is there any way to solve this compatibility problem of having to create multiple thread networks with dataset?

@taoyinzhou , when commissioning multiple nodes out-of-band, you should  provide the exact same Operational Dataset. This is most easily done using the octet form.

1. To retrieve the dataset:
```
> dataset active -x
0e080000000000010000000300001035060004001fffe002084eb74ab03c56e6d00708fdc7fe165c83a67805108e2104f183e698da87e96efc1e45aa51030f4f70656e5468726561642d383631310102861104108d6273023d82c841eff0e68db86f35740c030000ff
Done
```

2. To set the dataset:
```
dataset set active 0e080000000000010000000300001035060004001fffe002084eb74ab03c56e6d00708fdc7fe165c83a67805108e2104f183e698da87e96efc1e45aa51030f4f70656e5468726561642d383631310102861104108d6273023d82c841eff0e68db86f35740c030000ff
Done
```

simonlingoogle, 2020-12-26:
> When joining a node to an existing network, it should be sufficient to set just the master key (and, optionally, channel and PAN ID) as long as the Active Timestamp is not configured

The issue we currently reveal in this case is that the joining node might fail to join (due to random packet loss) and become a Leader after a short period of time. It will generate an active operational dataset using active timestamp 0. 

When it finally attaches to the other device, they won't synchronize the active dataset if they are both using active timestamp 0. 

This projects a potential connectivity risk to network configurations in this way.

jwhui, 2020-12-26:
> > When joining a node to an existing network, it should be sufficient to set just the master key (and, optionally, channel and PAN ID) as long as the Active Timestamp is not configured
> 
> The issue we currently reveal in this case is that the joining node might fail to join (e.x. due to radio interference) and become a Leader after a short period of time. It will generate an active operational dataset using active timestamp 0.
> 
> When it finally attaches to the other device, they won't synchronize the active dataset if they are both using active timestamp 0.

The intent is that a device must not become a leader if the Active Operational Dataset is incomplete: https://github.com/openthread/openthread/blob/471cf8bedf77796600ce7e2b4deaeb834cd97e44/src/core/thread/mle_router.cpp#L189

At the same time, we should also work to remove:
https://github.com/openthread/openthread/blob/471cf8bedf77796600ce7e2b4deaeb834cd97e44/src/core/meshcop/dataset_manager_ftd.cpp#L307

All existing docs have moved away from using `GenerateLocal()` (for example, [src/cli/README_DATASET.md](https://github.com/openthread/openthread/blob/master/src/cli/README_DATASET.md)).

taoyinzhou, 2020-12-26:
@jwhui I have a small question, the thread seems to have the ability to switch channels automatically, is there such a situation, when one of the devices is shut down, the other thread device switched channels, after a while the original one came online, will it become two thread networks because the channels are different?

Moandor-y, 2020-12-26:
> The Extended PAN ID is intended to provide a unique identifier for a Thread network, so it should be randomly generated.
>
> When joining a node to an existing network, it should be sufficient to set [...]

For 2 nodes, set the same channel, pan id, master key (using corresponding commands, without dataset), then start them at the same time, there is a chance that they both become leaders and generate their own datasets. They will merge into one network while keeping the different parameters. 

Although this is not the recommended way of forming networks, OTNS is currently using it. Maybe we should make changes to OTNS?

jwhui, 2020-12-28:
> @jwhui I have a small question, the thread seems to have the ability to switch channels automatically, is there such a situation, when one of the devices is shut down, the other thread device [...]

@taoyinzhou , Thread uses the Active Timestamp to determine which Operational Dataset takes precedent. You can enable[ `OPENTHREAD_CONFIG_ANNOUNCE_SENDER_ENABLE`](https://github.com/openthread/openthread/blob/471cf8bedf77796600ce7e2b4deaeb834cd97e44/src/core/config/announce_sender.h#L49) to ensure that devices on different channels will find each other.

jwhui, 2020-12-28:
> Although this is not the recommended way of forming networks, OTNS is currently using it. Maybe we should make changes to OTNS?

@Moandor-y , I think we should update OTNS to use the recommended method for forming Thread networks.

taoyinzhou, 2020-12-29:
@jwhui ,
1. Does OPENTHREAD_CONFIG_ANNOUNCE_SENDER_ENABLE apply to MTD, so that the end device can also follow the channel switching of OTBR?
2. Could this solve the problem that if the end device is turned off first and the OTBR channel is switched, can the end device be switched back when it is powered on?
3. Do you recommended to set the channel at the beginning and not to let it switch channels? Because I am worried that when the OTBR or the end device is not powered on, the other devices may switch channels.
4. Is it possible to call the function [otchannelmanagersetautochannelselectionenabled ](https://openthread.io/reference/group/api-channel-manager#otchannelmanagersetautochannelselectionenabled)or use the command "channel manage auto 0" to achieve the function of not switching channels?

simonlingoogle, 2020-12-29:
> > Although this is not the recommended way of forming networks, OTNS is currently using it. Maybe we should make changes to OTNS?
> 
> @Moandor-y , I think we should update OTNS to use the recommended method for forming Thread networks.

Agree that OTNS is not using the ideal way of configuring network parameters. However, it does reveal a potential connectivity issue we currently have. I think it's better that we resolve the actual underlying issue if we are to change the default parameters.

Currently, two nodes becoming Leader at the same time does not have connectivity issue since they are default parameters. 

```
node 1> dataset active
Active Timestamp: 0
Channel: 11
Channel Mask: 0x07fff800
Ext PAN ID: dead00beef00cafe
Mesh Local Prefix: fdde:ad00:beef:0::/64
Master Key: 00112233445566778899aabbccddeeff
Network Name: OpenThread
PAN ID: 0xface
PSKc: 2985393c5fc15f1b2bcaa51524663105
Security Policy: 672, onrcb
Done

node 2> dataset active
Active Timestamp: 0
Channel: 11
Channel Mask: 0x07fff800
Ext PAN ID: dead00beef00cafe
Mesh Local Prefix: fdde:ad00:beef:0::/64
Master Key: 00112233445566778899aabbccddeeff
Network Name: OpenThread
PAN ID: 0xface
PSKc: 83223340a72089ef71223a1ff26c9f87
Security Policy: 672, onrcb
Done
```

jwhui, 2020-12-29:
> Agree that OTNS is not using the ideal way of configuring network parameters. However, it does reveal a potential connectivity issue we currently have. I think it's better that we resolve the actual underlying issue if we are to change the default parameters.
> 
> Currently, two nodes becoming Leader at the same time does not have connectivity issue since they are using the same default parameters.

I believe we should remove [`ActiveDataset::GenerateLocal()`](https://github.com/openthread/openthread/blob/471cf8bedf77796600ce7e2b4deaeb834cd97e44/src/core/meshcop/dataset_manager_ftd.cpp#L307). A Thread device should never form a Thread network using default values. Yes, this will require changes to anything that relies on this behavior.

simonlingoogle, 2020-12-29:
> I believe we should remove [`ActiveDataset::GenerateLocal()`](https://github.com/openthread/openthread/blob/471cf8bedf77796600ce7e2b4deaeb834cd97e44/src/core/meshcop/dataset_manager_ftd.cpp#L307). A [...]

Agree that the `GenerateLocal` behavior should be removed. 

@Moandor-y Will you take on this fix, which may involve fixing many test scripts, including OTNS tests?

Moandor-y, 2020-12-29:
> Agree that the `GenerateLocal` behavior should be removed.
> 
> @Moandor-y Will you take on this fix, which may involve fixing many test scripts, including OTNS tests?

I agree. Just to confirm my understanding, by removing `GenerateLocal`, the major change is that if the user doesn't specify a dataset, no matter if the user has set some parameters directly, the node won't form a new network.

@simonlingoogle can you fix the OTNS part? I can take the rest of them. I think we can start after <https://github.com/openthread/openthread/pull/5998> is merged as it's changing many tests.

simonlingoogle, 2020-12-29:
> I agree. Just to confirm my understanding, by removing `GenerateLocal`, the major change is that if the user doesn't specify a dataset, no matter if the user has set some parameters directly, the [...]

I will fix OTNS part.

jwhui, 2021-01-06:
> Just to confirm my understanding, by removing `GenerateLocal`, the major change is that if the user doesn't specify a dataset, no matter if the user has set some parameters directly, the node won't form a new network.

Correct. OpenThread should not fabricate its own Operational Dataset without the user explicitly telling it to do so.

jwhui, 2022-12-07:
Closing stale issue.

closed 2022-12-07 by jwhui