openthread/openthread #6013: [efr32] entropy: otPlatEntropyGet may fail

dismirlian, 2020-12-26:
**Describe the bug**
The efr32 platform code calls `mbedtls_hardware_poll`, which uses the TRNG (at least in efr32mg1x):

https://github.com/openthread/openthread/blob/a6e8848a5eb9f037693afe11583857d6a8ff96bc/examples/platforms/efr32/src/entropy.c#L51-L55

The TRNG might not have any new samples available, so `partialLen` may be 0. For instance, this is the call stack during startup: 

```
otPlatEntropyGet() at ot-platform-entropy.c:44 0x2117c	
ot::RandomManager::Entropy::HandleMbedtlsEntropyPoll() at random_manager.cpp:175 0x47c02	
entropy_gather_internal() at entropy.c:289 0x19db2	
mbedtls_entropy_func() at entropy.c:375 0x19e72	
mbedtls_ctr_drbg_reseed() at ctr_drbg.c:345 0x1704c	
mbedtls_ctr_drbg_seed() at ctr_drbg.c:419 0x170fa	
ot::RandomManager::CryptoCtrDrbg::Init() at random_manager.hpp:129 0x314e4	
ot::RandomManager::RandomManager() at random_manager.cpp:70 0x3152a	
ot::Instance::Instance() at linked_list.hpp:108 0x475b0	
ot::Instance::InitSingle() at new.hpp:45 0x30aa0	
otInstanceInitSingle() at instance_api.cpp:60 0x461ee	
```

mbedTLS requests 128 bytes of entropy, but the TRNG has only 16 words in its buffer, so the second call to `mbedtls_hardware_poll` returns 0 because it returns immediately if there are no words available; see sl_crypto/trng.c:

```C
int mbedtls_trng_poll( mbedtls_trng_context *ctx,
                       unsigned char *output, size_t len, size_t *olen )
{
    TRNG_TypeDef *trng = ctx->trng;
    unsigned char *outptr = output;
    size_t output_len = 0;
    size_t count = 0;
    size_t available;
    int ret = 0;

    while (len > 0)
    {
        available = trng->FIFOLEVEL * 4;
        if (available == 0)
        {
            break;
        }

```

This is not a failure; for example, during initialization, sl_crypto/trng.c has the following waiting loop:

```C
    /* Wait for TRNG to complete startup tests and start filling the FIFO. */
    for (i=0; (TRNG0->FIFOLEVEL == 0) && (i<TRNG_STARTUP_TEST_WAIT_RETRY); i++);
    EFM_ASSERT(i<TRNG_STARTUP_TEST_WAIT_RETRY);
```

In my case, this caused my device to fail attach after joining successfully (don't know exactly why...).

I removed line 54, and everything works as expected, because the TRNG fills its FIFO given enough time.

I don't know what the best fix would be.

Thanks!

romacdon, 2021-01-06:
@dismirlian you are correct.  That is an issue and we also recently found this in our Simplicity Studio version and fixed it the exact same way.  We removed that same line.  That is the correct fix.  We are in the process of updating GitHub to support the latest Silabs GSDK and will include this fix with that update but if you want to push in that change now that would be great.

suveshpratapa, 2021-01-08:
@dismirlian @romacdon Opened a PR to fix the efr32 example https://github.com/openthread/openthread/pull/6050

PR #6050 (2021-01-08): [efr32] fix for trng entropy API
When obtaining entropy data, trng_get_random checks the status for any noise, and if so, cleans its queues and keeps the entropy accumulator of mbedtls running.

In the process, TRNG might not have any new samples available, so a zero length output is valid, and we should keep running rather than exit fatally.

Resolves #6013
closed 2021-01-08 by jwhui