openthread/openthread #6014: [random manager] should we check the return value of mbedtls_ctr_drbg_seed?

dismirlian, 2020-12-26:
Related to #6013, I experienced a hard-to-debug issue, which was masked by the fact that `RandomManager::CryptoCtrDrbg::Init` doesn't check the return value of `mbedtls_ctr_drbg_seed`, leaving the user unaware that the random hardware generator failed.

https://github.com/openthread/openthread/blob/7fe84c59cdf6ce1070ac2d877d826779a04df140/src/core/common/random_manager.cpp#L191-L195

Is this intended? Should we print an error message? Should we assert?

Thanks.

jwhui, 2021-01-06:
@dismirlian , thanks for raising this issue.

At a minimum, I think it would be great to emit a critical-level log and assert. Would you be willing to submit a PR for that?

dismirlian, 2021-01-08:
Hi @jwhui I can submit a PR. Would it be sufficient to assert? Do you think there should be also a critical level message?

jwhui, 2021-01-08:
Yes, I think we should have both a critical-level log and assert (as mentioned in https://github.com/openthread/openthread/issues/6014#issuecomment-755770014).

PR #6071 (2021-01-15): [random-manager] fail if unable to seed the CTR DRBG
Resolves #6014.
closed 2021-01-15 by jwhui