openthread/openthread #6015: Platform test benches

dismirlian, 2020-12-26:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
Although OT core is exhaustively tested, the platform abstractions are not.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
OpenThread could provide a test bench for the platforms. These test benches would test the platform compliance to what OT core expects. Of course, these should be run directly on hardware. Ideally, they would include all the APIs (radio, alarm, entropy, flash, etc.).

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
None.

**Additional context** Add any other context or screenshots about the feature request here.
None.

jwhui, 2021-01-06:
@dismirlian , thanks for raising this issue.

I think it would be great to have a test suite that validates a given platform implementation. Would you be willing to start this effort?

dismirlian, 2021-01-14:
Hi @jwhui, I'm sure I don't fully understand all the details and use cases of the platform API, and also I'm quite busy right now. Maybe, with your help (the OT team), I can kick this off in some weeks.

jwhui, 2023-06-27:
Closing stale issue.

closed 2023-06-27 by jwhui