openthread/openthread #6016: Commissioner commands lead to "Error 7: InvalidArgs"

MarcelloDavis, 2020-12-27:
Hello,

I have just built my own openthread mesh-network based on a Raspberry Pi 3B with a nRF52840 Dongle working as the borderrouter and a Thunderboard Sense 2 (EFR32MG12) as a child device. When I type `RouterTable` on the RPi device, it shows me a table with two entries: The borderrouter itself and the Thunderboard. Also, when I ping the borderrouter IP-adress from my Thunderboard, I receive and answer displayed in the Thunderboard console. So far so good, everything's working.

But it seems there is an issue with my borderrouter. It started with the commissioning process on the RPi: When I tried to commission the Thunderboard into my thread-network (using the command `commissioner joiner add * J01NME`), I get an error saying: "`Error 7: InvalidArgs`". In fact, the RPi knows the commissioner commands (_`commissioner state` provides the output `active`_), but it seems like it isn't familiar with the syntax. Possibly I am using an outdated commissioner version?

So I stopped the thread network, formed it again and this time I set the Thunderboard as commissioner. On the Thunderboard I am able to type the exact same command `commissioner joiner add * J01NME` and this time I get the output `Done`.

Also, when I typed in `commissioner sessionid` on the RPi, the output was some number such as 1400 (I tried it several times). Shouldn't it be 0, when the RPi itself acts as commissioner? Or may the number differ from 0?

_If you want to follow my building tutorial, these are the very steps I did in order to set up my RPi (as a borderrouter). These steps can also be found in the official openthread documentary https://openthread.io/guides/border-router/build:
On the RPi:_
```
git clone https://github.com/openthread/ot-br-posix.git
cd ot-br-posix
./script/bootstrap
./script/setup
OTBR_AGENT_OPTS="-I wpan0 spinel+hdlc+uart:///dev/ttyACM0"
```
_For the nRF52840 Dongle:_
```
git clone https://github.com/openthread/openthread.git
cd openthread
./script/bootstrap
./bootstrap
make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 TMF_PROXY=1 DHCP6_SERVER=1 DHCP6_CLIENT=1 DISABLE_SPI=1 SNTP_CLIENT=1 DISABLE_BUILTIN_MBEDTLS=1 USB=1 BOOTLOADER=USB
cd output/nrf52840/bin
arm-none-eabi-objcopy -O ihex ot-rcp ot-rcp.hex
nrfutil pkg generate --hw-version 52 --sd-req 0x0 --application-version 1 --application ot-rcp.hex out.zip
sudo nrfutil dfu usb-serial -pkg out.zip -p /dev/ttyACM0 -b 115200
```
_Then reboot the RPi and I can see there are three services running:_
```
avahi-daemon.service
otbr-agent.service
otbr-web.service
```
_As the borderrouter is now set up, I form a network using:_
```
dataset init new
dataset
```
_and get an output something like the following:_
```
Active Timestamp: 1
Channel: 13
Channel Mask: 07fff800
Ext PAN ID: d63e8e3e495ebbc3
Mesh Local Prefix: fd3d:b50b:f96d:722d/64
Master Key: dfd34f0f05cad978ec4e32b0413038ff
Network Name: OpenThread-8f28
PAN ID: 0x8f28
PSKc: c23a76e98f1a6483639b1ac1271e2e27
Security Policy: 0, onrcb
Done
```
_then I commit this dataset as active and start the thread network:_
```
dataset commit active
ifconfig up
thread start
```
_Then I installed the commissioner on my RPi accoriding to the openthread documentary https://openthread.io/guides/commissioner/build :_
```
git clone https://github.com/openthread/ot-commissioner
cd ot-commissioner
./script/bootstrap.sh
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/usr/local -GNinja ..
ninja -j1
( ./tests/commissioner-test )
ninja install
( commissioner-cli -h )
```
_and finally I changed the pskc in the non-ccm-config.json file to the one the `dataset` output delivered._
_Now that the commissioner should be installed and configured, I tried the commissioning as follows:_
`commissioner start`
_(here I get the output `Done`).
As I am trying to set the pskc, I run into the error 7: InvalidArgs:_
`commissioner joiner add * J01NME`

I hope that you can help me and thank you very much in advance!
Marcel

wgtdkp, 2020-12-28:
Hi,

The standalone commissioner-cli uses different CLI commands, you can check
the *commissioner* command with command *help commissioner*.
See also
https://openthread.io/guides/border-router/external-commissioning?comm=ot-commissioner
for an example of commissioning with commissioner-cli.

BRs,
Kangping

MarcelloDavis, 2020-12-28:
@wgtdkp thank you very much for your quick response!
I'll try this out.

So I installed the standalone commissioner. Is it also possible that I don't have to install the standalone commissioner and just use the commissioner that I've set in the build configuration:
`make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 [...]`
so that I can just use the command `commissioner add joiner` without `ot-cli`?

wgtdkp, 2020-12-28:
Yes, you can just use the commissioner built directly from openthread with
COMMISSIONER=1.

closed 2021-01-05 by jwhui