openthread/openthread #6028: ot-daemon is not compatible with time sync

TPolzer, 2020-12-31:
I built the RCP for the nrf52840 dongle and ot-daemon at 22b043bf62328e89773fa61c9bb80770f81a4bef with the following commands:
```
make -f examples/Makefile-nrf52840 -j20 USB=1 BOOTLOADER=USB TIME_SYNC=1
make -f src/posix/Makefile-posix DAEMON=1 TIME_SYNC=1 -j20
```
To make ot-daemon build, I implemented `uint16_t otPlatTimeGetXtalAccuracy(void)` as `return 40;` (I don't there's a way to query clock accuracy under Linux?).

I tried running ot-daemon, but it crashes with a nullpointer dereference shortly after I issue the `thread start` command:
```
(gdb) bt
#0  0x000055555557400c in ot::Mac::TxFrame::SetTimeIeOffset (aOffset=0 '\000', this=0x555555624bf0 <sRadioSpinel+8784>)
    at ~redacted/openthread-head/src/posix/../../src/core/mac/mac_frame.hpp:1350
#1  ot::Mac::Mac::BeginTransmit (this=0x55555561f380 <ot::gInstanceRaw+38208>)
    at ~redacted/openthread-head/src/posix/../../src/core/mac/mac.cpp:1234
#2  0x000055555555a8b5 in main (argc=2, argv=0x7fffffffe588) at ~redacted/openthread-head/src/posix/../../src/posix/main.c:347
(gdb) list
1350	    void SetTimeIeOffset(uint8_t aOffset) { mInfo.mTxInfo.mIeInfo->mTimeIeOffset = aOffset; }
(gdb) p mInfo.mTxInfo
$7 = {mAesKey = 0x0, mIeInfo = 0x0, mTxDelay = 0, mTxDelayBaseTime = 0, mMaxCsmaBackoffs = 4 '\004', mMaxFrameRetries = 3 '\003', mIsARetx = false, 
  mCsmaCaEnabled = true, mCslPresent = false, mIsSecurityProcessed = false}
(gdb) p *this
$9 = {<ot::Mac::Frame> = {<otRadioFrame> = {mPsdu = 0x555555624abf <sRadioSpinel+8479> "Y\230\215g~\377\377\376\377\r", mLength = 126, mChannel = 11 '\v', 
      mRadioType = 0 '\000', mInfo = {mTxInfo = {mAesKey = 0x0, mIeInfo = 0x0, mTxDelay = 0, mTxDelayBaseTime = 0, mMaxCsmaBackoffs = 4 '\004', 
          mMaxFrameRetries = 3 '\003', mIsARetx = false, mCsmaCaEnabled = true, mCslPresent = false, mIsSecurityProcessed = false}, mRxInfo = {
          mTimestamp = 0, mAckFrameCounter = 0, mAckKeyId = 0 '\000', mRssi = 0 '\000', mLqi = 0 '\000', mAckedWithFramePending = false, 
          mAckedWithSecEnhAck = false}}}, <No data fields>}, <No data fields>}
(gdb) p sRadioSpinel
$11 = {mInstance = 0x555555615e40 <ot::gInstanceRaw>, mRxFrameBuffer = {<ot::Hdlc::FrameWritePointer> = {
      mWritePointer = 0x5555556229b6 <sRadioSpinel+22> "\203\006\066g~\247\361\335N\273\246\201\306\025\026\027\030\031\032\346(3-01602-g22b043bf6; NRF52840; Dec 31 2020 13:25:42", mRemainingLength = 8188}, 
    mBuffer = "\000\000\000\000\203\006\066g~\247\361\335N\273\246\201\306\025\026\027\030\031\032\346(3-01602-g22b043bf6; NRF52840; Dec 31 2020 13:25:42\000\226\026", '\000' <repeats 8113 times>, mWriteFrameStart = 0x5555556229b2 <sRadioSpinel+18> ""}, mSpinelInterface = {
    mReceiveFrameCallback = 0x5555555b4670 <ot::Spinel::RadioSpinel<ot::Posix::HdlcInterface, RadioProcessContext>::HandleReceivedFrame(void*)>, 
    mReceiveFrameContext = 0x5555556229a0 <sRadioSpinel>, mReceiveFrameBuffer = @0x5555556229a8, mSockFd = 4, mBaudRate = 115200, mHdlcDecoder = {
      mState = ot::Hdlc::Decoder::kStateSync, mWritePointer = @0x5555556229a8, 
      mFrameHandler = 0x5555555b7220 <ot::Posix::HdlcInterface::HandleHdlcFrame(void*, otError)>, mContext = 0x5555556249c0 <sRadioSpinel+8224>, 
      mFcs = 65535, mDecodedLength = 0}}, mCmdTidsInUse = 0, mCmdNextTid = 4 '\004', mTxRadioTid = 0 '\000', mWaitingTid = 0 '\000', mWaitingKey = 0, 
  mPropertyFormat = 0x0, mPropertyArgs = {{gp_offset = 24, fp_offset = 48, overflow_arg_area = 0x7fffffffdf50, reg_save_area = 0x7fffffffde90}}, 
  mExpectedCommand = 0, mError = OT_ERROR_NONE, mRxPsdu = '\000' <repeats 126 times>, 
  mTxPsdu = "Y\230\215g~\377\377\376\377\r\000\000\000\000\000\300\276\242\245o[\ni䵖\203i\303\315\376I\373\360\024\351\024\351\354\063\000\000\204\000\000\000\000\002\000\000\000\000\001\066\001a\001b\001b\001e\001\064\001d\001d\001\064\001\067\001a\001b\001\062\001c\001c\001\067\001\060\001\060\001\060\001\060\001\060\001\060\001\060\001\060\001\060\001\060\001\060\001\060\001\060\001\070\001e\001f\003ip6\000\000\000\000\000\000", 
  mAckPsdu = '\000' <repeats 126 times>, mRxRadioFrame = {mPsdu = 0x555555624a40 <sRadioSpinel+8352> "", mLength = 0, mChannel = 0 '\000', 
    mRadioType = 0 '\000', mInfo = {mTxInfo = {mAesKey = 0x0, mIeInfo = 0x0, mTxDelay = 0, mTxDelayBaseTime = 0, mMaxCsmaBackoffs = 0 '\000', 
        mMaxFrameRetries = 0 '\000', mIsARetx = false, mCsmaCaEnabled = false, mCslPresent = false, mIsSecurityProcessed = false}, mRxInfo = {
        mTimestamp = 0, mAckFrameCounter = 0, mAckKeyId = 0 '\000', mRssi = 0 '\000', mLqi = 0 '\000', mAckedWithFramePending = false, 
        mAckedWithSecEnhAck = false}}}, mTxRadioFrame = {mPsdu = 0x555555624abf <sRadioSpinel+8479> "Y\230\215g~\377\377\376\377\r", mLength = 126, 
    mChannel = 11 '\v', mRadioType = 0 '\000', mInfo = {mTxInfo = {mAesKey = 0x0, mIeInfo = 0x0, mTxDelay = 0, mTxDelayBaseTime = 0, 
        mMaxCsmaBackoffs = 4 '\004', mMaxFrameRetries = 3 '\003', mIsARetx = false, mCsmaCaEnabled = true, mCslPresent = false, 
        mIsSecurityProcessed = false}, mRxInfo = {mTimestamp = 0, mAckFrameCounter = 0, mAckKeyId = 0 '\000', mRssi = 0 '\000', mLqi = 0 '\000', 
        mAckedWithFramePending = false, mAckedWithSecEnhAck = false}}}, mAckRadioFrame = {mPsdu = 0x555555624b3e <sRadioSpinel+8606> "", mLength = 0, 
    mChannel = 0 '\000', mRadioType = 0 '\000', mInfo = {mTxInfo = {mAesKey = 0x0, mIeInfo = 0x0, mTxDelay = 0, mTxDelayBaseTime = 0, 
        mMaxCsmaBackoffs = 0 '\000', mMaxFrameRetries = 0 '\000', mIsARetx = false, mCsmaCaEnabled = false, mCslPresent = false, 
        mIsSecurityProcessed = false}, mRxInfo = {mTimestamp = 0, mAckFrameCounter = 0, mAckKeyId = 0 '\000', mRssi = 0 '\000', mLqi = 0 '\000', 
        mAckedWithFramePending = false, mAckedWithSecEnhAck = false}}}, mTransmitFrame = 0x0, mExtendedAddress = {m8 = "~ºt\335N\273\246"}, 
  mShortAddress = 65534, mPanId = 32359, mRadioCaps = 31 '\037', mChannel = 11 '\v', mRxSensitivity = -100 '\234', mTxError = OT_ERROR_NONE, 
  mVersion = "OPENTHREAD/20191113-01602-g22b043bf6; NRF52840; Dec 31 2020 13:25:42", '\000' <repeats 59 times>, mIeeeEui64 = {
    m8 = "\364\316\066\354\317`", <incomplete sequence \321>}, 
  mState = ot::Spinel::RadioSpinel<ot::Posix::HdlcInterface, RadioProcessContext>::kStateReceive, mIsPromiscuous = false, mIsReady = true, 
  mSupportsLogStream = false, mIsTimeSynced = false, mDiagMode = false, mDiagOutput = 0x0, mDiagOutputMaxLen = 0, mTxRadioEndUs = 18446744073709551615, 
  mRadioTimeRecalcStart = 18446744073709551615, mRadioTimeOffset = 0, mMaxPowerTable = {static kPowerDefault = 30 '\036', 
    mPowerTable = '\036' <repeats 16 times>}}
```

jwhui, 2020-12-31:
The time sync feature is not currently supported in RCP mode.

Note that the time sync feature is experimental and not a part of the Thread specification.

jwhui, 2023-06-27:
Closing stale issue.

closed 2023-06-27 by jwhui