openthread/openthread #6029: ot-daemon doesn't properly forward multicast from TUN

TPolzer, 2020-12-31:
I built the RCP for the nrf52840 dongle and ot-daemon at 22b043b with the following commands:
```
make -f examples/Makefile-nrf52840 -j20 USB=1 BOOTLOADER=USB
make -f src/posix/Makefile-posix DAEMON=1 -j20
```
I then joined to the existing  thread network I have:
```
> state
child
Done
> ipaddr
fdbf:e492:40f3:de5a:0:ff:fe00:b401
2a02:168:6425:b00:6377:795c:31d6:597d
fdbf:e492:40f3:de5a:9b8c:4dc5:c457:a1b9
fe80:0:0:0:603d:bdff:51da:f957
Done
```
Multicast from the cli looks good (I have 5 other devices in the network):
```
> ping ff03::1 16 4
Done
> 24 bytes from fdbf:e492:40f3:de5a:da97:213d:4159:6742: icmp_seq=26 hlim=64 time=10ms
24 bytes from fdbf:e492:40f3:de5a:397c:f3ce:1bce:ce60: icmp_seq=26 hlim=64 time=77ms
24 bytes from fdbf:e492:40f3:de5a:8b48:86dd:a002:56da: icmp_seq=26 hlim=64 time=116ms
24 bytes from fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b: icmp_seq=26 hlim=64 time=141ms
24 bytes from fdbf:e492:40f3:de5a:da97:213d:4159:6742: icmp_seq=27 hlim=64 time=12ms
24 bytes from fdbf:e492:40f3:de5a:397c:f3ce:1bce:ce60: icmp_seq=27 hlim=64 time=104ms
24 bytes from fdbf:e492:40f3:de5a:8b48:86dd:a002:56da: icmp_seq=27 hlim=64 time=118ms
24 bytes from fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b: icmp_seq=27 hlim=64 time=147ms
24 bytes from fdbf:e492:40f3:de5a:da97:213d:4159:6742: icmp_seq=28 hlim=64 time=10ms
24 bytes from fdbf:e492:40f3:de5a:397c:f3ce:1bce:ce60: icmp_seq=28 hlim=64 time=43ms
24 bytes from fdbf:e492:40f3:de5a:8b48:86dd:a002:56da: icmp_seq=28 hlim=64 time=69ms
24 bytes from fdbf:e492:40f3:de5a:8195:2838:718e:72ab: icmp_seq=28 hlim=64 time=80ms
24 bytes from fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b: icmp_seq=28 hlim=64 time=119ms
24 bytes from fdbf:e492:40f3:de5a:da97:213d:4159:6742: icmp_seq=29 hlim=64 time=11ms
24 bytes from fdbf:e492:40f3:de5a:397c:f3ce:1bce:ce60: icmp_seq=29 hlim=64 time=54ms
24 bytes from fdbf:e492:40f3:de5a:8b48:86dd:a002:56da: icmp_seq=29 hlim=64 time=109ms
24 bytes from fdbf:e492:40f3:de5a:8195:2838:718e:72ab: icmp_seq=29 hlim=64 time=115ms
24 bytes from fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b: icmp_seq=29 hlim=64 time=129ms
```
But multicast from the host (i.e. through the TUN device) only hits the parent node:
```
$ ping -Iwpan0 -I fdbf:e492:40f3:de5a:9b8c:4dc5:c457:a1b9 -c4 -s 16 ff03::1
PING ff03::1(ff03::1) from fdbf:e492:40f3:de5a:9b8c:4dc5:c457:a1b9 wpan0: 16 data bytes
24 bytes from fdbf:e492:40f3:de5a:9b8c:4dc5:c457:a1b9: icmp_seq=1 ttl=64 time=0.060 ms
24 bytes from fdbf:e492:40f3:de5a:da97:213d:4159:6742: icmp_seq=1 ttl=64 time=10.9 ms (DUP!)
24 bytes from fdbf:e492:40f3:de5a:9b8c:4dc5:c457:a1b9: icmp_seq=2 ttl=64 time=0.063 ms
24 bytes from fdbf:e492:40f3:de5a:da97:213d:4159:6742: icmp_seq=2 ttl=64 time=11.6 ms (DUP!)
24 bytes from fdbf:e492:40f3:de5a:9b8c:4dc5:c457:a1b9: icmp_seq=3 ttl=64 time=0.065 ms
24 bytes from fdbf:e492:40f3:de5a:da97:213d:4159:6742: icmp_seq=3 ttl=64 time=15.7 ms (DUP!)
24 bytes from fdbf:e492:40f3:de5a:9b8c:4dc5:c457:a1b9: icmp_seq=4 ttl=64 time=0.056 ms

--- ff03::1 ping statistics ---
4 packets transmitted, 4 received, +3 duplicates, 0% packet loss, time 3006ms
rtt min/avg/max/mdev = 0.056/5.482/15.656/6.409 ms
```

TPolzer, 2020-12-31:
Unicast (mostly...) works:
```
$ ping -s16 -c4 -I wpan0 fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b
PING fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b(fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b) from fdbf:e492:40f3:de5a:0:ff:fe00:b401 wpan0: 16 data bytes

--- fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b ping statistics ---
4 packets transmitted, 0 received, 100% packet loss, time 3080ms

$ ping -s16 -c4 -I wpan0 -I fdbf:e492:40f3:de5a:9b8c:4dc5:c457:a1b9 fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b
PING fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b(fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b) from fdbf:e492:40f3:de5a:9b8c:4dc5:c457:a1b9 wpan0: 16 data bytes
24 bytes from fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b: icmp_seq=1 ttl=64 time=46.1 ms
24 bytes from fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b: icmp_seq=2 ttl=64 time=45.0 ms
24 bytes from fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b: icmp_seq=3 ttl=64 time=48.4 ms
24 bytes from fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b: icmp_seq=4 ttl=64 time=47.5 ms

--- fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3004ms
rtt min/avg/max/mdev = 45.006/46.746/48.362/1.298 ms
```

jwhui, 2020-12-31:
IPv6 Hop Limit value on linux defaults to 1 for multicast messages. Try setting the Hop Limit or TTL value on the host as an argument to `ping6` - see [ping6 man page](https://linux.die.net/man/8/ping6).

TPolzer, 2020-12-31:
Thanks! That indeed fixes the multicast ping. It's still a bit weird that I'm unable to ping other nodes from the rloc mesh local address (fdbf:e492:40f3:de5a:0:ff:fe00:b401), but it's not an issue for me.

closed 2020-12-31 by TPolzer