openthread/openthread #6030: Commissioning Failures

gabekassel, 2021-01-01:
**Describe the bug** A clear and concise description of what the bug is.

I've been seeing lots of failures around commissioning a Thread device onto my network. Phone app is able to petition, but commissioning never completes successfully. I can't tell if this is a misconfiguration on the border router or some kind of incompatibility. 

> Jan  1 22:41:57 eero daemon.info /usr/sbin/ot-daemon[6327]: Thread state changed, flag: 512
> Jan  1 22:41:57 eero daemon.info /usr/sbin/ot-daemon[6327]:     Thread Netdata changed.  Notify registered clients
> Jan  1 22:42:05 eero daemon.info ot-daemon[6327]: ~ Discovery Request from c286e3691f6f87a0: version=2,joiner=1
> Jan  1 22:42:09 eero daemon.info /usr/sbin/ot-daemon[6327]: handleThreadStateChanged: flags = 200
> Jan  1 22:42:09 eero daemon.info /usr/sbin/ot-daemon[6327]: Thread state changed, flag: 512
> Jan  1 22:42:09 eero daemon.info /usr/sbin/ot-daemon[6327]:     Thread Netdata changed.  Notify registered clients
> Jan  1 22:42:26 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MAC-----: Received Beacon Request
> Jan  1 22:42:26 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MAC-----: Sending Beacon, name:eero-thread, xpanid:31de7789a2582331, id:3, ver:2, joinable:yes, native:no
> Jan  1 22:42:26 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:28, seqnum:232, type:Beacon, src:0x0000, dst:None, sec:no, ackreq:no
> Jan  1 22:42:26 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:28, seqnum:230, type:Beacon, src:0x0000, dst:None, sec:no, ackreq:no
> Jan  1 22:42:27 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Forwarded request to leader on c/la
> Jan  1 22:42:27 eero daemon.warn /usr/sbin/ot-daemon[6327]: [WARN]-MESH-CP-: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered
> Jan  1 22:42:27 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: received keep alive
> Jan  1 22:42:27 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: sent keep alive response
> Jan  1 22:42:27 eero daemon.warn /usr/sbin/ot-daemon[6327]: [WARN]-MESH-CP-: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered
> Jan  1 22:42:27 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Sent to commissioner
> Jan  1 22:42:30 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MAC-----: Received IPv6 UDP msg, len:489, chksum:1ba6, from:1aabeec536e3ff30, sec:no, prio:normal, rss:-21.0
> Jan  1 22:42:30 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MAC-----:     src:[fe80:0:0:0:18ab:eec5:36e3:ff30]:1000
> Jan  1 22:42:30 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MAC-----:     dst:[fe80:0:0:0:ccd4:89b3:e3c:7bb7]:1000
> Jan  1 22:42:30 eero daemon.warn /usr/sbin/ot-daemon[6327]: [WARN]-MESH-CP-: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered
> Jan  1 22:42:30 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: JoinerRouter::HandleUdpReceive
> Jan  1 22:42:30 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Sent relay rx
> Jan  1 22:42:30 eero daemon.warn /usr/sbin/ot-daemon[6327]: [WARN]-MESH-CP-: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered
> Jan  1 22:42:30 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Sent to commissioner
> Jan  1 22:42:30 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Sent to commissioner on c/rx
> Jan  1 22:42:42 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Forwarded request to leader on c/la
> Jan  1 22:42:42 eero daemon.warn /usr/sbin/ot-daemon[6327]: [WARN]-MESH-CP-: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered
> Jan  1 22:42:42 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: received keep alive
> Jan  1 22:42:42 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: sent keep alive response
> Jan  1 22:42:42 eero daemon.warn /usr/sbin/ot-daemon[6327]: [WARN]-MESH-CP-: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered
> Jan  1 22:42:42 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Sent to commissioner
> Jan  1 22:42:50 eero daemon.info srp-mdns-proxy[3173]: router advertisement from fe80::18ce:773a:3617:2224 (LUA) to ff02::1 hop_limit 255 on br-wan: checksum = 4619 cur_hop_limit = 0 flags = 0 router_lifetime = 0 reachable_time = 0 retransmission_timer = 0
> Jan  1 22:42:50 eero daemon.info srp-mdns-proxy[3173]:   source link layer address e0:2b:96:a1:b3:c3
> Jan  1 22:42:50 eero daemon.info srp-mdns-proxy[3173]:   route info: fdbc:1791:add0:: (ULA)/64 0 1800
> Jan  1 22:42:57 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Forwarded request to leader on c/la
> Jan  1 22:42:57 eero daemon.warn /usr/sbin/ot-daemon[6327]: [WARN]-MESH-CP-: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered
> Jan  1 22:42:57 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: received keep alive
> Jan  1 22:42:57 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: sent keep alive response
> Jan  1 22:42:57 eero daemon.warn /usr/sbin/ot-daemon[6327]: [WARN]-MESH-CP-: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered
> Jan  1 22:42:57 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Sent to commissioner
> Jan  1 22:43:01 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-UTIL----: ChannelMonitor: 3 [00 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 ]
> Jan  1 22:43:03 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MAC-----: Received IPv6 UDP msg, len:489, chksum:1aa6, from:1aabeec536e3ff30, sec:no, prio:normal, rss:-23.125
> Jan  1 22:43:03 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MAC-----:     src:[fe80:0:0:0:18ab:eec5:36e3:ff30]:1000
> Jan  1 22:43:03 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MAC-----:     dst:[fe80:0:0:0:ccd4:89b3:e3c:7bb7]:1000
> Jan  1 22:43:03 eero daemon.warn /usr/sbin/ot-daemon[6327]: [WARN]-MESH-CP-: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered
> Jan  1 22:43:03 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: JoinerRouter::HandleUdpReceive
> Jan  1 22:43:03 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Sent relay rx
> Jan  1 22:43:03 eero daemon.warn /usr/sbin/ot-daemon[6327]: [WARN]-MESH-CP-: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered
> Jan  1 22:43:03 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Sent to commissioner
> Jan  1 22:43:03 eero daemon.info /usr/sbin/ot-daemon[6327]: [INFO]-MESH-CP-: Sent to commissioner on c/rx

**To Reproduce** Information to reproduce the behavior, including:
Form network on BR with below dataset:

> Active Timestamp: 1
> Channel: 25
> Channel Mask: 0x07fff800
> Ext PAN ID: 31de7789a2582331
> Mesh Local Prefix: fd1f:2c15:d7f7:0::/64
> Master Key: 60151a61c91aabe6b659b65330eaac74
> Network Name: eero-thread
> PAN ID: 0xd99a
> PSKc: 4138a3c74a73125ed3e6c20f4be416d5
> Security Policy: 700, nrcb

Using ThreadGroup app, join wifi network then on joiner run:
ifconfig up
joiner start J01NU5

1. Git commit id
OT-Daemon: 22b043bf62328e89773fa61c9bb80770f81a4bef
MG1 RCP: 22b043bf62328e89773fa61c9bb80770f81a4bef
End Device: 9380ffd07c39d7a2cfe9bca8cf2c7649873e6da2

2. IEEE 802.15.4 hardware platform
RCP: EFR32MG1P
ED: EFR32MG12P

3. Build steps
4. Network topology
Single BR/router and single ED

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

gabekassel, 2021-01-02:
I did see one success tonight (which I didn't capture logs for), but then multiple failures. Here's additional logging during failures. 
[otdaemon.log](https://github.com/openthread/openthread/files/5759625/otdaemon.log)

gabekassel, 2021-01-02:
Commissioning from ot-daemon cli works perfectly

```
> Commissioner: active

> commissioner joiner add 14b457fffe6da684 J01NU5
Done
> ~ Discovery Request from fea82f9ed810fbf1: version=2,joiner=1
Commissioner: Joiner start 1aabeec536e3ff30
Commissioner: Joiner connect 1aabeec536e3ff30
Commissioner: Joiner finalize 1aabeec536e3ff30
Commissioner: Joiner end 1aabeec536e3ff30
```

wgtdkp, 2021-01-03:
@gabekassel It looks like the BR and joiner device are not compatible with each other. Is it possible for you to use the same commit for both the BR and joiner?

gabekassel, 2021-01-04:
@wgtdkp i tried that. i find commissioning works, but only intermittently. most attempts fail. here's another log where it was failing.

let me know any suggestions to further debug
[Uploading ot1.log…]()

gabekassel, 2021-01-04:
I was able to successfully commission just now and captured tcpdump on BR (didn't have OT logs handy)

I've noticed when things stop working, the joiner reports "Not Found" or "Security" as failure reason. I've also noticed the commissioning app can't petition (asks for commissioning credential again). I do see discover requets on the BR from the joiner whenever I run joiner command. 

```
01:19:23.892465 IP6 fd97:739d:386a:1:880e:4dc4:644b:5151.46893 > fd97:739d:386a:1::1.49191: UDP, length 72
01:19:23.894779 IP6 fd97:739d:386a:1::1.49191 > fd97:739d:386a:1:880e:4dc4:644b:5151.46893: UDP, length 45
01:19:25.652880 IP6 fd97:739d:386a:1::1.49191 > fd97:739d:386a:1:880e:4dc4:644b:5151.46893: UDP, length 486
01:19:25.805800 IP6 fd97:739d:386a:1:880e:4dc4:644b:5151.46893 > fd97:739d:386a:1::1.49191: UDP, length 115
01:19:25.911616 IP6 fd97:739d:386a:1::1.49191 > fd97:739d:386a:1:880e:4dc4:644b:5151.46893: UDP, length 502
01:19:26.030837 IP6 fd97:739d:386a:1:880e:4dc4:644b:5151.46893 > fd97:739d:386a:1::1.49191: UDP, length 696
01:19:26.849335 IP6 fd97:739d:386a:1::1.49191 > fd97:739d:386a:1:880e:4dc4:644b:5151.46893: UDP, length 318
01:19:27.088179 IP6 fd97:739d:386a:1:880e:4dc4:644b:5151.46893 > fd97:739d:386a:1::1.49191: UDP, length 138
01:19:27.135611 IP6 fd97:739d:386a:1::1.49191 > fd97:739d:386a:1:880e:4dc4:644b:5151.46893: UDP, length 148
01:19:27.161082 IP6 fd97:739d:386a:1:880e:4dc4:644b:5151.46893 > fd97:739d:386a:1::1.49191: UDP, length 128
01:19:27.196705 IP6 fd97:739d:386a:1::1.49191 > fd97:739d:386a:1:880e:4dc4:644b:5151.46893: UDP, length 92
01:19:27.210271 IP6 fd97:739d:386a:1:880e:4dc4:644b:5151.46893 > fd97:739d:386a:1::1.49191: UDP, length 102
01:19:29.662815 IP6 fd97:739d:386a:1:880e:4dc4:644b:5151.46893 > fd97:739d:386a:1::1.49191: UDP, length 58
01:19:29.664307 IP6 fd97:739d:386a:1::1.49191 > fd97:739d:386a:1:880e:4dc4:644b:5151.46893: UDP, length 33
01:19:29.665395 IP6 fd97:739d:386a:1::1.49191 > fd97:739d:386a:1:880e:4dc4:644b:5151.46893: UDP, length 45
```

```
> ~ Discovery Request from aec105f19f1c9c71: version=2,joiner=1
~ Discovery Request from 0269b24ce1fa51f8: version=2,joiner=1
```

wgtdkp, 2021-01-04:
@gabekassel The log shows that the message buffers are exhausted, could you do the following?
1. enlarge message buffers for `ot-daemon`: add below code to file `openthread-core-posix-config.h`
    ```c++
    /**
     * @def OPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS
     *
     * The number of message buffers in the buffer pool.
     *
     */
    #define OPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS 256
    ```
2. share you build configuration of `ot-daemon`.

gabekassel, 2021-01-04:
Building with that change now. Here are my current ot-daemon build flags

```
-GNinja -DOT_PLATFORM=posix \
    -DOT_DAEMON=1 \
    -DOT_DIAGNOSTIC=1 \
    -DOT_BORDER_ROUTER=1 \
    -DOT_BORDER_AGENT=1 \
    -DOT_COAP_OBSERVE=1 \
    -DOT_COMMISSIONER=1 \
    -DOT_CHANNEL_MANAGER=1 \
    -DOT_CHANNEL_MONITOR=1 \
    -DOT_CHILD_SUPERVISION=1 \
    -DOT_CTI=1 \
    -DOT_DATASET_UPDATER=1 \
    -DOT_DHCP6_CLIENT=1 \
    -DOT_DHCP6_SERVER=1 \
    -DOT_DNS_CLIENT=1 \
    -DOT_DYNAMIC_LOG_LEVEL=1 \
    -DOT_FULL_LOGS=1 \
    -DOT_IP6_FRAGM=1 \
    -DOT_JAM_DETECTION=1 \
    -DOT_JOINER=1 \
    -DOT_LEGACY=1 \
    -DOT_MAC_FILTER=1 \
    -DOT_MAX_POWER_TABLE=1 \
    -DOT_MTD_NETDIAG=1 \
    -DOT_SERVICE=1 \
    -DOT_SLAAC=1 \
    -DOT_SNTP_CLIENT=1 \
    -DOT_TMF_PROXY=1 
```

gabekassel, 2021-01-04:
Things look **much** butter with that change. Just commissioned five times without issue. 

Is this buffer change unique to my platform?

jwhui, 2021-01-04:
> Things look **much** butter with that change. Just commissioned five times without issue.
> 
> Is this buffer change unique to my platform?

@gabekassel , I don't think this is unique to your platform. The external commissioner increases message buffer utilization because it needs to maintain a DTLS session with the external commissioner and proxy messages between the commissioner, leader, and joiner. I just submitted #6030 to increase the default message buffer pool size for posix build.

gabekassel, 2021-01-04:
Pasting PR link for future us https://github.com/openthread/openthread/pull/6037

closed 2021-01-05 by jwhui

OlivierGre, 2022-08-18:
While doing an external commissioning (with an Android phone used as commissioner), I'm also seeing this error on the raspberry pi otbr agent:
`MESH-CoP------: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered`
I have 20 messages like this. 
The commissioning is successful though.

In my env, OPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS is already set to 256. Shall I increase it more? 

In my configuration, there is an openthread library (https://github.com/openthread/openthread) in the otbr-agent, in the RCP and in the Joiner Device. If I change OPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS, shall I rebuild all those devices or is it needed only for the otbr-agent?

> @gabekassel It looks like the BR and joiner device are not compatible with each other. Is it possible for you to use the same commit for both the BR and joiner?
@wgtdkp  In my case, the version used for the RCP and Joiner device is older than the version used in the OTBR agent. For my information, how do you see compatibility problems?

Thank you

wgtdkp, 2022-08-19:
@OlivierGre 
```
MESH-CoP------: Failed to notify commissioner on ProxyRx (c/ur): DestinationAddressFiltered
```
The log actually doesn't indicate an error, it's printed when the BR filters out messages which are not targeting the commissioner.

I think you are set if you can successfully commissioning new joiner devices.

PR #6037 (2021-01-05): [posix] increase message buffer pool to 256
Resolves #6030