openthread/openthread #6031: [devsite] seeming mistakes in Hardware Codelab

Irving-cl, 2021-01-02:
**Describe the bug**
1. `NCP` are still used in some places. [Codelab #Step4](https://openthread.io/codelabs/openthread-hardware#4) says, "**Set up the RCP Joiner**". And `ot-ctl` is used to manage the Thread device. However, the title of [#Step8](https://openthread.io/codelabs/openthread-hardware#7) is "Add the NCP Joiner". And there are many other occurrences of `NCP` and `wpantund`, `wpanctl`. It seems that there's no NCP in this codelab. I guess it's from the last version. Should these be updated? Other occurrences: 
    * [Step7](https://openthread.io/codelabs/openthread-hardware#6), in the code block to show `ot-ctl` for RCP Joiner, `wpanctl:utun7> scan`.
    * Since Step8, `NCP` and `wpanctl` command are still used in text, code blocks and topology graph.

Thoughts? @Vyrastas @jwhui

Vyrastas, 2021-01-05:
Yeah, I have a long-standing task to update this Codelab to remove the references to wpantund and wpanctl, but I haven't had time lately.  I'll try to take a look later this week but if someone else (a Googler) wants to update, the Codelabs are now on openthread.io so they can be edited like any other openthread.io source file.

Irving-cl, 2021-01-06:
@Vyrastas 

Ok, I think I can do it as I just played with it recently.

jwhui, 2021-01-12:
Codelab updates have been published.

closed 2021-01-12 by jwhui