openthread/openthread #6032: Getting Started: Developing with OpenThread APIs build errors

ricardocirio, 2021-01-03:
Hi,

I followed the getting started guide: [https://openthread.io/codelabs/openthread-apis](url) and got errors during build related to the use of the **otUdpBind** and **otUdpSend** functions without the required _otInstance_ parameter. I solved it by adding the parameter to the calls.

1. Git commit id: 22b043bf62328e89773fa61c9bb80770f81a4bef
2. IEEE 802.15.4 hardware platform: nrf52840
3. Build steps: as described in the guide.

The issue is in the code from step 7 (API: Use multicast to turn on an LED).

Irving-cl, 2021-01-03:
Hi @ricardocirio,

Thanks for raising the issue! The two APIs do need a `otInstance` parameter now. The document needs to be updated.

Irving-cl, 2021-01-20:
The Codelab has been updated.

closed 2021-01-20 by Irving-cl