openthread/openthread #6033: OT-Daemon/RCP Incompatibility

gabekassel, 2021-01-03:
**Describe the bug**
I observed that an RCP running from around commit c00cb8a4c4f6e2a619770ffbb4abfd0f36a7fdc2 is not compatible with host ot-daemon from current ToT (6368cee5e7fc46348f168c6330f454087ad0a2fb). Since, in my system, I use ot-daemon not only to manage the Thread network but also to manage the RCP firmware (added custom version and a bootload launching command via diagnostic commands), this creates a problem. 

Is there a suggestion for handling this situation? Might it be possible to launch ot-daemon with some "basic capability" that is always backwards compatible? My only thought was to install an additional copy of ot-daemon from a previous commit, but then it gets complicated to know which ot-daemon works with which RCP firmware. 

This is ot-daemon against EFR31MG1P RCP.

```
root@eero:~# /usr/sbin/ot-daemon -d5 spinel+hdlc+uart:///dev/ttyQHS0?ncp-dataset -v
/usr/sbin/ot-daemon[3974]: Running OPENTHREAD/ga898de065; POSIX; Jan 3 2021 19:04:28
/usr/sbin/ot-daemon[3974]: Thread version: 2
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: Wait response: tid=0 key=0
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => Framing error 6: [
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => 00 C8 00 6F C2 C0]
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => Framing error 6: [
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => 00 C8 00 6F C2 C0]
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => Framing error 6: [
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => 00 C8 00 6F C2 C0]
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => Framing error 6: [
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => 00 C8 00 6F C2 C0]
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => Framing error 6: [
/usr/sbin/ot-daemon[3974]: [WARN]-PLAT----: Error decoding hdlc frame: Parse
/usr/sbin/ot-daemon[3974]: [INFO]-PLAT----: RCP reset: RESET_SOFTWARE
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: Wait response: tid=1 key=1
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: Wait response: tid=2 key=2
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: Wait response: tid=3 key=8
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => Framing error 6: [
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => 00 C8 00 6F C2 C0]
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => Framing error 6: [
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => 00 C8 00 6F C2 C0]
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => Framing error 6: [
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => 00 C8 00 6F C2 C0]
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => Framing error 6: [
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => 00 C8 00 6F C2 C0]
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => Framing error 6: [
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => 00 C8 00 6F C2 C0]
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => Framing error 6: [
/usr/sbin/ot-daemon[3974]: [DEBG]-PLAT----: RCP => 00 C8 00 6F C2 C0]
/usr/sbin/ot-daemon[3974]: [CRIT]-PLAT----: HandleRcpTimeout() at /usr/src/debug/openthread-core/git+AUTOINC+a898de065f-r0/git/src/lib/spinel/radio_spinel_impl.hpp:2164: RadioSpinelNoResponse

```

Moandor-y, 2021-01-05:
Hi @gabekassel , I ran an OT daemon compiled with <https://github.com/openthread/openthread/commit/22b043bf62328e89773fa61c9bb80770f81a4bef> and it works fine with an nrf52840 RCP compiled with <https://github.com/openthread/openthread/commit/c00cb8a4c4f6e2a619770ffbb4abfd0f36a7fdc2>. Also tried running against a simulated RCP and didn't see the issue. Is this issue only happening with EFR32MG1P?

gabekassel, 2021-01-05:
I haven't tried with another RCP but can tomorrow. But the RCP api version did increment recently.

Moandor-y, 2021-01-05:
The RCP API version was changed in <https://github.com/openthread/openthread/pull/5999>, which is a backward-compatible change. If RCP incompatibility was introduced then `SPINEL_MIN_HOST_SUPPORTED_RCP_API_VERSION` must be increased. There should be some other reason.

abtink, 2021-01-05:
@gabekassel On a quick glance at the logs, I don't think this is related to RCP API version and RCP-host compatibility. The API version is tracking the spinel definitions (spinel msg format) and whether there is a incompatible change.

The error here seems to  indicate a more fundamental problem with the UART/driver interface between RCP and host:
- The error `"RCP => Framing error 6:"` is indicating a `PARSE_ERROR` during HDLC decoding of a received frame on RCP. 
-  Basically the received HDLC frame is somehow corrupted (i.e., does not have correct HDLC FCS).  
- Maybe some bytes are being dropped?

gabekassel, 2021-01-05:
Thanks @abtink - I just made two firmware trying to bisect and haven't been able to reproduce. I'm comfortable considering this transient for the moment and see if it happens again.

closed 2021-01-05 by gabekassel

gabekassel, 2021-01-05:
Moving to https://github.com/openthread/openthread/issues/6042