openthread/openthread #6053: Cannot ping external IPv6-addresses from a child device

MarcelloDavis, 2021-01-09:
Hello,

I have successfully set up a Thread mesh network using a Raspberry Pi 3B with a nRF52840 USB dongle servicing as the borderrouter, and a Thunderboard Sense 2 (EFR32MG12) as a member of the network.
Via `> router table` I can see both participants:
[image]
And I can also ping the borderrouter using the Thunderboard:
[image]
[image]

However, when I tried pinging an external IPv6 address, I cannot receive sent back bytes:
[image]
[image]

After saying `Done`, there is no further output in the console. Does that mean that I cannot access webpages from the Thunderboard via the borderrouter?

When I ping the same IP-address from the RPi borderrouter, i get an answer saying: "`64 bytes from edge-star-mini6-shv-01-dus.facebook.com (2a03:...)`"

Thank you very much in advance.
Marcel

jwhui, 2021-01-09:
To send packets to an external address, the Thread network must be configured with an external route (either a default route or more-specific route). In doing so, you also need to configure the Thread network with a globally routable IPv6 address.
```
sudo ot-ctl prefix add <ipv6-prefix> paros med
sudo ot-ctl netdata register
```
Without an external route, the Thread network will not know to route packets to a border router. Without a globally routable IPv6 address, the external network will not be able to route responses back to your Thread network.

MarcelloDavis, 2021-01-10:
@jwhui thank you very much! I'll try this out

MarcelloDavis, 2021-01-12:
@jwhui I started a new thread network and added the lines you provided like this:
```
sudo ot-ctl dataset init new
sudo ot-ctl dataset commit active
sudo ot-ctl prefix add fd11:22::/64 paros med
sudo ot-ctl ifconfig up
sudo ot-ctl thread start
sudo ot-ctl netdata register
```
And on my Thunderboard I typed:
```
dataset masterkey <masterkey>
dataset commit active
routerselectionjitter 1
ifconfig up
thread start
```

I can join the network with my Thunderboard, but that did not solve the problem that I cannot ping external addresses. I also tried `pasor med` instead of `paros med`, but that didn't help. Do you know what I'm doing wrong?

Best regards!

jwhui, 2021-01-12:
@MarcelloDavis , `fd11:22::/64` is a unique local address and not a globally routable prefix. If you want to receive packets from IPv6 hosts on the public internet, you need to have a globally routable IPv6 prefix.

maxikrie, 2021-01-24:
I think I ran into the same issue (it may however well be that I have a lack of understanding the overall concept - i am new to Openthread and not an export on networking). 

I have setup a border router on Raspberry Pi 4 following the instructions on https://openthread.io/guides/border-router/build (with automatic Wi-Fi AP setup) with RCP running on a nrf52840 DK. I have used the Web GUI with default settings to form a network and successfully commissioned a second nrf52840 DK as a FTD, which is now connected to the network (following https://openthread.io/guides/border-router/external-commissioning?comm=ot-commissioner). The RP4 is also connected over Ethernet to my home router, which provides Internet connection. Last, another PC is connected to the Access Point provided by the RP4 (BorderRouter-AP) and can access the Internet (whatever is required in terms of rerouting traffic hence works).

To give a better overview of my setup:

-  this is the output of ifconfig on the border router
[image]

- this the output of ipaddr on the RCP connected to the border router
[image]

- this is the output of ipaddr of the FTD
[image]

- this the output of ifconfig on the PC connected to the BorderRouter-AP)
[image]

I can ping, for example, the FTD from either RCP or border router (I am pinging the fd11:22:.. address as this is the mesh local address as specified during forming the network, if I understand this correctly)
[image]

I now would like to ping (and eventually send and receive UDP packets) from the PC connected to the BorderRouter-AP, but this is doesn't work. I think this is exactly what the OP tried to do. 

From the guide it seems quite straight-forward to also ping the Internet from the FTD with
> ping 64:ff9b::808:808

However, this also doesn't work (even though it works from the border router itself). Some research pointed me to tayga as NAT64 translator and the configuration in /etc/tayga.conf. I see that the prefix is set to 'prefix 64:ff9b::/96', but I am not sure what to make of this or if I should change this.

Any help is much appreciated, thanks!

maxikrie, 2021-01-24:
P.S. I rechecked and the fd11:22:xx address have now disappeared on both RCP and FTD for some reason. I guess this was not really the mesh local address, but rather the respective longer of the fdb4:42db:c94e:893c:xx addresses?

I can ping the longer address of the FTD from the RCP, but not from the border router
[image]

I can also send a UDP message from RCP to FTD.

maxikrie, 2021-01-27:
I could at least shed some light on the non-persistence of the 'on-mesh' prefix. I had rebooted the border router, which causes the 'on-mesh prefix' to disappear and needs to be re-added with 

> sudo ot-ctl prefix add fd11:22::/64 pasor
> sudo ot-ctl netdata register

as described in https://openthread.io/guides/border-router/external-commissioning?comm=ot-commissioner (see Manual). This issue describes the part about non-persistence after reboot https://github.com/openthread/ot-br-posix/issues/439#issue-625843068

I still cannot ping from another PC connected to the border router AP. 

@MarcelloDavis Did you get it working? 
@jwhui Can you maybe assist?

jwhui, 2021-01-27:
It appears that the PC only has a link-local IPv6 address. That is not sufficient if you want the PC to communicate with devices that are not on Wi-Fi/Ethernet (e.g. a Thread device).

One way to get things going is by manually configuring IPv6 addresses and routes on your Wi-Fi/Ethernet link. You can take a look at [Practical Networking for Linux Admins: IPv6 Routing](https://www.linux.com/topic/networking/practical-networking-linux-admins-ipv6-routing/) for information on how to do that.

jwhui, 2021-02-10:
Closing stale issue.

closed 2021-02-10 by jwhui