openthread/openthread #6058: Issues with COAP on Borderrouter

MarcelloDavis, 2021-01-11:
Hello,

I installed the OpenThread ot-rcp example on my Borderrouter (Raspberry Pi 3B with nordic nRF52840 Dongle) and an ot-cli-ftd example for my Thunderboard Sense 2 (EFR32MG12).

**For the Borderrouter, I essentially followed this tutorial:**

```
git clone https://github.com/openthread/ot-br-posix.git
cd ot-br-posix
./script/bootstrap
./script/setup
```
`OTBR_AGENT_OPTS="-I wpan0 spinel+hdlc+uart:///dev/ttyACM0" #I checked that this is the port of my dongle`
```
git clone https://github.com/openthread/openthread.git
cd openthread
./script/bootstrap
./bootstrap
make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 TMF_PROXY=1 DHCP6_SERVER=1 DHCP6_CLIENT=1 DISABLE_SPI=1 SNTP_CLIENT=1 DISABLE_BUILTIN_MBEDTLS=1 USB=1 BOOTLOADER=USB
cd output/nrf52840/bin
arm-none-eabi-objcopy -O ihex ot-rcp ot-rcp.hex
nrfutil pkg generate --hw-version 52 --sd-req 0x0 --application-version 1 --application ot-rcp.hex out.zip
sudo nrfutil dfu usb-serial -pkg out.zip -p /dev/ttyACM0 -b 115200
```

**For the Thunderboard I followed the instructions on Github** (https://github.com/openthread/openthread/tree/master/examples/platforms/efr32/efr32mg12):
```
cd <path-to-openthread>
./script/bootstrap
```
```
cd <path-to-openthread>/third_party
mkdir silabs
cd <path-to-Simplicity-Studio>/developer/sdks
cp -rf gecko_sdk_suite <path-to-openthread>/third_party/silabs/
```
```
cd <path-to-openthread>
./bootstrap
```
`make -f examples/Makefile-efr32mg12 BOARD=BRD4166A COMMISSIONER=1 JOINER=1 COAP=1 DHCP6_CLIENT=1 DHCP6_SERVER=1`
```
cd <path-to-openthread>/output/efr32mg12/bin
arm-none-eabi-objcopy -O ihex ot-cli-ftd ot-cli-ftd.hex
```

While I can see and use the `coap` commands on my Thunderboard, I cannot see or use them on the borderrouter. I have already tried a rebuild on the RPi, but that doesn't show any effect. Maybe there is an issue with the SDK on my borderrouter?

Thanks in advance, and best regards!

jwhui, 2021-01-11:
The RCP only includes the IEEE 802.15.4 MAC/PHY and the higher layers run on the host. If you want to include CoAP in otbr-agent, you can try adding the `-DOT_COAP=ON` build option to [otbr_install()](https://github.com/openthread/ot-br-posix/blob/master/script/_otbr#L56).

MarcelloDavis, 2021-01-12:
@jwhui oh, yes I totally forgot about that, sorry! And thank you very much for your quick response!
Can I alternatively build a ncp design just by changing rcp to ncp in this line:
`arm-none-eabi-objcopy -O ihex ot-rcp ot-rcp.hex`
so that I have the application layer running on the host processor?

Best regards

MarcelloDavis, 2021-01-12:
@jwhui thank you very much for your quick response! I tried adding the option right in the otbr_options:
```
    otbr_options=(
        "-DBUILD_TESTING=OFF"
        "-DCMAKE_INSTALL_PREFIX=/usr"
        "-DOTBR_DBUS=ON"
        "-DOTBR_WEB=ON"
        "-DOTBR_REST=ON"
        "-DOT_COAP=ON"
        "${otbr_options[@]}"
    )
```
But this led to an error so that I couldn't finish the ./script call.
Then I tried putting it in 
```
if [[ ${REFERENCE_DEVICE} == "1" ]]; then
        otbr_options+=(
            "-DOT_REFERENCE_DEVICE=ON"
            "-DOT_DHCP6_CLIENT=ON"
            "-DOT_DHCP6_SERVER=ON"
            "-DOT_COAP=ON"
```
And this time I could run the ./script call, but after complete setup I wasn't able to find the coap command when typing `ot-ctl help`.

Do you have any advice? And thank you very much in advance

jwhui, 2021-01-12:
I just applied this diff:
```
diff --git a/script/_otbr b/script/_otbr
index 98da066..95d438c 100644
--- a/script/_otbr
+++ b/script/_otbr
@@ -67,6 +67,7 @@ otbr_install()
         "-DOTBR_DBUS=ON"
         "-DOTBR_WEB=ON"
         "-DOTBR_REST=ON"
+        "-DOT_COAP=ON"
         "${otbr_options[@]}"
     )
```

And successfully built OTBR with the following commands:
```
git clone https://github.com/openthread/ot-br-posix.git
cd ot-br-posix
./script/bootstrap
./script/setup
```

The coap command appears as expected:
```
> help
bufferinfo
ccathreshold
channel
child
childip
childmax
childtimeout
coap
...
```

Can you describe what error you ran into?

jwhui, 2021-01-29:
Closing stale issue.

closed 2021-01-29 by jwhui