openthread/openthread #6063: RadioSpinelNoResponse serial errors on CC2538

VGuenat, 2021-01-12:
**Describe the bug**
After operating correctly for some time, the `otbr-agent` (from `openthread/ot-br-posix`) crashes due to a serial `RadioSpinelNoResponse` errors. It sometimes happens after a few minutes, then after a few seconds. I post the issue here as the problem seems to come from the `spinel` part of the code, not the `ot-br-posix` one.

**To Reproduce**

1. Git commit id: 01d169ad876583e7ff45307156dd01d1689748bb (submodule for `ot-br-posix` commit ID 07145f02b6ddafb5efb0b26442710fc491059259)
2. IEEE 802.15.4 hardware platform: custom hardware based on the CC2538. The serial connection is encapsulated in a USB connection via an FTDI chip. No HW flow control is present (CC2538's UART0 is used).
3. Build steps
    - for the module: `make -f examples/Makefile-cc2538`
    - for the OTBR, it is the standard build process from the guide [here](https://openthread.io/guides/border-router/build), but adapted to CentOS.
4. Network topology:
    - 10 nodes based on EFR32 platform running Zephyr + OpenThread MTD (custom HW there as well)
    - 1 node based on CC2538 platform running `ot-rcp`. That's the node to which the OTBR is connected.

**Console/log output** 
```
Nov  3 00:30:32 info otbr-agent[9678]:  [INFO]-MAC-----: Received IPv6 UDP msg, len:84, chksum:9cff, from:463f612f3e1bcd3a, sec:no, prio:net, rss:-60.0
Nov  3 00:30:32 info otbr-agent[9678]:  [INFO]-MAC-----:     src:[fe80:0:0:0:443f:612f:3e1b:cd3a]:19788
Nov  3 00:30:32 info otbr-agent[9678]:  [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
Nov  3 00:30:32 info otbr-agent[9678]:  [INFO]-MLE-----: Receive Parent Request (fe80:0:0:0:443f:612f:3e1b:cd3a)
Nov  3 00:30:32 info otbr-agent[9678]:  [INFO]-MLE-----: Delay Parent Response (fe80:0:0:0:443f:612f:3e1b:cd3a)
Nov  3 00:30:32 info otbr-agent[9678]:  [INFO]-MAC-----: Received IPv6 UDP msg, len:84, chksum:91f9, from:8e76752696786dd3, sec:no, prio:net, rss:-55.0
Nov  3 00:30:32 info otbr-agent[9678]:  [INFO]-MAC-----:     src:[fe80:0:0:0:8c76:7526:9678:6dd3]:19788
Nov  3 00:30:32 info otbr-agent[9678]:  [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
Nov  3 00:30:32 info otbr-agent[9678]:  [INFO]-MLE-----: Receive Parent Request (fe80:0:0:0:8c76:7526:9678:6dd3)
Nov  3 00:30:32 info otbr-agent[9678]:  [INFO]-MLE-----: Delay Parent Response (fe80:0:0:0:8c76:7526:9678:6dd3)
Nov  3 00:30:32 warning otbr-agent[9678]:  [WARN]-PLAT----: radio tx timeout
Nov  3 00:30:32 crit otbr-agent[9678]:  [CRIT]-PLAT----: HandleRcpTimeout() at /ot-br-posix/third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2164: RadioSpinelNoResponse
Nov  3 00:30:32 info avahi-daemon[3519]:  Withdrawing workstation service for wpan0.
Nov  3 00:30:32 notice systemd[1]:  otbr-agent.service: main process exited, code=exited, status=6/NOTCONFIGURED
Nov  3 00:30:32 notice systemd[1]:  Unit otbr-agent.service entered failed state.
Nov  3 00:30:32 warning systemd[1]:  otbr-agent.service failed.
Nov  3 00:30:37 info systemd[1]:  otbr-agent.service holdoff time over, scheduling restart.
Nov  3 00:30:37 info systemd[1]:  Stopped Border Router Agent.
Nov  3 00:30:37 info systemd[1]:  Started Border Router Agent.
Nov  3 00:30:37 info otbr-agent[9738]:  Running 0.2.0-07145f0-dirty
Nov  3 00:30:37 info otbr-agent[9738]:  Thread interface wpan0
```
There are also quite a lot of warnings, as shown below:
```
Nov  3 00:40:47 info otbr-agent[9994]:  [INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
Nov  3 00:40:47 info otbr-agent[9994]:  [INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:ce53, to:0xffff, sec:no, prio:net
Nov  3 00:40:47 info otbr-agent[9994]:  [INFO]-MAC-----:     src:[fe80:0:0:0:c0f5:f8d7:6f8b:e427]:19788
Nov  3 00:40:47 info otbr-agent[9994]:  [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
Nov  3 00:41:08 info otbr-agent[9994]:  [INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
Nov  3 00:41:08 info otbr-agent[9994]:  [INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:fecd, to:0xffff, sec:no, prio:net
Nov  3 00:41:08 info otbr-agent[9994]:  [INFO]-MAC-----:     src:[fe80:0:0:0:c0f5:f8d7:6f8b:e427]:19788
Nov  3 00:41:08 info otbr-agent[9994]:  [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
Nov  3 00:41:46 warning otbr-agent[9994]:  [WARN]-PLAT----: processTransmit: Drop
Nov  3 00:41:48 info otbr-agent[9994]:  [INFO]-MAC-----: Received IPv6 UDP msg, len:84, chksum:8203, from:269f23663169a84d, sec:no, prio:net, rss:-61.0
Nov  3 00:41:48 info otbr-agent[9994]:  [INFO]-MAC-----:     src:[fe80:0:0:0:249f:2366:3169:a84d]:19788
Nov  3 00:41:48 info otbr-agent[9994]:  [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
Nov  3 00:41:48 info otbr-agent[9994]:  [INFO]-MLE-----: Receive Parent Request (fe80:0:0:0:249f:2366:3169:a84d)
Nov  3 00:41:48 warning otbr-agent[9994]:  [WARN]-MLE-----: Failed to process Parent Request: NoBufs
Nov  3 00:41:49 info otbr-agent[9994]:  [INFO]-MAC-----: Received IPv6 UDP msg, len:84, chksum:9a3d, from:269f23663169a84d, sec:no, prio:net, rss:-61.0
Nov  3 00:41:49 info otbr-agent[9994]:  [INFO]-MAC-----:     src:[fe80:0:0:0:249f:2366:3169:a84d]:19788
Nov  3 00:41:49 info otbr-agent[9994]:  [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
Nov  3 00:41:49 info otbr-agent[9994]:  [INFO]-MLE-----: Receive Parent Request (fe80:0:0:0:249f:2366:3169:a84d)
Nov  3 00:41:49 warning otbr-agent[9994]:  [WARN]-MLE-----: Failed to process Parent Request: NoBufs
Nov  3 00:41:50 info otbr-agent[9994]:  [INFO]-MAC-----: Received IPv6 UDP msg, len:84, chksum:ea9e, from:269f23663169a84d, sec:no, prio:net, rss:-61.0
Nov  3 00:41:50 info otbr-agent[9994]:  [INFO]-MAC-----:     src:[fe80:0:0:0:249f:2366:3169:a84d]:19788
Nov  3 00:41:50 info otbr-agent[9994]:  [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
Nov  3 00:41:50 info otbr-agent[9994]:  [INFO]-MLE-----: Receive Parent Request (fe80:0:0:0:249f:2366:3169:a84d)
Nov  3 00:41:50 warning otbr-agent[9994]:  [WARN]-MLE-----: Failed to process Parent Request: NoBufs
Nov  3 00:41:50 info otbr-agent[9994]:  [INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
Nov  3 00:41:50 info otbr-agent[9994]:  [INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:6c2e, to:0xffff, sec:no, prio:net
Nov  3 00:41:50 info otbr-agent[9994]:  [INFO]-MAC-----:     src:[fe80:0:0:0:c0f5:f8d7:6f8b:e427]:19788
Nov  3 00:41:50 info otbr-agent[9994]:  [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
```

**Additional context**
This bug mostly appears during traffic bursts e.g. when all 10 nodes are sending their Child Update request. I noticed a similar issue was reported before in #5662 where HW flow control seemed to solve the problems.

I understand the fact of having a custom hardware platform makes it harder to get help here, but any insight into why this is happening and possible workarounds would be appreciated.

jwhui, 2021-01-12:
@VGuenat , thanks for raising this issue.

We recently merged #6037, which might help address the "NoBufs" errors on the host side.

You might also take a look at #6043, which might help address lost frames being sent from the RCP to host. Not sure if this will help with the "NoResponse" timeout, but worth a try.

VGuenat, 2021-01-13:
Thank you for the comment. I tried using the latest openthread commit in the otbr-agent but didn't see a difference.

I will swap the CC2538 module on Monday with one based on the EFR32 and see if it solves the problem, just to be sure on which side of the serial connection the problem lies.

jwhui, 2021-01-13:
@VGuenat , note that #6043 applies to the RCP. Did you test with the latest commit in both otbr-agent and RCP?

VGuenat, 2021-01-15:
@jwhui Indeed I forgot it the first time. I tried yesterday with both the otbr-agent and the RCP using the latest available commits and the issue persisted. As said before I will try on Monday with an EFR32 (both with and without HW flow control). I will also try with a different linux platform, to see if there is a link there.

I'll put the results here when I'm done.

VGuenat, 2021-01-19:
@jwhui I did some more testing yesterday, replacing the CC2538-based module by a EFR32MG21 Development Kit (BRD4180A). The results were good and no crashes were detected. Testing the CC2538-based module with the otbr-agent on my laptop confirmed the assumption that the error lies in the firmware and/or BSP.

It seems strange to have problems with the CC2538 since it is a Thread certified component. I assume the files in _examples/platforms/cc2538_ have been tested with some hardware. Do you know what kind of hardware was used there? Maybe the problem comes from my module's hardware design and looking at the reference design used in OpenThread may provide some insights.

On a side note, I see a lot of warnings saying **[WARN]-MLE-----: Failed to process Parent Request: NoBufs**. Since the latest commits put the default number of message buffers to 256, I don't understand how my setup could exhaust these (as a reminder, there are 10 MTD nodes, all children of the router, with a child timeout of 240 seconds and no application traffic). Do you have any ideas?

jwhui, 2021-01-27:
> Do you know what kind of hardware was used there? Maybe the problem comes from my module's hardware design and looking at the reference design used in OpenThread may provide some insights.

For CC2538 testing, we use the [Zolertia Firefly](https://zolertia.io/product/firefly/).

> On a side note, I see a lot of warnings saying **[WARN]-MLE-----: Failed to process Parent Request: NoBufs**. Since the latest commits put the default number of message buffers to 256, I don't [...]

That seems odd to me. Can you provide the output of `bufferinfo` to validate the number of available message buffers?

jwhui, 2021-02-10:
Closing stale issue.

closed 2021-02-10 by jwhui