openthread/openthread #6067: [csl] coordinate between CSL and data poll tx requests

abtink, 2021-01-13:
This is follow up from discussion in https://github.com/openthread/openthread/pull/6038#discussion_r556290446.

Basically the idea is to coordinate/manage between `DataPollSender` and CSL modules 
Example of this is when we need to send fast polls `DataPollSender::SendFastPolls()` or in general other APIs that control polling rate.
- If CSL sample period is same or shorter than fast poll interval, then we can skip polling.
- Otherwise we may need to send (fast) polls (to rx expected response).

I think the `DataPollSender` and/or CSL modules would be best place for this new code/optimization.

Thoughts?

Irving-cl, 2021-01-14:
SGTM.

So in general, the logic would be like: when we try to send data poll, we check that if CSL Sample would happen soon (we may need to decide how long is 'soon' in our case), then `DataPollSender` skip polling. And CSL Sampling goes as always. So basically we can do this by adding some code to `DataPollSender`.

abtink, 2021-01-14:
Yes. Thanks. This sounds good to me.

We wait about ~100ms after poll (for timeout) so I think around same interval may make sense for how "soon" before CSL sample to skip the data poll.  

Thoughts?

jwhui, 2021-01-14:
Do we want to make it configurable - at least make it a compile-time option? I could see some applications not caring as much about the latency. For example, some applications may be fine to wait a full second if the latency is acceptable.