openthread/openthread #6093: bootstrap script fails in docker environment container

deimi, 2021-01-19:
**Describe the bug**
`script/bootstrap` does not fully work in a docker build environment.

**To Reproduce** 

1. `docker run -it openthread/environment bash`
2. `cd openthread`
3. `script/bootstrap`

**Console/log output**
```
...
+ sudo ln -s /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-addr2line /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-ar /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-as /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-c++ /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-c++filt /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-cpp /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-elfedit /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++ /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc-9.3.1 /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc-ar /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc-nm /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc-ranlib /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcov /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcov-dump /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcov-tool /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gdb /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gdb-add-index /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gdb-add-index-py /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gdb-py /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gprof /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-ld /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-ld.bfd /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-nm /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-objcopy /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-objdump /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-ranlib /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-readelf /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-size /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-strings /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-strip /usr/local/bin/.
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-addr2line': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-ar': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-as': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-c++': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-c++filt': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-cpp': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-elfedit': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-g++': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gcc': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gcc-9.3.1': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gcc-ar': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gcc-nm': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gcc-ranlib': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gcov': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gcov-dump': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gcov-tool': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gdb': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gdb-add-index': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gdb-add-index-py': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gdb-py': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-gprof': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-ld': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-ld.bfd': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-nm': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-objcopy': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-objdump': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-ranlib': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-readelf': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-size': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-strings': File exists
ln: failed to create symbolic link '/usr/local/bin/./arm-none-eabi-strip': File exists
root@80946ca54a59:/openthread#
```

**Additional context**  
When doing manual builds, this is not a problem as the next steps for the actual building will work fine. But when using the docker image in an automated CI pipeline, the job will fail as the bootstrap script returns non zero.

Irving-cl, 2021-01-19:
@deimi 

The docker image has everything we need to run openthread. So we don't need to install anything in the docker. What CI pipeline do you mean? Is it possible to skip the step calling `script/bootstrap`?

deimi, 2021-01-19:
@Irving-cl 
We are using a gitlab pipeline in our project to build the nordic dongle firmware.

```
...
  script:
    - cd /openthread
    - ./script/bootstrap
    - ./bootstrap
    - make -f examples/Makefile-nrf52840 UDP_FORWARD=1 USB=1 BOOTLOADER=USB LINK_RAW=1
    - cd output/nrf52840/bin/
    - arm-none-eabi-objcopy -O ihex ot-ncp-ftd nrf52840-dongle-ncp-ftd.hex
...
```

I was just following the instructions as explained here https://openthread.io/guides/build.  
But you are right, I tried now and the build works fine if we skip the script/bootstrap part. So in my opinion this should either be "fixed" like in my pull request or the documentation should be more precise.

P.s. a few month ago as we set up our project pipeline, the script was working fine.

Irving-cl, 2021-01-19:
@deimi 

Sounds good to me. Thanks! I think we would also put update the guide later.

closed 2021-01-19 by jwhui