openthread/openthread #6098: Transmitting sensor data from thread network to external IPv6 networj

ToMiller, 2021-01-20:
GitHub Issues are for bugs and feature requests. To make bugs and feature requests more easy to find and organize, we close issues that are deemed out of scope for GitHub Issues.

The openthread-users Google Group is the recommended place for users to discuss OpenThread and interact directly with the OpenThread community. https://groups.google.com/forum/#!forum/openthread-users

maxikrie, 2021-01-24:
I am currently trying to achieve something similar and I have decided for DDS employing following open source implementation with support for constrained devices - https://github.com/eProsima/Fast-DDS. So far it looks promising, but I hit an obstacle with the thread - I can let you know more once this is resolved.

An alternative would be MQTT-SN, but I don't like that beyond the embedded devices MQTT is TCP based, which is too much overhead for the data I am interested in. Maybe somebody else with more experience can comment on MQTT(-SN) and ideally also DDS or CoAP.

maxikrie, 2021-01-25:
https://github.com/openthread/openthread/issues/6053#issue-782606606

jwhui, 2021-01-27:
> However I find it difficult to find the right approach to send the data to an external network. I don't need to send the data to the borderrouter and then from there send it to an IPv6 address, right?

Yes, given that Thread is an IPv6 network, a packet's destination can be any IPv6 host that is reachable from the Thread device.

> I found several approaches, like CoAP and MQTT, but I can't find no example for my EFR32, so I guess I have to create the script on my own. So, would you suggest going with CoAP or MQTT?

Both CoAP or MQTT are IP-based protocols. Either approach should work, but which is better will depend on your specific use case.

> And is there any good tutorial I can follow to solve my problem?

Unfortunately, we do not have any tutorials specific to CoAP or MQTT. However, I believe some silicon vendors supporting OpenThread have examples and tutorials as part of their SDKs.

jwhui, 2021-02-10:
Closing stale issue.

closed 2021-02-10 by jwhui