openthread/openthread #6108: Manual overrider LEADER Role to a specific Node when it reconnects to the Thread Network

uttaravadina, 2021-01-22:
**Is your feature request related to a problem? Please describe.** A feature that allows a Node to be assigned itself as the LEADER Role when it re-joins the network. It might be the case where a certain node in the Network holds more weightage because of being much more capable in terms of certain parameters.

*Parameters*:
   - Open physical space around the node
   - Resistance to interference because of physical positioning
   - Uninterrupted Power Source to the Node
   - High resistance to physical damage
   - Close proximity to the Border Router
   - Strong specs in terms of processing power, storage and radio.
 
**Describe the solution you'd like** Allowing the API call otThreadBecomeLeader, otThreadBecomeRouter, otThreadSetParentPriority, otThreadSetRouterDowngradeThreshold, otThreadSetRouterUpgradeThreshold and otThreadSetRouterSelectionJitter in the production to change settings.

**Describe alternatives you've considered** Tried the CLI `state leader` command.

**Additional context** Right now the above-listed APIs, when used in production to change settings, make it non-compliant with Thread SPECs which limits the above-mentioned use-case. An extra network data parameter should be added so that nodes can know if the non-compliant APIs can be manually overridden in the Network or not.

uttaravadina, 2021-01-22:
@jwhui it's completely understandable that changing such settings in production can cause the network to become unstable but there will be corner cases arising where such requirement ensures that the network reaches a certain state by overriding those APIs. Some thread networks might want to isolate themselves as the environment is relatively small and stable or certain devices by default need to be on top in hierarchical order because of the above-mentioned parameters.

Eg: A powerful (low constraint) node which is very close in proximity to a Border Router might want to stay Leader after it reconnects itself to the Network to allow a semi-static environment rather than a completely dynamic environment.

jwhui, 2021-01-27:
@uttaravadina , thanks for raising this issue.

Implementing this change would require a change to the Thread Specification. Currently, whether the partition is a singleton (compose of a single leader) takes precedence over a higher weight value. In other words, a partition with multiple routers will have higher priority over a partition with a single leader even though the latter has a higher weight value. I have submitted [SPEC-973](https://threadgroup.atlassian.net/browse/SPEC-973) for discussion within the Thread Group.

jwhui, 2021-11-06:
With #6809, the weight value now always takes highest precedence.

closed 2021-11-06 by jwhui