openthread/openthread #6115: [github-actions] test_mlr_multicast_routing_across_thread_pans.py fails intermittently

wgtdkp, 2021-01-26:
**Describe the bug**
tests/scripts/thread-cert/backbone/test_mlr_multicast_routing_across_thread_pans.py fails intermittently.

**Console/log output**
artifacts: [thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/5873362/thread-1-2-backbone-results.zip)

jwhui, 2021-05-10:
[logs_75340.zip](https://github.com/openthread/openthread/files/6453342/logs_75340.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/6453343/thread-1-2-backbone-results.zip)

jwhui, 2021-05-26:
[logs_79846.zip](https://github.com/openthread/openthread/files/6549809/logs_79846.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/6549812/thread-1-2-backbone-results.zip)

jwhui, 2021-06-30:
[logs_83684.zip](https://github.com/openthread/openthread/files/6743726/logs_83684.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/6743727/thread-1-2-backbone-results.zip)

jwhui, 2021-08-13:
[logs_91706.zip](https://github.com/openthread/openthread/files/6980073/logs_91706.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/6980077/thread-1-2-backbone-results.zip)

jwhui, 2021-09-29:
[logs_11862.zip](https://github.com/openthread/openthread/files/7254854/logs_11862.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/7254855/thread-1-2-backbone-results.zip)

jwhui, 2022-02-17:
[logs_114989.zip](https://github.com/openthread/openthread/files/8091599/logs_114989.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/8091600/thread-1-2-backbone-results.zip)

jwhui, 2022-03-18:
https://github.com/openthread/openthread/runs/5595883416?check_suite_focus=true

jwhui, 2022-03-25:
https://github.com/openthread/openthread/runs/5697022967?check_suite_focus=true

jwhui, 2025-03-11:
Stale issue.

closed 2025-03-11 by jwhui