openthread/openthread #6145: Unable to ping in 868Mhz band

SwathiRamachandra, 2021-02-05:
Hi,
I have two nodes leader and child. There is no output from ping command in leader device. The ping command is successful. But I could see from network analyzer of Silabs IDE, the ping response is sent from leader and child node has replied for the ping response, see below image. In leader device in IP6::HandleDatagram(), HasUnicastAddress() has not passed after ICMP6 reply from child.

Reproduction Steps:
Node A.
channel 0
panid 0xacdb
ifconfig up
thread start
state -> leader
commissioner start
commissioner joiner add <eui64Id> PSK001
ping <LLA>

Node B.
factoryreset
ifconfig up
joiner start PSK001
thread start
state -> child
ipaddr

>router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
 | 16 | 0x4000 |            63 |               0 |         0 |         0 |      0 | ae1fd572816050e9 |
 | 32 | 0x8000 |            63 |               0 |         3 |         3 |      0 | 321c723dfec3a1d7 |

1. Git commit id - 6fda0786edab8e821e1647cc4be08b4a7059cf4a
2. IEEE 802.15.4 hardware platform - EFR32MG12, BRD4163a Radio board
3. Build steps:
     make -f examples/Makefile-efr32mg12 COMMISSIONER=1 COAP=1 COAPS=1 DTLS=1 JOINER=1 DNS_CLIENT=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 BOARD=BRD4163A

what are the possibilities, where a ping command could fail?

Thanks,
Swathi

[image]

jwhui, 2021-02-05:
@SwathiRamachandra , the BRD4163A board is not a board that is currently included in our repo. I assume there is some custom platform code that you are running?

I noted that the Extended MAC address in the router table does not seem to match the MAC Src/Dest addresses in SiL Network Analyzer. In Thread, the link-local address should be derived from the Extended MAC address.

When processing the ICMPv6 Echo Reply, in `Ip6::HandleDatagram()`, can you indicate what the datagram's IPv6 Destination Address is? Also, can you provide the output of `ipaddr` on the Leader? That would allow us to check if there is a mismatch.

SwathiRamachandra, 2021-02-05:
@jwhui Sorry, the above image was from older Thread network configuration and the Ext MAC was created with another Thread config. Sorry for the confusion. 
Below I have created the Thread network again and the respective network analyzer image.

The datagram's destination address in ICMPv6 ECHO reply is fe80:0:0:0:5481:5727:1cd7:433b

ipaddr of Leader:
fdde:ad00:beef:0:0:ff:fe00:fc30
fdde:ad00:beef:0:0:ff:fe00:fc00
fdde:ad00:beef:0:0:ff:fe00:3000
fdde:ad00:beef:0:836d:6d50:d6ef:5a0a
fe80:0:0:0:3c2f:f3e3:c0ca:1b6c
Done

ipaddr of Child:
fdde:ad00:beef:0:0:ff:fe00:d400
fdde:ad00:beef:0:77a5:45ac:60e9:32d7
fe80:0:0:0:5481:5727:1cd7:433b
Done

router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
| 12 | 0x3000 |       63 |         0 |     0 |      0 |   0 | 3e2ff3e3c0ca1b6c |
| 53 | 0xd400 |       63 |         0 |     3 |      3 |  26 | 568157271cd7433b |

[image]

Thanks,
Swathi

jwhui, 2021-02-05:
Given that the ICMPv6 Echo Reply's MAC Dest is 3E2FF3E3C0CA1B6C, I would expect the IPv6 Destination Address to be fe80:0:0:0:3c2f:f3e3:c0ca:1b6c. However, you indicated that the IPv6 Destination Address is fe80:0:0:0:5481:5727:1cd7:433b, which is the address assigned to the peer. In that case, the ICMPv6 Echo Reply won't be processed correctly.

Thread uses 6LoWPAN header compression. For link-local addresses, the IPv6 address is derived from the MAC Header on reception. So I'm confused as to why the IPv6 Dest Address is not matching the MAC Dest address. I'm not aware of any issues like this currently in the code base.

SwathiRamachandra, 2021-02-10:
@jwhui Thanks for your help,
I am able to ping between the device in 868MHz band now, with the latest OpenThread commit.

Thanks,
Swathi

closed 2021-02-10 by SwathiRamachandra