openthread/openthread #6151: Implement (intentionally) unreliable multicast

TPolzer, 2021-02-07:
**Is your feature request related to a problem? Please describe.** 

What I want to achieve seems like it should be a relatively common issue:

I have a set of lights, each of them a full thread device, and a dimmer (a minimal thread device running on battery).
It would be nice to be able to smoothly regulate brightness with low latency for a group of lights.

That works pretty well (with <10bytes payload UDP packets sent at 25Hz) over thread with a single light, using unicast.

For a group I wanted to use multicast, but it turns out that multicast retransmissions block a lot of airtime, which in the scenario of continuous dimming is useless to the application, as it's sending updates invalidating earlier messages every ~40ms anyway.

**Describe the solution you'd like**

It would be great to have a per message header option to control retries.

**Describe alternatives you've considered**

For my application I see two alternatives:
- I can blanket disable multicast retries on all my Thread devices, but that of course then affects all multicast (including MLE?). Also obviously wouldn't work if I had devices by a different vendor on my network (which I guess isn't a big concern currently...).
- I can use a low update frequency and hide it by interpolating/extrapolating brightness changes. Works, but not really great either.

jwhui, 2021-02-10:
@TPolzer , thank you for submitting this issue.

Unfortunately, the Thread Specification does not provide the mechanism that you are proposing. Thread uses MPL [RFC 7731](https://tools.ietf.org/html/rfc7731), which includes a sequence number in the IPv6 Hop-by-Hop Option header for duplicate suppression and message ordering. MPL maintains a sliding window to determine which messages have been received. If a message falls behind the sliding window, MPL will drop the message and not attempt to forward it.

Note that only Thread routers perform MPL forwarding, so the number of active forwarders is limited by the fact that Thread limits active routers to at most 32. Also, Currently, MPL Timer Expirations is set to 2:
https://github.com/openthread/openthread/blob/f2bb6cfad3d537136ff6674defe042e6253dad8a/src/core/thread/mle_types.hpp#L200
Setting it to 1 may help, but it only reduces the traffic by half, at the cost of reduced reliability.

Another consideration is that Thread is an IPv6 transport. Having a Thread-specific mechanism for controlling multicast behavior does not help non-Thread devices.

TPolzer, 2021-02-10:
Thanks for the detailed response!

I'm aware this feature request doesn't fit well with the existing multicast design/infrastructure, i.e. short term I'll have to pick workarounds.

On the other hand from an end user perspective, this seems like a reasonable thing to ask for in a smart home setting and under normal conditions it doesn't seem impossible (from the 802.15.4 imposed constraints), so I'm wondering how much effort it would take to make it possible with Thread (assuming that non Thread devices don't matter - either don't exist or are connected with high speed/low latency, e.g. over Ethernet).

TPolzer, 2021-02-15:
Turns out my application level retransmits were more aggressive than intended, sorry.  
After fixing that, the behavior in a smallish network is now acceptable at 25 Hz multicast (don't have the hardware to test with a big network).

That said, it's still easily (visually) distinguishable from a wired dimmer. I guess the lowest hanging fruit now would be for the sleepy end device to be able to immediately send multicast messages to all routers in range (not just its parent).

jwhui, 2021-03-10:
> Turns out my application level retransmits were more aggressive than intended, sorry.
> After fixing that, the behavior in a smallish network is now acceptable at 25 Hz multicast (don't have the hardware to test with a big network).

Great to hear.

> That said, it's still easily (visually) distinguishable from a wired dimmer. I guess the lowest hanging fruit now would be for the sleepy end device to be able to immediately send multicast messages to all routers in range (not just its parent).

Implementing this would require the SED to maintain security associations with multiple parents and, in turn, increase the average power consumption on the SED.

TPolzer, 2021-03-10:
I'll close the thread here then. It'd be nice to have a faster multicast, but at this point I guess it's not worth the amount of work necessary.

closed 2021-03-10 by TPolzer