openthread/openthread #6152: How to add user commands to CLI?

purvisb, 2021-02-08:
I'm trying to figure out how to add my own commands to the OT CLI. I'm starting by adding a command that prints a string when called. I already achieved this two ways, but I'm really not sure if what I'm doing is proper.

One way I achieved this was by reverse engineering some stuff in the openthread source code and guessed and it somehow worked. But I feel like this isn't proper. Another way I achieved this was via the nrf sdk. Although I'm told I may want to pivot away from using this, and that I should stick to editing the openthread source code. I'm just not sure which approach is ideal.

I'll explain my two methods. You probably don't need to read all of this to answer my question. I'm just including it for posterity.

In openthread/src/cli/cli.cpp, I copied another random function and named it `Processtest`. And just added the line `OutputLine("Hello World");` and removed any other functionality from that function. Then in openthread/src/cli/cli.hpp, I added `otError Processtest(uint8_t aArgsLength, char *aArgs[]);` to the `#if OPENTHREAD_FTD` if statement. In cli.hpp, I then added `{"test", &Interpreter::Processtest},` AFTER the `#if OPENTHREAD_CONFIG_SRP_CLIENT_ENABLE || OPENTHREAD_CONFIG_SRP_SERVER_ENABLE` if statement.

The other way, was using the nrf sdk. In the path: nRF5_SDK_for_Thread_and_Zigbee_v4.1.0_32ce5f8 (1)/examples/thread/cli/ftd/usb/pca10056/blank/ses, I opened the thread_cli_ftd_usb_pca10056.emProject via segger. Then I went to the main file, added a function titled "test" with the line `otCliOutputFormat("hello world\n\r");` Then in the main function, I added the line `otCliCommand testCommand = {"test", test};` as the first line. Then I added 
 `otCliSetUserCommands(&testCommand,1);` AFTER the app_sched_execute(); line. I compiled this and it also worked.

jwhui, 2021-02-10:
@purvisb , it really depends on your project goals - both approaches work and are acceptable. Directly modifying the OpenThread CLI component does reduce dependencies on external projects. At the same time, implementing the CLI in the nRF SDK avoids having to modify the OpenThread component.

purvisb, 2021-02-11:
@jwhui , We are working with GPIO and SAADC drivers to connect sensors to nrf52840-mdk chips. Then we are going to send this data to a SQL database via COAP. 

Are we losing any functionality by using the nRF SDK? The OpenThread stack is pre-compiled so we can't modify that. But I believe everything in [this API](https://openthread.io/reference/api-updates) is accessible in the sdk (I need to confirm that). So losing access to the internals of OpenThread might not be important.  This might be a better question for their community.

Given our project, would you recommend using nRF SDK? Or should we stick with OpenThread exclusively? To me it mostly seems to be a difference in work flow.

jwhui, 2021-02-11:
@purvisb , I don't think you'll be missing any functionality if you are fine with pre-built binary and the existing OpenThread APIs. You would likely receive better support from Nordic if you used their nRF SDK constructs. That said, if you have any plan to switch away from nRF SDK, then you may want to avoid binding to it.

purvisb, 2021-02-12:
@jwhui Is there a simpler way to add a command using only the Open thread Source code? Presumably with otCliSetUserCommands?

Likewise if I wanted to import a driver to use in my CLI command, how would I go about doing that? I already tried some things that failed, which prompted me to try the nRF SDK.

jwhui, 2021-02-12:
> @jwhui Is there a simpler way to add a command using only the Open thread Source code? Presumably with otCliSetUserCommands?

Yes, you can use `otCliSetUserCommands()` to include your own command table.

> Likewise if I wanted to import a driver to use in my CLI command, how would I go about doing that? I already tried some things that failed, which prompted me to try the nRF SDK.
> 
> [Like this saadc driver for example](https://github.com/openthread/openthread/blob/4db030c6c9d7138e5e7f5859ecd549474e1fd717/third_party/NordicSemiconductor/nrfx/hal/nrf_saadc.h)

The OpenThread examples in this repo run in a bare-metal environment that does not provide many services that an RTOS may provide. If you are looking to integrate other peripheral drivers, it largely depends on the peripheral driver's dependencies, if any. The SDK will provide more services and support. I do not have any experience with the nRF SAADC peripiheral.

purvisb, 2021-02-21:
@jwhui In the nRF SDK - I got the command to work, but with a warning.

I initialize an otCliCommand
`otCliCommand readCommand = {"read", read};`

read is defined as
`static void read(void *aContext, uint8_t aArgsLength, char *aArgs[])`

Then I add the command
`otCliSetUserCommands(&readCommand,1);`

This compiles. But on the line where I initialize readCommand - I get this error:
`
initialization of ‘void (*)(uint8_t,  char **)’ {aka ‘void (*)(unsigned char,  char **)’} from incompatible pointer type ‘void (*)(void *, uint8_t,  char **)’ {aka ‘void (*)(void *, unsigned char,  char **)’} [-Wincompatible-pointer-types]`

What gives? My syntax seems to be the same as how its used [here](https://github.com/openthread/openthread/blob/3d21a846d92d6df4ae3559a184462537fa1413bd/src/posix/main.c) and [here](https://openthread.io/reference/struct/ot-cli-command).

wgtdkp, 2021-02-21:
@purvisb I believe the nRF SDK is not including the latest OpenThread definitions documented [here](https://openthread.io/reference/struct/ot-cli-command).

The OT CLI command callback signature has been changed in this PR https://github.com/openthread/openthread/pull/5587.

For the nRF SDK, I think you should define the `read` command as:
```c++
static void read(uint8_t aArgsLength, char *aArgs[])
```

purvisb, 2021-02-21:
That fixes the warning. I don't have any further questions on this thread at this time. Thank you!

closed 2021-02-21 by purvisb