openthread/openthread #6159: Outdated nrf_security library documentation for nRF52840 crypto cell 310 support

nealjack, 2021-02-09:
It appears the [documentation](https://github.com/openthread/openthread/tree/master/examples/platforms/nrf528xx/nrf52840#cryptocell-310-support) for cryptocell support with external applications is outdated. The library `libmbedcrypto_cc310_backend.a` no longer exists in `third_party/NordicSemiconductor/libraries/nrf_security/lib` among others.

The current contents of `third_party/NordicSemiconductor/libraries/nrf_security/lib` are:
```
libmbedcrypto_cc3xx.a   libmbedcrypto_shared.a     libmbedtls_tls_vanilla.a   libnrf_cc310_platform_0.9.4.a
libmbedcrypto_oberon.a  libmbedtls_base_vanilla.a  libmbedtls_x509_vanilla.a
```

How has the configuration and library includes changed for building external apps with cryptocell support? I tried including all the new libraries, sans the oberon one, and my builds are failing with many undefined mbedtls reference errors like the below:
```
libmbedtls_base_vanilla.a(ctr_drbg.c.obj): in function `mbedtls_ctr_drbg_random_with_add':
/home/map5/repo/ncs/mbedtls/library/ctr_drbg.c:564: undefined reference to `mbedtls_aes_crypt_ecb'
```

LuDuda, 2021-02-14:
@MarekPorwisz can you look at it?

MarekPorwisz, 2021-02-15:
It seems that somehow I forgot to update this part of doc, while I updated the[ RTOS part](https://github.com/openthread/openthread/tree/master/examples/platforms/nrf528xx/nrf52840#cryptocell-310-rtos-support).

For some reason, the order of the libraries does matter to some compilers. 
Here is an order of linking used in [nrf platform samples](https://github.com/openthread/openthread/blob/4538a5fb7cdf06507f61a41b99f1672b97b81803/examples/platforms/nrf528xx/nrf52840/Makefile.platform.am#L43) that was passing for all the compiles in the CI:
```
libmbedcrypto_shared.a
libmbedtls_tls_vanilla.a
libmbedtls_x509_vanilla.a
libmbedcrypto_cc3xx.a
libnrf_cc310_platform_0.9.4.a
libmbedcrypto_oberon.a
libmbedtls_base_vanilla.a
```
Different linker missing function errors were observed with different linking order.
Please let us know if changing the linking order resolved the issue.

Best regards,
/\/\arek

nealjack, 2021-02-15:
@MarekPorwisz thanks! With that library ordering, I am able to successfully build an external application.

However, my thread applications no longer function. I turned on openthread logging and am seeing a lot of failed attachments:

```
[0000048489] [INFO]-MLE-----: AttachState Idle -> Start
[0000048489] [NOTE]-MLE-----: Attach attempt 8 unsuccessful, will try again in 32.161 seconds
[0000050805] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000051347] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000052873] [DEBG]-MAC-----: Received frame from short address 0x0000
[0000052874] [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:28, seqnum:73, type:Beacon, src:0x0000,[0000055134] [DEBG]-MAC-----: Received frame from short address 0x0000
[0000055135] [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:28, seqnum:74, type:Beacon, src:0x0000,[0000055176] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000055183] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000055191] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000055197] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000055205] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000055213] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000055216] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000055223] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000055229] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000055237] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000055245] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000055417] [DEBG]-MAC-----: Received frame from short address 0x0000
[0000055417] [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:28, seqnum:75, type:Beacon, src:0x0000,[0000055699] [DEBG]-MAC-----: Received frame from short address 0x0000
[0000055700] [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:28, seqnum:76, type:Beacon, src:0x0000,[0000055981] [DEBG]-MAC-----: Received frame from short address 0x0000
[0000055981] [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:28, seqnum:77, type:Beacon, src:0x0000,[0000056146] [DEBG]-MAC-----: Request to start operation "EnergyScan"
[0000056146] [DEBG]-MAC-----: Starting operation "EnergyScan"
[0000056146] [DEBG]-UTIL----: ChannelMonitor: channel: 11, rssi:-98
[0000056146] [DEBG]-UTIL----: ChannelMonitor: channel: 15, rssi:-99
[0000056147] [DEBG]-UTIL----: ChannelMonitor: channel: 19, rssi:-99
[0000056147] [DEBG]-UTIL----: ChannelMonitor: channel: 23, rssi:-97
[0000056147] [DEBG]-MAC-----: [0000056425] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000056493] [DEBG]-MAC-----: Frame rx failed, error:DestinationAddressFiltered
[0000058049] [DEBG]==============================[RX len=069]===============================
[0000058050] [DEBG]| 41 D8 3D CE FA FF FF A2 | C3 50 2E 9A C4 6E BA 7F | AX=Nz.."CP..Dn:.
[0000058051] [DEBG]| 3B 01 F0 4D 4C 4D 4C 02 | 6A 00 15 44 58 00 00 FF | ;.pMLML.j..DX...
[0000058051] [DEBG]| FF FF FF 80 23 9C 6A 94 | AB 8F B6 B4 FB C0 1B E0 | ....#.j.+.64{@.`
[0000058052] [DEBG]| 67 B0 E9 0C FD 6C 2E 7B | C4 D6 FE 19 EF 09 58 9D | g0i.}l.{DV~.o.X.
[0000058054] [DEBG]-MLE-----: Receive UDP message
[0000058055] [WARN]-MLE-----: Failed to process UDP: Security
```

I think that last line there `Failed to process UDP: Security` might be indicative of the mbedtls/cc310 backends not working correctly. Are there any compilation flags or configuration that I might be missing? I'm building my libraries with the following make command:

```
make -f examples/Makefile-nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COAP=1 COMMISSIONER=1 DISABLE_BUILTIN_MBEDTLS=1 DNS_CLIENT=1 DIAGNOSTIC=1 EXTERNAL_HEAP=1 JOINER=1 LINK_RAW=1 MAC_FILTER=1 MTD_NETDIAG=1 SERVICE=1 UDP_FORWARD=1 ECDSA=1 SNTP_CLIENT=1 COAPS=1 DHCP6_SERVER=1 DHCP6_CLIENT=1 CHANNEL_MANAGER=1 CHANNEL_MONITOR=1 DISABLE_TRANSPORTS=1
```

I saw in some recent commits that the heap API has changed - perhaps it's a memory issue with mbedtls. I have set up my platform alloc and free like so:

```
void* otPlatCAlloc(size_t n, size_t size)
{
    void *p_ptr = NULL;

    p_ptr = nrf_calloc(n, size);

    return p_ptr;
}

void otPlatFree(void *p_ptr)
{
    nrf_free(p_ptr);
}

static void platform_init(void)
{
    APP_ERROR_CHECK(nrf_mem_init());

    int ret;

    ret = mbedtls_platform_set_calloc_free(otPlatCAlloc, otPlatFree);
    ASSERT(ret == 0);

    ret = mbedtls_platform_setup(NULL);
    ASSERT(ret == 0);
}
```

If anyone has suggestions for things to try, I would appreciate them!

MarekPorwisz, 2021-02-17:
First thing that came to my mind, but is only a guess without diving into the issue, is that recently there was and update that was removing the use of default masterkey. From that point you need to perform the commissioning or set masterkey manually (in addition to setting channel, and panid).

nealjack, 2021-02-17:
@MarekPorwisz That was it! Setting a master key worked and my device now joins the network.

PR #6181 (2021-02-18): [nrf52840] update docs to list new cc310/mbedtls libs
Fixes #6159
closed 2021-02-18 by jwhui