openthread/openthread #6164: Nordic 52840 DTLS JOIN failure

krbvroc1, 2021-02-11:
I have not been able to get Nordic support on this issue, maybe openthread can help and I believe Nordic monitors here.

- I first started using Nordic 'examples/thread/cli' which works fine joining a Thread network.
- Next I tried Nordic 'examples/thread/dfu/client' and I disabled Nordic special 'autocommission' (hardcoded masterkey) feature from their demo. This demo fails joining a Thread network. The DTLS ClientHello Handshake fails.

It appears the difference between these two is that the second example enables the 'nrf_crypto' security library.

In the first example, on a successful handshake, the DTLS packet looks as follows and Wireshark shows an 'Extension of Unknown type 256 - length 330'. What is this extension?

```
Datagram Transport Layer Security
    DTLSv1.2 Record Layer: Handshake Protocol: Client Hello
        Content Type: Handshake (22)
        Version: DTLS 1.2 (0xfefd)
        Epoch: 0
        Sequence Number: 0
        Length: 438
        Handshake Protocol: Client Hello
            Handshake Type: Client Hello (1)
            Length: 426
            Message Sequence: 0
            Fragment Offset: 0
            Fragment Length: 426
            Version: DTLS 1.2 (0xfefd)
            Random: 4bb461a7f5194340ca7a1806b121a8cfdb5c378ab9cbbf38…
                GMT Unix Time: Apr  1, 2010 05:04:39.000000000 EDT
                Random Bytes: f5194340ca7a1806b121a8cfdb5c378ab9cbbf38b111b0e8…
            Session ID Length: 0
            Cookie Length: 0
            Cipher Suites Length: 4
            Cipher Suites (2 suites)
                Cipher Suite: TLS_ECJPAKE_WITH_AES_128_CCM_8 (0xc0ff)
                Cipher Suite: TLS_EMPTY_RENEGOTIATION_INFO_SCSV (0x00ff)
            Compression Methods Length: 1
            Compression Methods (1 method)
                Compression Method: null (0)
            Extensions Length: 380
            Extension: signature_algorithms (len=12)
                Type: signature_algorithms (13)
                Length: 12
                Signature Hash Algorithms Length: 10
                Signature Hash Algorithms (5 algorithms)
                    Signature Algorithm: ecdsa_secp521r1_sha512 (0x0603)
                        Signature Hash Algorithm Hash: SHA512 (6)
                        Signature Hash Algorithm Signature: ECDSA (3)
                    Signature Algorithm: ecdsa_secp384r1_sha384 (0x0503)
                        Signature Hash Algorithm Hash: SHA384 (5)
                        Signature Hash Algorithm Signature: ECDSA (3)
                    Signature Algorithm: ecdsa_secp256r1_sha256 (0x0403)
                        Signature Hash Algorithm Hash: SHA256 (4)
                        Signature Hash Algorithm Signature: ECDSA (3)
                    Signature Algorithm: SHA224 ECDSA (0x0303)
                        Signature Hash Algorithm Hash: SHA224 (3)
                        Signature Hash Algorithm Signature: ECDSA (3)
                    Signature Algorithm: ecdsa_sha1 (0x0203)
                        Signature Hash Algorithm Hash: SHA1 (2)
                        Signature Hash Algorithm Signature: ECDSA (3)
            Extension: supported_groups (len=20)
                Type: supported_groups (10)
                Length: 20
                Supported Groups List Length: 18
                Supported Groups (9 groups)
                    Supported Group: secp521r1 (0x0019)
                    Supported Group: secp384r1 (0x0018)
                    Supported Group: secp256r1 (0x0017)
                    Supported Group: secp256k1 (0x0016)
                    Supported Group: brainpoolP256r1 (0x001a)
                    Supported Group: secp224r1 (0x0015)
                    Supported Group: secp224k1 (0x0014)
                    Supported Group: secp192r1 (0x0013)
                    Supported Group: secp192k1 (0x0012)
            Extension: ec_point_formats (len=2)
                Type: ec_point_formats (11)
                Length: 2
                EC point formats Length: 1
                Elliptic curves point formats (1)
                    EC point format: uncompressed (0)
            Extension: Unknown type 256 (len=330)
                Type: Unknown (256)
                Length: 330
                Data: 41045eec00943e591e16f0eeaec1cacc22dffe701dc2d9fe…
```

In the failing case, when nrf_crypto is used, that 'extension' is missing:

```
Datagram Transport Layer Security
    DTLSv1.2 Record Layer: Handshake Protocol: Client Hello
        Content Type: Handshake (22)
        Version: DTLS 1.2 (0xfefd)
        Epoch: 0
        Sequence Number: 0
        Length: 104
        Handshake Protocol: Client Hello
            Handshake Type: Client Hello (1)
            Length: 92
            Message Sequence: 0
            Fragment Offset: 0
            Fragment Length: 92
            Version: DTLS 1.2 (0xfefd)
            Random: 20b689ede8f4ad3e54292b2610550b03803057681db200d3…
                GMT Unix Time: May 24, 1987 01:49:33.000000000 EDT
                Random Bytes: e8f4ad3e54292b2610550b03803057681db200d30aff1262…
            Session ID Length: 0
            Cookie Length: 0
            Cipher Suites Length: 4
            Cipher Suites (2 suites)
                Cipher Suite: TLS_ECJPAKE_WITH_AES_128_CCM_8 (0xc0ff)
                Cipher Suite: TLS_EMPTY_RENEGOTIATION_INFO_SCSV (0x00ff)
            Compression Methods Length: 1
            Compression Methods (1 method)
                Compression Method: null (0)
            Extensions Length: 46
            Extension: signature_algorithms (len=12)
                Type: signature_algorithms (13)
                Length: 12
                Signature Hash Algorithms Length: 10
                Signature Hash Algorithms (5 algorithms)
                    Signature Algorithm: ecdsa_secp521r1_sha512 (0x0603)
                        Signature Hash Algorithm Hash: SHA512 (6)
                        Signature Hash Algorithm Signature: ECDSA (3)
                    Signature Algorithm: ecdsa_secp384r1_sha384 (0x0503)
                        Signature Hash Algorithm Hash: SHA384 (5)
                        Signature Hash Algorithm Signature: ECDSA (3)
                    Signature Algorithm: ecdsa_secp256r1_sha256 (0x0403)
                        Signature Hash Algorithm Hash: SHA256 (4)
                        Signature Hash Algorithm Signature: ECDSA (3)
                    Signature Algorithm: SHA224 ECDSA (0x0303)
                        Signature Hash Algorithm Hash: SHA224 (3)
                        Signature Hash Algorithm Signature: ECDSA (3)
                    Signature Algorithm: ecdsa_sha1 (0x0203)
                        Signature Hash Algorithm Hash: SHA1 (2)
                        Signature Hash Algorithm Signature: ECDSA (3)
            Extension: supported_groups (len=20)
                Type: supported_groups (10)
                Length: 20
                Supported Groups List Length: 18
                Supported Groups (9 groups)
                    Supported Group: secp521r1 (0x0019)
                    Supported Group: secp384r1 (0x0018)
                    Supported Group: secp256r1 (0x0017)
                    Supported Group: secp256k1 (0x0016)
                    Supported Group: brainpoolP256r1 (0x001a)
                    Supported Group: secp224r1 (0x0015)
                    Supported Group: secp224k1 (0x0014)
                    Supported Group: secp192r1 (0x0013)
                    Supported Group: secp192k1 (0x0012)
            Extension: ec_point_formats (len=2)
                Type: ec_point_formats (11)
                Length: 2
                EC point formats Length: 1
                Elliptic curves point formats (1)
                    EC point format: uncompressed (0)
```

cc @LuDuda

jwhui, 2021-02-11:
> In the first example, on a successful handshake, the DTLS packet looks as follows and Wireshark shows an 'Extension of Unknown type 256 - length 330'. What is this extension?

It is the `ecjpake_key_kp_pair` extension. You can find more details in [Thread 1.1.1](https://www.threadgroup.org/support#specifications) Section 7.4.7.3.

@LuDuda , can you help with the Nordic SDK example?

LuDuda, 2021-02-11:
Hello @krbvroc1, I am assuming this is the [Devzone ticket](https://devzone.nordicsemi.com/f/nordic-q-a/71115/thread-sdk-ota-bug) you created couple days ago? 

I can look at this problem and reply there. But first i would need to understand better what are you using on the Border Router side? Have you tried with the newest ot-br-posix? 

Both examples use nrf_security, the difference is that DFU client uses additional applicaition security (look at /libraries/crypto), so i would bet that there is some conflict in mbedTLS headers or background DFU module reconfigures something.

Note that nRF5 SDK for Thread and Zigbee is now in maintainance mode, and you should consider using nRF Connect SDK to get the newest OpenThread integration and other new features.

krbvroc1, 2021-02-11:
Hi @LuDuda - yes that is the Devzone ticket. Initially I was running the Nordic provided ot-br image on RaspPi. However, I have recently switched to running the latest ot-br-posix on RaspPi and that is what I am currently using. (Using 52840 dongle RCP rather than NCP on that RaspPi).

I have been conflicted about using nRF5 SDK for Thread for new development. But it seems from what I have seen on the Devzone is that Thread OTA DFU is not supported on nRF Connect SDK. So I am reluctant to switch if I cannot perform Thread OTA DFU. nRF Connect SDK seems less mature too.

There are big differences between the sdk_config.h between the two examples and the working one does not include any #defines or references to NRF_CRYPTO_*

jwhui, 2021-03-01:
@krbvroc1 , has this issue been resolved?

krbvroc1, 2021-03-01:
No, never got a response here or on the Devzone ticket. In parallel, I have been trying to rewrite my code to use the nRF Connect SDK which is based on Zephyr / Openthread.

pszkotak, 2021-03-02:
Hi @krbvroc1, we are working on this issue.

pszkotak, 2021-03-10:
@krbvroc1, thanks for giving us time to carefully look at it.
Please find result of our troubleshooting in the devzone ticket.
If there is further assistance needed lets continue there.

jwhui, 2021-03-10:
Discussion moved to [DevZone ticket](https://devzone.nordicsemi.com/f/nordic-q-a/71115/thread-sdk-ota-bug).

closed 2021-03-10 by jwhui