openthread/openthread #6168: [nrf52840] CSL transmission fails when using a dongle as CSL Transmitter and a DK as CSL Receiver

Irving-cl, 2021-02-13:
**Describe the bug**
For some reason, I'm using a nrf52840 dongle as CSL Transmitter and a nrf52840 DK as CSL Receiver. When I ping from the Transmitter to the Receiver, it fails. (The Receiver didn't receive it and there's no ack.) However it can work in the opposite way: If I ping from the Receiver to the Transmitter, it works. And the Receiver can receive the ping reply. I also tried using 2 dongles and 2 DKs. In both cases the ping succeeded.

Here is a pcap, the time of failed CSL transmission seems correct to me:
[image]

**To Reproduce** Information to reproduce the behavior, including:
1. Git commit id: [4538a5f](https://github.com/openthread/openthread/commit/4538a5fb7cdf06507f61a41b99f1672b97b81803)
2. IEEE 802.15.4 hardware platform: **nrf52840 DK(PCA10056 v1.1.0)**, **dongle (PCA10059 v1.0.0)**
3. Build steps
#### Dongle:
```
OT_CMAKE_BUILD_DIR=dg_build ./script/cmake-build nrf52840 USB_trans_bl -DOT_THREAD_VERSION=1.2 -DOT_CSL_RECEIVER=ON
arm-none-eabi-objcopy -O ihex dg_build/examples/apps/cli/ot-cli-ftd ot-cli-ftd.hex
nrfutil pkg generate --debug-mode --hw-version 52 --sd-req 0 --application ot-cli-ftd.hex --key-file private.pem ot-cli-ftd.zip
```
#### DK:
```
OT_CMAKE_BUILD_DIR=dk_build ./script/cmake-build nrf52840 UART_trans -DOT_THREAD_VERSION=1.2 -DOT_CSL_RECEIVER=ON
arm-none-eabi-objcopy -O ihex dg_build/examples/apps/cli/ot-cli-ftd ot-cli-ftd.hex
```

Any idea for the problem? @edmont @hubertmis 
Is it possible that this is caused by the different versions of hardware?

hubertmis, 2021-02-13:
@Irving-cl , what's CSL Period for attached pcap?

Irving-cl, 2021-02-13:
@hubertmis, it's 3125 (500ms).

hubertmis, 2021-02-13:
It seems dongle clock drifts from the sniffer clock back ~10 us per 1s (10 PPM, it's within tolerance). For the successful transmissions (no 23, 28), the transmitter is a little too late (a number of dozens of microseconds, within tolerance). However, the transmitters clock keeps drifting and if you wait long enough, it transmits the frame earlier than expected. Frames 30-33 are transmitted ~80 us too early, and frames 46-49 ~490 us too early.

I would verify if CSL receiver is prepared for too early transmissions. If receiver would be enabled just before expected transmission (without taking into account the transmitter drift), the result would be exactly what you observe.

Also, to verify timings of CSL receiver and transmitter, you can play with [https://github.com/openthread/openthread/blob/4538a5fb7cdf06507f61a41b99f1672b97b81803/third_party/NordicSemiconductor/drivers/radio/nrf_802154_debug.c#L54](debug gpio mode) in the radio driver and verify with a logic analyzer when receiver opens its window and when transmitter transmits the frame.

hubertmis, 2021-02-22:
@Irving-cl, was my comment of any help? Have you narrowed down the issue?

Irving-cl, 2021-02-23:
@hubertmis, Thanks very much. I'll try it soon. Sorry I was busy with something else.

Irving-cl, 2021-02-23:
@hubertmis, the problem only occurs when I use one dongle and one DK. How can we debug with the dongle?

hubertmis, 2021-02-23:
@Irving-cl, it depends what debugging do you mean. The easiest one would be to use TC2050 cable and SWD/JTAG debugger like described here: https://infocenter.nordicsemi.com/topic/ug_nrf52840_dongle/UG/nrf52840_Dongle/hw_swd_if.html?cp=4_0_5_5_6

If you are interested in GPIO debugging to verify timing, you would need to select available pins and solder connectors to them.

However, I think I've found the issue here:

https://github.com/openthread/openthread/blob/b2f7bda6b46a9d2f308ed0c777fd574c21f7b90d/examples/platforms/nrf528xx/src/radio.c#L1071

It should be
```
uint32_t diff = (cslPeriodInUs - (curTime % cslPeriodInUs) + (sCslSampleTime % cslPeriodInUs) + (kCslSampleWindow / 2)) % cslPeriodInUs;
```

With current implementation the transmitter aims at the beginning of CSL window, but it should aim to the center of it.

Irving-cl, 2021-02-23:
@hubertmis,

As I remember, we intentionally put it to the beginning of the CSL window according to the experience. I think at that time we haven't used the `transmit_at` solution. I would definitely try to put it to the center. I'll keep you updated.

For debugging, I'm planning to purchase one logic analyzer. I think it will be helpful to debug CSL in the future. Do you think a logic analyzer with sampling frequency of **100M** would be enough for our case? As I understand, it samples every 10ns.

hubertmis, 2021-02-23:
> Do you think a logic analyzer with sampling frequency of **100M** would be enough for our case?

It depends what M stands for. If that's MHz, it would be more than enough. With radio timing accuracy we are aiming now 1us. 2MHz logic is enough to measure such timings. 100MHz is 50 times better than enough :)

Irving-cl, 2021-02-23:
@hubertmis Yeah, it's MHz. It seems that even the cheapest ones have maximum frequency of 100MHz. So I'm curious about that.

hubertmis, 2021-02-23:
It's possible that the device can keep up 100MHz only when you are sampling one channel. For two channels it can degrade to 50MHz, 4 channels 25 MHz, 8 channels 12.5 MHz etc.

I have never used more than 8 channels for CSL tests. Even with such limitation, the cheapest one seems fine (as long as it supports enough number of channels).

Irving-cl, 2021-02-24:
@hubertmis, I just did more experiments. I did the same thing (2 device, a child attach to the parent and the parent ping the child, CSL period is **3125, 500ms**) for three cases:
* Use 2 dongles
* Use 2 DKs
* Use one Dongle as parent and one DK as child

Just as before, the ping succeeded in the first two cases and failed in the last case. Here are the detailed results:
### dongle to dongle
[image]
Data processed:
| Packet No.  | packet time | packet len  | packet send time | phase       | next sample time |
| ----------- | ----------- | ----------- | ---------------- | ----------- | ---------------- |
| 20          | 12.836621 | 102 | 12.833357 | 1614 | **13.091597** |
| 22          | 13.093847 | 69 | **13.091639** | N/A | N/A |
| 23          | 13.094840 | 17 | 13.094296 | 3108 | **13.591576** |
| 25          | 20.093198 | 48 | **20.091662** | N/A | N/A |
### DK to DK
[image]
Data processed:
| Packet No.  | packet time | packet len  | packet send time | phase       | next sample time |
| ----------- | ----------- | ----------- | ---------------- | ----------- | ---------------- |
| 19            | 8.198137 | 102 | 8.194873 | 3034 | **8.680313** |
| 22            | 8.682569 | 69 | **8.680361** | N/A | N/A |
| 23            | 8.683562 | 17 | 8.683018 | 3109 | **9.180458** |
| 26            | 15.182319 | 48 | **15.180783** | N/A | N/A |
### dongle to DK
[image]
| Packet No.  | packet time | packet len  | packet send time | phase       | next sample time |
| ----------- | ----------- | ----------- | ---------------- | ----------- | ---------------- |
| 17            | 26.768816 | 102 | 26.765552 | 1644 | **27.028592** |
| 19            | 27.030817 | 69 | **27.028609** | N/A | N/A |
| 20            | 27.031810 | 17 | 27.031266 | 3109 | **27.528706** |
| 23            | 38.030177 | 48 | **38.028641** | N/A | N/A |

If we take a look at the values I bolded, we can see that, for each successful CSL transmission, the `next sample time` is slightly smaller than the `packet send time` of next packet (in one period). And for the only one failed CSL transmission, this is not the case (`xx.x28641` is earlier than `xx.x28706` ). This explains why it doesn't receive the frame. But I don't understand why this happens when we use one dongle and one DK. We can reproduce it each time when we use one dongle and one DK.

hubertmis, 2021-02-24:
> But I don't understand why this happens when we use one dongle and one DK.

It's most probably caused by different properties of crystals used in dongle and DK. I expect that dongle crystal drifts slightly back in your experiments, while DK crystal drifts slightly forward. If both devices in the tests have similar drift, they can communicate with each other (it explain why dongle-dongle and DK-DK works).

However, if transmitter's clock drifts away from the receiver clock, the communication is harder. The receiver must take into account relative drift and extend RX window to allow transmitter fit inside. Because of this, receiver shall keep RX window open some safety margin after time reported in CSL Phase (in case transmitter drifts forward), and before time reported in CSL Phase (in case transmitter drifts back).

It seems that CSL implementation in OpenThread does not add safety margin before the time reported in CSL Phase, what I mentioned in one of previous posts. When you run tests transmitting from dongle to DK you can reproduce "drift back" condition, that is not correctly handled in the code, and you catch the problem.

edmont, 2021-03-16:
It seems @hubertmis proposal of centering the receive window is working well for me: https://github.com/openthread/openthread/pull/6287

hubertmis, 2021-03-16:
@Irving-cl, could you try PR #6287 on your setup to verify if it fixes the problem you reported?

Irving-cl, 2021-03-16:
@edmont @hubertmis 

Thanks solving the issue! I'll verify it ASAP.

Irving-cl, 2021-03-17:
@edmont @hubertmis 

I tried the latest main branch (commit 12d5f793e) but it still cannot work (using one dk and one dongle).

Here is how I build the `ot-cli-ftd` for dongle and dk:
```
# build DK
$ OT_CMAKE_BUILD_DIR=dk_build ./script/cmake-build nrf52840 UART_trans -DOT_THREAD_VERSION=1.2 -DOT_CSL_RECEIVER=ON -DOT_JOINER=ON
$ arm-none-eabi-objcopy -O ihex dk_build/examples/apps/cli/ot-cli-ftd ot-cli-ftd.hex

# build dongle
$ OT_CMAKE_BUILD_DIR=dg_build ./script/cmake-build nrf52840 USB_trans_bl -DOT_THREAD_VERSION=1.2 -DOT_CSL_RECEIVER=ON
$ arm-none-eabi-objcopy -O ihex dg_build/examples/apps/cli/ot-cli-ftd ot-cli-ftd.hex
$ nrfutil pkg generate --debug-mode --hw-version 52 --sd-req 0 --application ot-cli-ftd.hex --key-file private.pem ot-cli-ftd.zip
```

Then I start the DK as leader:
```
> factoryreset
> channel 14
Done
> panid 0xface
Done
> masterkey ff00112233445566778899aabbccddee
Done
> mode rdn
Done
> ifconfig up
Done
> thread start
Done
```
And start the dongle as SSED:
```
> factoryreset
> channel 14
Done
> panid 0xface
Done
> masterkey ff00112233445566778899aabbccddee
Done
> mode -
Done
> csl period 3125
Done
> ifconfig up
Done
> thread start
Done
```

And when I ping from DK (leader) to dongle (SSED), still, there is no reply. Could you please take a look?

edmont, 2021-03-18:
@Irving-cl could you please give a try to this? https://github.com/openthread/openthread/pull/6311

Irving-cl, 2021-03-18:
@edmont sure, will have a try soon.

Irving-cl, 2021-03-22:
@edmont 

I just tried with the latest commit 97d30e31. It seems that it still doesn't work. And it even cannot ping successfully when I use 2 DKs.

In addition, I tried the same thing with the main branch(without this PR). CSL transmission still doesn't work. It seems that it's affected by #6287. CSL transmission works after I reverted the change of #6287.

Does CSL transmission work on current main branch on your side?

edmont, 2021-03-22:
Sorry, I keep adding changes to the adaptative receive window PR. https://github.com/openthread/openthread/pull/6287 is actually reverted by https://github.com/openthread/openthread/pull/6311. The latest should be used for both Leader and SSED.

I just tested with the latest commit (https://github.com/openthread/openthread/pull/6311/commits/2ad609cfc173b5255408684c7daabc9cbe8a5634) and it worked well with the default parameters.

Then I switched to:
        otPlatRadioGetCslAccuracy() = 2
        kCslWorstCrystalPpm  = 2, ///< Worst possible crystal accuracy, in units of ± ppm.
        kCslWorstUncertainty = 1, ///< Worst possible scheduling uncertainty, in units of 100 us.

This worked for attaching, but not for pinging (which works for me with two DKs). This demonstrates how the @hubertmis' hypothesis seems true: DK and dongle use different crystals and their drift is more noticeable when small receive windows are used.

Irving-cl, 2021-03-30:
@edmont @hubertmis 

I just verified with the latest commit(a43e34824) for #6311. It works for both cases (DK-to-DK, DK-to-dongle). Thanks for the change! Let's close the issue after #6311 merged.

BTW if you have time, please help double check the DK-to-dongle case. I think your tests would be better.

edmont, 2021-03-31:
Hi @Irving-cl. Please find attached two capture files: [dk_leader-dg_ssed.zip](https://github.com/openthread/openthread/files/6236886/dk_leader-dg_ssed.zip)

I used commit https://github.com/openthread/openthread/commit/f81eebeb48fe72d0bfe3427a8cba0a17ef3f77d0 for building for the `default.pcapng` test, where you can see how the Leader (DK) is able to transmit via CSL even after 36 seconds (frame #27). 

For the `5ppm.pcapng` test I made these changes:
```
diff --git a/src/core/mac/sub_mac.cpp b/src/core/mac/sub_mac.cpp
index 40fd47cc4..cfad16ed3 100644
--- a/src/core/mac/sub_mac.cpp
+++ b/src/core/mac/sub_mac.cpp
@@ -1050,8 +1050,8 @@ void SubMac::GetCslWindowEdges(uint32_t &ahead, uint32_t &after)
         elapsed = static_cast<uint32_t>(curTime - mCslLastSync.GetValue());
     }
 
-    semiWindow = elapsed * (Get<Radio>().GetCslAccuracy() + mCslParentDrift) / 1000000;
-    semiWindow += mCslParentUncert * kUsPerUncertUnit;
+    semiWindow = elapsed * (5 + 5) / 1000000;
+    semiWindow += 1 * kUsPerUncertUnit;
 
     ahead = (semiWindow + kCslReceiveTimeAhead > semiPeriod) ? semiPeriod : semiWindow + kCslReceiveTimeAhead;
     after = (semiWindow + kMinCslWindow > semiPeriod) ? semiPeriod : semiWindow + kMinCslWindow;
```
In this case CSL works for delivering the Child Update Response, but not 15 s after.

This means the combined drift between the two devices is bigger than 10 ppm.

Irving-cl, 2021-04-02:
@edmont 

Thanks for the testing! Since it works, we can close this issue now.

closed 2021-04-02 by Irving-cl