openthread/openthread #6170: CoAP doesn't work when using out-of-band commissioning

purvisb, 2021-02-15:
**Describe the bug** 
CoAP times out in a thread partition made via out-of-band commissioning. CoAP works as expected in a thread partition made via in-band commissioning.

**To Reproduce** 
compilation command:
`make -f examples/Makefile-nrf52840 USB=1 JOINER=1 COMMISSIONER=1 COAP=1`

Use this script on two nrf52840-mdk SOCs. They automatically join each other. Note that $1 is the name of the symlink file in /dev.
```
#!/bin/bash
sudo -v

logPath="/root/openthread"
networkName="TestThread"
panid="0x1234"
xpanid="0123456789ABCDEF"
key="00112233445566778899aabbccddeeff"
terminalName="/dev/$1"
utunName=$1
channel=22

#create screen daemon
screen -dmS $1 /dev/$1 115200

#screen -S $1 -X stuff "state^M"

#configure
screen -S $1 -X stuff "dataset init new^M"
screen -S $1 -X stuff "dataset panid $panid^M"
screen -S $1 -X stuff "dataset extpanid $xpanid^M"
screen -S $1 -X stuff "dataset networkname $networkName^M"
screen -S $1 -X stuff "dataset masterkey $key^M"
screen -S $1 -X stuff "dataset channel $channel^M"
screen -S $1 -X stuff "prefix add fd01:2345:6789:0:1111:1111:1111:1113/64^M"

#start
screen -S $1 -X stu1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology
ff "dataset commit active^M"
screen -S $1 -X stuff "ifconfig up^M"
screen -S $1 -X stuff "thread start^M"
screen -S $1 -X stuff "netdata register^M"
screen -S $1 -X stuff "state router^M"

screen -S $1 -X stuff "dataset active^M"
```

**Expected behavior** 
`coap get` should receive a response. 

**Console/log output** 

**Additional context** 
The client keeps sending address solicit packets to the server. And IEE 802.15.4 ACKs are sent out (presumably in response). Theres no other exchange of information besides that.

When I use that script on a leader, then join another SOC to the thread partition via in-band commisioning (without running that script on it), CoAP works as expected.

jwhui, 2021-02-16:
I believe this is due to a mismatch in your Active Operational Dataset. Rather than setting each individual parameter separately, I recommend either (1) setting the entire Active Operational Dataset as hex-encoded data or (2) specifying only the master key and channel and letting the device retrieve the full dataset after attaching.

For (1), you can retrieve the Active Operational Dataset from the first node using:
```
dataset active -x
0e080000000000010000000300001035060004001fffe002084eb74ab03c56e6d00708fdc7fe165c83a67805108e2104f183e698da87e96efc1e45aa51030f4f70656e5468726561642d383631310102861104108d6273023d82c841eff0e68db86f35740c030000ff
Done
```
Then set it on another node using:
```
dataset set active 0e080000000000010000000300001035060004001fffe002084eb74ab03c56e6d00708fdc7fe165c83a67805108e2104f183e698da87e96efc1e45aa51030f4f70656e5468726561642d383631310102861104108d6273023d82c841eff0e68db86f35740c030000ff
Done
```

See [src/cli/README_DATASET.md](https://github.com/openthread/openthread/blob/master/src/cli/README_DATASET.md) for more details.

For (2): follow the Quick Start example [src/cli/README_DATASET.md](https://github.com/openthread/openthread/blob/master/src/cli/README_DATASET.md).

Hope that helps.

purvisb, 2021-02-17:
I will implement that later this week, I'll let you know if that works. Thank you.

purvisb, 2021-02-21:
@jwhui When I type `dataset active -x` on an SOC CLI - it just returns the human readable format.

wgtdkp, 2021-02-21:
@purvisb I believe you are using the nRF SDK but not building from the latest OpenThread code, right?

The `dataset active -x` command is introduced 5 months ago with this PR https://github.com/openthread/openthread/pull/5549. The equivalent command before is `dataset active binary`. You can check all available `dataset` sub-commands with `dataset help`.

purvisb, 2021-02-21:
@wgtdkp yes I am using the nRF SDK and I am not building from the latest OpenThread code. `dataset active binary` still returns the human readable format. Heres what `dataset help` outputs:
```
> dataset help
help
active
activetimestamp
channel
channelmask
clear
commit
delay
extpanid
init
masterkey
meshlocalprefix
mgmtgetcommand
mgmtsetcommand
networkname
panid
pending
pendingtimestamp
pskc
securitypolicy
Done
```
Any further suggestions? It might just be better for me to import the most recent version of OpenThread.

wgtdkp, 2021-02-22:
@purvisb Okay, it seems your OpenThread version is too old... One way is to update the OpenThread, while another way is to set the dataset fields one by one.

jwhui, 2021-03-01:
@purvisb , has this issue been resolved?

purvisb, 2021-03-01:
I'm working on updating my OpenThread binary [here](https://devzone.nordicsemi.com/f/nordic-q-a/72146/update-openthread-in-nrf-sdk).

This issue has not been resolved. But once I update the binary, the solution you provided will probably work. I'm fine with closing this issue if you want. I'll try and reopen it or make a new issue if I encounter an issue with your aforementioned solution.

jwhui, 2021-03-10:
@purvisb , thanks for the update. I'll close this for now. Feel free to reopen if you run into more issues.

closed 2021-03-10 by jwhui