openthread/openthread #6188: CoAP add resource and sending POST's

purvisb, 2021-02-21:
I'm trying to add a resource via my embedded program. heres my code
```
static void resource(void *aContext, uint8_t aArgsLength, char *aArgs[]){

   otCoapResource mResource;

   memset(&mResource, 0, sizeof(mResource));

   mResource.mUriPath = "test-resource";
   mResource.mContext = aContext;
   mResource.mHandler = coap_test_handler;
   mResource.mNext = NULL;

   otError error = otCoapStart(thread_ot_instance_get(), OT_DEFAULT_COAP_PORT);
   otCliOutputFormat(otThreadErrorToString(error));
    error = otCoapAddResource(thread_ot_instance_get(),&mResource);
    otCliOutputFormat(otThreadErrorToString(error));
   otCliOutputFormat("done\n\r");
```

Nothing ends in an error. But when I go into the CLI and type `coap resource` - it outputs (null). Likewise sending a GET to the "test-resource" resource fails.

Also I'm struggling to figure out how to send messages from a client to the server running a resource. I've hunted around for examples and tried them with no luck. Is there an example on sending a CoAP message to a server that I can refer to?

wgtdkp, 2021-02-21:
> Nothing ends in an error. But when I go into the CLI and type coap resource - it outputs (null).

The CLI command `coap resource` outputs the CLI CoAP resource which is set by command `coap resource <resource-url>`.

> Likewise sending a GET to the "test-resource" resource fails

Could you confirm if you can ping the CoAP server?

purvisb, 2021-02-22:
Figured this out. Issue was that the CoAP CLI and CoAP in my embedded program don't really talk to each other.

closed 2021-02-22 by purvisb