openthread/openthread #6201: SED battery drained quickly because of frequent re-attach sequences

dismirlian, 2021-02-24:
Under the following circumstances a SED's battery is drained very quickly:
- device with very marginal connectivity (de-attaches frequently)
- default `OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_MINIMUM_INTERVAL` (251ms)
- "fast" poll period (in our case, 11s)

This happens because the SED de-attaches frequently, and tries to attach very fast (the default `OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_MINIMUM_INTERVAL` causes 5 attach attempts in less than ~8s). A workaround may be to increase `OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_MINIMUM_INTERVAL`, but this would result in slow re-attaches for devices with robust links.

Another idea would be to implement some kind of "backoff": OT could guarantee that there is some minimum (configurable) interval _between_ re-attach sequences. What do you think?

Thank you.

jwhui, 2021-02-24:
@dismirlian , thanks for raising this issue.

As you probably know, OpenThread actually implements a backoff mechanism using the following configurations:

- [OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_ENABLE](https://github.com/openthread/openthread/blob/d24327fd71a96bda0619f6b6331184e313b73909/src/core/config/mle.h#L121)
- [OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_ENABLE](https://github.com/openthread/openthread/blob/d24327fd71a96bda0619f6b6331184e313b73909/src/core/config/mle.h#L135)
- [OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_MAXIMUM_INTERVAL](https://github.com/openthread/openthread/blob/d24327fd71a96bda0619f6b6331184e313b73909/src/core/config/mle.h#L149)
- [OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_JITTER_INTERVAL](https://github.com/openthread/openthread/blob/d24327fd71a96bda0619f6b6331184e313b73909/src/core/config/mle.h#L163)

Currently, the backoff interval should only be reset to the minimum after successfully attaching. Is this what you are observing? Or are you proposing to increase the backoff interval even after the device is successfully attaching?

dismirlian, 2021-02-24:
Hi @jwhui, thanks for the (incredibly) fast answer. Yes, I'm aware of those configurations. I'll try to explain better:

After a detach, the default `OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_*` values cause attach attempts at roughly these times (in ms):

0
251
753
1757
3765
7781
15813
31877
64005

As you see, there are 6 attach attempts in ~8s.

If the connectivity is marginal, one of these attempts may succeed. However, also because the connectivity is so marginal, the following SED poll may fail, so the device detaches after (3?) poll attempts. Immediately, the SED starts the re-attach sequence. So in practice we observe the following sequence:
- The SED polls and fails -> detaches
- The SED performs re-attach sequence, and succeeds after maybe 4 to 7 attempts (2~16s)
- The SED polls after 11s and fails -> detaches
- The SED performs re-attach sequence, and succeeds after maybe 4 to 7 attempts (2~16s)
........

If the poll period is relatively fast (say, 11s), then the time **between** those re-attach sequences is small, and the battery is drained very quickly.

In one particular situation, we observed 509 detaches and 510 (successful) attaches in 2hs (this means many more unsuccessful attach attempts!).

A quick fix may be to increase `OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_MINIMUM_INTERVAL` to something similar to our poll period, which would be ok regarding power consumption.

What I propose (just one possibility of many) is that OT guarantees a minimum time between those re-attach sequences. So, for example:

_For a SED with marginal link:_
- The SED polls and fails -> detaches
- The SED performs re-attach sequence, and succeeds after maybe 4 to 7 attempts (2~16s)
- The SED polls after 11s and fails -> detaches
- **OT waits (say) 19s before trying to re-attach, because the previous attach happened less than (say) 30s ago**
**In this case, OT would guarantee 30s between re-attach sequences.**
- The SED performs re-attach sequence, and succeeds after maybe 4 to 7 attempts (2~16s)

_For a SED with robust link:_
- The SED polls and fails -> detaches
- The SED performs re-attach sequence, and succeeds after maybe 4 to 7 attempts (2~16s)
- ... some hours pass ....
- The SED polls and fails -> detaches
- The SED performs re-attach sequence, and succeeds after maybe 4 to 7 attempts (2~16s)
**In this case, OT performs the re-attach sequence immediately**

This would allow SEDs with robust links to re-attach quickly, while preserving battery life for SEDs with marginal links.

Thanks!
Diego.

jwhui, 2021-02-24:
@dismirlian , thanks for clarifying!

Yes, I think applying some kind if hysteresis will certainly help in this situation. As you propose, there could be a single build-time configuration that specifies the minimum time delay between successive attaches and an implementation can always set to 0 to keep the behavior as it is today. Probably something that is easy to add in [`Mle::GetAttachStartDelay()`](https://github.com/openthread/openthread/blob/d24327fd71a96bda0619f6b6331184e313b73909/src/core/thread/mle.cpp#L558).

Would you like to submit a PR with this proposal?

abtink, 2021-02-24:
Thanks @dismirlian for raising this issue.

It's an interesting practical situation. 
- An SED node with such a poor link-quality is practically not usable (basically should be considered as not being part of the Thread network). 
- But I agree that it would be good to have mechanism to preserve its battery. Current reattach backoff mechasnim resets the to min wait time on a successful attach.

I can try to help with this and submit a PR on this later.

- We can add a feature that says don't try to attach again within some window of time after the last successful attach (say ~30s after last attach). This sounds reasonable but I am not sure if it may cause other side-effects (in cases/situation we cannot think of now).

- So maybe we need a bit more complex condition for delaying reattach: Something like, "delay if within some window of last attach and the last successful attach required more than x attempts".

Thoughts?

dismirlian, 2021-02-24:
@jwhui, I'd prefer that this be implemented by an expert (because it touches the very core), if it ever gets implemented.

@abtink, thanks for looking into this.

> * An SED node with such a poor link-quality is practically not usable (basically should be considered as not being part of the Thread network).

In practice we see that this SED is quite useable during some periods, and quite unuseable in others (we have yet to find out **why** this happens...). When it's attached, we transmit the messages we have cached, and that's ok in our use case.

> I can try to help with this and submit a PR on this later.

Excellent, thanks!!

> * We can add a feature that says don't try to attach again within some window of time after the last successful attach (say ~30s after last attach). This sounds reasonable but I am not sure if it [...]

Well, I can't imagine in which cases the first alternative would have undesirable side-effects, but the second alternative sounds even better to me.

jwhui, 2021-02-24:
> @jwhui, I'd prefer that this be implemented by an expert (because it touches the very core), if it ever gets implemented.

Sounds good. I've assigned to @abtink.

> Well, I can't imagine in which cases the first alternative would have undesirable side-effects, but the second alternative sounds even better to me.

I think it's all a tradeoff. At first glance, if we require X attempts, then the energy cost during poor connectivity is multiplied by X. It's not yet obvious to me whether that is desirable.

abtink, 2021-02-25:
Submitted PR https://github.com/openthread/openthread/pull/6202.

It's a different idea where we add a (build-time configurable) delay before resetting the current attach backoff interval upon a successful attach. This way if the device detaches before the delay wait time, we resume from the same attach backoff interval as before. Another way of thinking about this is that _if we detach too quickly then it is like as if we never attached in the first place._

dismirlian, 2021-02-25:
@abtink that was fast. I think it's a very good idea, and would be useful in our particular scenario.

Thank you very much for the support.

closed 2021-02-25 by jwhui