openthread/openthread #6211: Running out of message buffer after a long time

Moandor-y, 2021-02-26:
**Describe the bug**

Message buffer is full of `6lo send` messages after running for a long time in a simulated environment.

**To Reproduce**

Commit: 33f15fac2faad5aa31920a24cbbc8b2fc49dbe08

Topology: a leader and an SED

Change `OPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS` to 280. Create a cert test like this to simulate a long running scenario:
```
#!/usr/bin/env python3

import unittest

import thread_cert

LEADER = 1
CHILD = 2

class TempTest(thread_cert.TestCase):
    TOPOLOGY = {
        LEADER: {
            'mode': 'rdn',
            'panid': 0xface,
            'whitelist': [CHILD]
        },
        CHILD: {
            'mode': 'n',
            'panid': 0xface,
            'router_selection_jitter': 1,
            'whitelist': [LEADER]
        },
    }

    def test(self):
        self.nodes[LEADER].start()
        self.simulator.go(5)
        self.assertEqual(self.nodes[LEADER].get_state(), 'leader')

        self.nodes[CHILD].start()
        self.simulator.go(7)
        self.assertEqual(self.nodes[CHILD].get_state(), 'child')

        while True:
            self.nodes[CHILD].ping('ff03::1', size=1000)
            self.simulator.go(10000)

            self.nodes[LEADER].send_command('bufferinfo')
            print(self.nodes[LEADER]._expect_results([
                r'total: \d+',
                r'free: \d+',
                r'6lo send: \d+ \d+',
                r'6lo reas: \d+ \d+',
                r'ip6: \d+ \d+',
                r'mpl: \d+ \d+',
                r'mle: \d+ \d+',
                r'arp: \d+ \d+',
                r'coap: \d+ \d+',
                r'coap secure: \d+ \d+',
                r'application coap: \d+ \d+',
            ]))

if __name__ == '__main__':
    unittest.main()
```

Run it with `OT_NODE_TYPE=rcp THREAD_VERSION=1.1 script/test cert` and it will print the buffer info periodically.

**Expected behavior**

Messages are mostly free.

**Console/log output**

Console:
```
1: ifconfig up
1: thread start
1: state
2: ifconfig up
2: thread start
ActiveOperationalDataset is not implemented yet.
2: state
2: ping ff03::1 1000
1: bufferinfo
['total: 251', 'free: 198', '6lo send: 52 52', '6lo reas: 0 0', 'ip6: 0 0', 'mpl: 0 0', 'mle: 0 0', 'arp: 0 0', 'coap: 0 0', 'coap secure: 0 0', 'application coap: 0 0']
2: ping ff03::1 1000
1: bufferinfo
['total: 251', 'free: 144', '6lo send: 104 104', '6lo reas: 0 0', 'ip6: 0 0', 'mpl: 0 0', 'mle: 0 0', 'arp: 0 0', 'coap: 0 0', 'coap secure: 0 0', 'application coap: 0 0']
2: ping ff03::1 1000
1: bufferinfo
['total: 251', 'free: 91', '6lo send: 156 156', '6lo reas: 0 0', 'ip6: 0 0', 'mpl: 0 0', 'mle: 0 0', 'arp: 0 0', 'coap: 0 0', 'coap secure: 0 0', 'application coap: 0 0']
2: ping ff03::1 1000
1: bufferinfo
['total: 251', 'free: 37', '6lo send: 208 208', '6lo reas: 0 0', 'ip6: 0 0', 'mpl: 0 0', 'mle: 0 0', 'arp: 0 0', 'coap: 0 0', 'coap secure: 0 0', 'application coap: 0 0']
2: ping ff03::1 1000
1: bufferinfo
['total: 251', 'free: 2', '6lo send: 242 242', '6lo reas: 0 0', 'ip6: 0 0', 'mpl: 0 0', 'mle: 0 0', 'arp: 0 0', 'coap: 0 0', 'coap secure: 0 0', 'application coap: 0 0']
2: ping ff03::1 1000
1: bufferinfo
['total: 251', 'free: 1', '6lo send: 243 243', '6lo reas: 0 0', 'ip6: 0 0', 'mpl: 0 0', 'mle: 0 0', 'arp: 0 0', 'coap: 0 0', 'coap secure: 0 0', 'application coap: 0 0']
2: ping ff03::1 1000
1: bufferinfo
['total: 251', 'free: 2', '6lo send: 242 242', '6lo reas: 0 0', 'ip6: 0 0', 'mpl: 0 0', 'mle: 0 0', 'arp: 0 0', 'coap: 0 0', 'coap secure: 0 0', 'application coap: 0 0']
2: ping ff03::1 1000
1: bufferinfo
['total: 251', 'free: 2', '6lo send: 242 242', '6lo reas: 0 0', 'ip6: 0 0', 'mpl: 0 0', 'mle: 0 0', 'arp: 0 0', 'coap: 0 0', 'coap secure: 0 0', 'application coap: 0 0']
2: ping ff03::1 1000
1: bufferinfo
['total: 251', 'free: 1', '6lo send: 243 243', '6lo reas: 0 0', 'ip6: 0 0', 'mpl: 0 0', 'mle: 0 0', 'arp: 0 0', 'coap: 0 0', 'coap secure: 0 0', 'application coap: 0 0']
```

Logs:
```
ot-cli[1421501]: [INFO]-MAC-----: Sent IPv6 UDP msg, len:89, chksum:8244, to:166e0a0000000002, sec:no, prio:net
ot-cli[1421501]: [INFO]-MAC-----:     src:[fe80:0:0:0:146e:a00:0:1]:19788
ot-cli[1421501]: [INFO]-MAC-----:     dst:[fe80:0:0:0:146e:a00:0:2]:19788
ot-cli[1421501]: [INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
ot-cli[1421501]: [INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:218f, to:0xffff, sec:no, prio:net
ot-cli[1421501]: [INFO]-MAC-----:     src:[fe80:0:0:0:146e:a00:0:1]:19788
ot-cli[1421501]: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
ot-cli[1421501]: [INFO]-UTIL----: ChannelMonitor: 3423 [00 0a 16 24 2f 37 4d 5d 66 74 82 8d 97 a1 b3 bc ]
ot-cli[1421504]: [INFO]-ICMP----: Sent echo request: (seq = 15)
ot-cli[1421501]: [INFO]-MEM-----: No available message buffer
ot-cli[1421501]: [NOTE]-MAC-----: Dropping rx frag frame, error:NoBufs, len:95, src:0x3001, dst:0x3000, tag:63568, offset:0, dglen:1056, sec:yes
ot-cli[1421501]: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:63568, offset:120, dglen:1056, sec:yes
ot-cli[1421501]: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:63568, offset:216, dglen:1056, sec:yes
ot-cli[1421501]: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:63568, offset:312, dglen:1056, sec:yes
ot-cli[1421501]: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:63568, offset:408, dglen:1056, sec:yes
ot-cli[1421501]: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:63568, offset:504, dglen:1056, sec:yes
ot-cli[1421501]: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:63568, offset:600, dglen:1056, sec:yes
ot-cli[1421501]: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:63568, offset:696, dglen:1056, sec:yes
ot-cli[1421501]: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:63568, offset:792, dglen:1056, sec:yes
ot-cli[1421501]: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:63568, offset:888, dglen:1056, sec:yes
ot-cli[1421501]: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:72, src:0x3001, dst:0x3000, tag:63568, offset:984, dglen:1056, sec:yes
ot-cli[1421504]: [INFO]-MAC-----: Sent IPv6 HopOpts msg, len:1056, chksum:0000, to:0x3000, sec:yes, prio:normal
ot-cli[1421504]: [INFO]-MAC-----:     src:[fd00:db8:0:0:8e08:4fe6:d517:2f78]
ot-cli[1421504]: [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:1]
ot-cli[1421504]: [INFO]-UTIL----: ChannelMonitor: 3421 [00 09 13 24 2f 38 4a 59 61 70 7f 8e 96 a7 b3 bc ]
ot-cli[1421501]: [INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
ot-cli[1421501]: [INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:03eb, to:0xffff, sec:no, prio:net
ot-cli[1421501]: [INFO]-MAC-----:     src:[fe80:0:0:0:146e:a00:0:1]:19788
ot-cli[1421501]: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
ot-cli[1421501]: [INFO]-UTIL----: ChannelMonitor: 3424 [00 0a 16 24 2f 37 4d 5c 66 74 81 8d 97 a1 b3 bc ]
ot-cli[1421504]: [INFO]-UTIL----: ChannelMonitor: 3422 [00 09 13 24 2f 38 4a 59 61 70 7f 8d 96 a6 b3 bd ]
```

**Additional context**

This issue is likely to be introduced by <https://github.com/openthread/openthread/pull/5638>. I've run these steps with commit 9f2b6788362fbbe2534960da062af5af4a676076 and didn't see this issue.

simonlingoogle, 2021-02-26:
Great finding! 

Just FYI, currently we have another issue that an indirectly message of a SED Child (ICMPv6 Echo Request of payload size 1232) could be removed before all its fragments are sent, causing the SED unable to fully receive all Ping Request fragments. Not sure if it's related to #5638 though. 

@abtink @Moandor-y

abtink, 2021-02-26:
Thanks for raising the issue. I will investigate this.

jwhui, 2021-02-26:
@Moandor-y , thanks for raising this issue.

@abtink , I believe we are not dequeuing the supervision messages in this case and they are getting stuck in the queue.

abtink, 2021-02-26:
Yes. The problem is that the message was removed from the child, but would not be dequeued and freed.

Submitted #6123 which should help address thi.

closed 2021-02-26 by jwhui