openthread/openthread #6218: [efr32] efr32 example is broken

wgtdkp, 2021-03-01:
**Describe the bug** A clear and concise description of what the bug is.
cannot build the efr32 example app with:
```shell
./bootstrap
BOARD=BRD4151A make -f examples/Makefile-efr32mg1
```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: d8a2cd9a70cfda7c2657645b4a57411eadaa4e61
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.
```shell
Link libraries                            :  
  Link maps                                 : yes
  OpenThread FTD support                    : no
  OpenThread MTD support                    : no
  OpenThread Radio Only support             : yes
  OpenThread CLI support                    : no
  OpenThread NCP support                    : yes
  OpenThread NCP Vendor Hook Source         : no
  OpenThread NCP Spinel Encrypter           : no
  OpenThread Vendor Extension Source        : no
  OpenThread builtin mbedtls support        : no
  OpenThread Examples                       : efr32mg1
  OpenThread Platform Libraries             : efr32mg1

  BUILD    efr32mg1
make -j6 -C build/efr32mg1 --no-print-directory all
  CHECK         .local-version
  GEN           .local-version
make --no-print-directory all-recursive
Making all in include
make  all-am
Making all in third_party
Making all in silabs
make[4]: *** No rule to make target 'gecko_sdk_suite/v3.1/util/third_party/crypto/mbedtls/library/aes.c', needed by 'gecko_sdk_suite/v3.1/util/third_party/crypto/mbedtls/library/libsilabs_mbedtls_a-aes.o'.  Stop.
make[3]: *** [Makefile:488: all-recursive] Error 1
make[2]: *** [Makefile:593: all-recursive] Error 1
make[1]: *** [Makefile:519: all] Error 2
make: *** [examples/Makefile-efr32mg1:279: do-build-cortex-m4] Error 2
```

**Additional context** Add any other context about the problem here.

wgtdkp, 2021-03-01:
@lmnotran Could you help on this?

lmnotran, 2021-03-01:
Yeah, I'll take a look at this

lmnotran, 2021-03-01:
@wgtdkp Could you check to make sure that you have Gecko SDK v3.1.1?

```
$ grep "sdk_version" third_party/silabs/gecko_sdk_suite/v3.1/gecko_sdk.slcs
sdk_version: "3.1.1"
```

This is the latest patch release and the mbedtls installation was moved between v3.1.0 and v3.1.1 so that might explain the error. I wasn't able to reproduce the error using your details above.

romacdon, 2021-03-01:
@wgtdkp I checked the build failures in the associated CHIP PR (project-chip/connectedhomeip#5062) here: https://github.com/project-chip/connectedhomeip/pull/5062/checks?check_run_id=2003750768.  The failures in that build do not appear top be the same failure you included in the description in this issue.  I'm guessing you probably tried to recreate the CHIP build failures in your local OpenThread repo and when you saw a failure you thought maybe it was the same as the CHIP failures.  I believe the issue you reported here is different and can be solved by making sure you have the correct GSDK loaded at third_party/silabs/gecko_sdk_suite/v3.1.  That would be v3.1.1 as @lmnotran mentioned.  

Now with that said it still does not solve your CHIP build issues.  There I see these errors:

```
2021-03-01T12:30:38.4669961Z ../../../examples/lock-app/efr32/third_party/connectedhomeip/third_party/openthread/repo/examples/platforms/efr32/src/system.c:60:10: fatal error: sl_device_init_nvic.h: No such file or directory
```
and
```
2021-03-01T12:30:38.4894580Z ../../../examples/lock-app/efr32/third_party/connectedhomeip/third_party/openthread/repo/examples/platforms/efr32/src/uart.c:51:10: fatal error: sl_uartdrv_usart_vcom_config.h: No such file or directory
```
I'm not actively involved in the CHIP project but it appears that the efr32 CHIP build scripts may need to be modified to make sure the correct include paths are specified.  I know the build env for CHIP does not use the openthread core autoconf/automake tools.

lmnotran, 2021-03-01:
A good starting point for which files and directories need to be added to the CHIP build are here

- [third_party/silabs/Makefile.am](https://github.com/openthread/openthread/pull/6154/files#diff-2699900c349405f9143dba91c68a853d34d6536085eb0631e8759f6bbea2df90)
- [examples/platforms/efr32/efr32_platform_defs.am](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32/efr32_platform_defs.am#L68-L103)

wgtdkp, 2021-03-02:
@romacdon @lmnotran Thanks for your information! I didn't have the `third_party/silabs/gecko_sdk_suite` after pulling all submodules:
```shell
git submodule update --recursive --progress --init
```

Seems I need to download the SDK to there manually? It may be good if we can add a submodule for the SDK.

wgtdkp, 2021-03-02:
@romacdon @lmnotran I am closing this issue as the CI failure has been fixed within project-CHIP. But I think we should also have the gecko_sdk within `openthread` (or another repo `ot-efr32`? @jwhui ) so that we can make sure the efre32 example can always build.

closed 2021-03-02 by wgtdkp