openthread/openthread #6222: Commissioning support for product line nrf52811

oc1tane, 2021-03-02:
Hi I am trying to implement Openthread for a device on a Nrf52811 controller. I have been warned about the lack of 192 kb memory and the lack of a crttography module. But nevertheless, the protocol does not take up much space, even a little remains for its software. And the acceleration can be transferred to the program.

So what I was trying to do:
- Build Openthread firmware for nrf52811
- I use libmbedcrypto-software.a in the makefile due to the fact that nrf52811 does not support hardware acceleration, instead of libmbedcrypto.a
- in the main code main.c I use the inclusion of Openthread by the function from the otJoinerStart API
- from the last repository Openthread collected libraries for nrf52811 with the key **JOINER = 1**

Where did I get my knowledge:
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.thread_zigbee.v2.0.0%2Fthread_openthread_libraries.html&cp=4_2_0_1_0_8_0&anchor=openthread_libs

jwhui, 2021-03-02:
@oc1tane , thanks for reaching out. This concern is not just code size, but also available RAM. The steps below should get you a build of MTD with JOINER enabled:

1. Set [`OPENTHREAD_CONFIG_MESSAGE_USE_HEAP_ENABLE`](https://github.com/openthread/openthread/blob/8f7024c3e9beb47a48cfc1e3185f5fce82fffba9/src/core/config/openthread-core-default-config.h#L140) to 1.
2. `./bootstrap`
3. `make -f examples/Makefile-nrf52811 JOINER=1`

Hope that helps.

oc1tane, 2021-03-03:
Thanks for the quick reply, indeed I did so...
Compiled the libraries of the thread downloaded through the git and slipped them into the directory of the SDK Nordic
**./openthread/output/nrf52811/lib (folder openthread from github)** -> (to)  **./external/openthread/lib/nrf52811/gcc (folder SDK Nordic v.4.0)**

> /usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: <PATH_TO_NORDICSDK>/v4.0/external/openthread/lib/nrf52811/gcc/libopenthread-nrf52811-sdk.a(libopenthread_nrf52811_sdk_a-alarm.o): in function `nrf_rtc_event_pending':
<PATH_TO_OPENTHREAD>/openthread/examples/../third_party/NordicSemiconductor/nrfx/hal/nrf_rtc.h:352: multiple definition of `RTC1_IRQHandler'; _build/nrf52811_xxaa/drv_rtc.c.o:<PATH_TO_NORDICSDK>/v4.0/components/libraries/timer/drv_rtc.c:349: first defined here
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: _build/nrf52811_xxaa/main.c.o: in function `joiner_timer_handler':
<PATH_TO_OPENTHREAD>/Other/magnetic/pca10056/blank/armgcc/../../../main.c:370: undefined reference to `otJoinerStart'
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: _build/nrf52811_xxaa/thread_utils.c.o: in function `platform_init':
<PATH_TO_NORDICSDK>/v4.0/components/thread/utils/thread_utils.c:129: undefined reference to `otHeapSetCAllocFree'
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: <PATH_TO_NORDICSDK>/v4.0/external/openthread/lib/nrf52811/gcc/libopenthread-nrf52811-sdk.a(libopenthread_nrf52811_sdk_a-system.o): in function `otSysDeinit':
<PATH_TO_OPENTHREAD>/openthread/examples/../examples/platforms/nrf528xx/src/system.c:122: undefined reference to `nrf5TransportDeinit'
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: <PATH_TO_NORDICSDK>/v4.0/external/openthread/lib/nrf52811/gcc/libopenthread-nrf52811-sdk.a(libopenthread_nrf52811_sdk_a-system.o): in function `otSysInit':
<PATH_TO_OPENTHREAD>/openthread/examples/../examples/platforms/nrf528xx/src/system.c:101: undefined reference to `nrf5TransportInit'
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: <PATH_TO_NORDICSDK>/v4.0/external/openthread/lib/nrf52811/gcc/libopenthread-nrf52811-sdk.a(libopenthread_nrf52811_sdk_a-system.o): in function `otSysProcessDrivers':
<PATH_TO_OPENTHREAD>/openthread/examples/../examples/platforms/nrf528xx/src/system.c:142: undefined reference to `nrf5TransportProcess'
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: <PATH_TO_NORDICSDK>/v4.0/external/openthread/lib/nrf52811/gcc/libopenthread-nrf52811-sdk.a(libopenthread_nrf52811_sdk_a-flash_nosd.o): in function `nrf5FlashPageErase':
<PATH_TO_OPENTHREAD>/openthread/examples/../examples/platforms/nrf528xx/src/flash_nosd.c:36: undefined reference to `nrfx_nvmc_page_erase'
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: <PATH_TO_NORDICSDK>/v4.0/external/openthread/lib/nrf52811/gcc/libopenthread-nrf52811-sdk.a(libopenthread_nrf52811_sdk_a-flash_nosd.o): in function `nrf5FlashWrite':
<PATH_TO_OPENTHREAD>/openthread/examples/../examples/platforms/nrf528xx/src/flash_nosd.c:48: undefined reference to `nrfx_nvmc_bytes_write'
collect2: error: ld returned 1 exit status

jwhui, 2021-03-03:
@oc1tane , for questions specific to the Nordic SDK, I suggest posting in Nordic DevZone.

/cc/ @LuDuda

LuDuda, 2021-03-04:
Hello @oc1tane,

As Jonathan mentioned, please create a ticker on [Nordic's DevZone](https://devzone.nordicsemi.com/).

To give you some hints:
1. It seems you use nRF5 SDK for T&Z in version 4.0, but you pasted instruction to version 2.0. I suggest to migrate to follow steps for 4.0 version - [link](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.0.0/thread_ot_libraries_building.html).
2. nRF5 SDK in version 4.0 was based on specific OpenThread commit. It is possible to upgrade it to the newest version, however there might be a lot of warnings/errors that you would need to solve, since master progressed a lot from that moment. I would suggest to checkout OpenThread to this [commit](https://github.com/openthread/openthread/commit/8a1992e2e42fb707babe9491a6a1456e553490e8) as a first step.
3. The error log `undefined reference to otJoinerStart'` might suggest that new libraries generated by you don't have Joiner support. Make sure to do `make -f examples/Makefile-nrf52811 clean` before building with new switches.

oc1tane, 2021-03-06:
I was able to compile and run the application except for an error that causes a heap in thread_utils. I commented out the platform_init () function and the firmware began to weigh about 178 kb, which allows it to be flashed to nrf52811, but the problem is that the entire timer began to work involuntarily, and the joiner does not occur when the button is pressed. Apparently this function is very important how can it be repaired?

`/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: _build/nrf52811_xxaa/thread_utils.c.o: in function `platform_init':
/Downloads/nRF5_SDK_for_Thread_and_Zigbee_v4.1.0_32ce5f8/components/thread/utils/thread_utils.c:129: undefined reference to `otHeapSetCAllocFree'
collect2: error: ld returned 1 exit status`

LuDuda, 2021-03-09:
@oc1tane as I mentioned, please create a ticker on Devzone, which is prefered platform to handle issues with nRF5 SDK.

The lack of `otHeapSetCAllocFree` suggests that either:
- you are not on the OT commit from the release - c6a258e3
- you missed the `EXTERNAL_HEAP=1` switch while building the libraries, for example:
`make -f examples/Makefile-nrf52811 JOINER=1 COAP=1 DIAGNOSTIC=1 DNS_CLIENT=1 EXTERNAL_HEAP=1 MAC_FILTER=1 MTD_NETDIAG=1`