openthread/openthread #6224: Compilation issues inside radio_spinel_impl.hpp

doru91, 2021-03-02:
The compilation issues seems to be caused by commit id 33f15fac2faad5aa31920a24cbbc8b2fc49dbe08

Compilation issues:

../openthread/src/lib/spinel/radio_spinel_impl.hpp: In member function 'otError ot::Spinel::RadioSpinel<InterfaceType, ProcessContextType>::RestoreDatasetFromNcp()':
../openthread/src/lib/spinel/radio_spinel_impl.hpp:420:19: error: no matching function for call to 'ot::Instance::Get()'
  420 |     Instance::Get().template Get<SettingsDriver>().Init();
      |                   ^
In file included from ../openthread/src/lib/spinel/radio_spinel_impl.hpp:45,
                 from ../openthread/src/lib/spinel/radio_spinel.hpp:1015,
                 from ../src/imx_rt/rt1060/radio.cpp:39:
../openthread/src/core/common/instance.hpp:298:43: note: candidate: 'template<class Type> Type& ot::Instance::Get()'
  298 |     template <typename Type> inline Type &Get(void);
      |                                           ^~~
../openthread/src/core/common/instance.hpp:298:43: note:   template argument deduction/substitution failed:
In file included from ../openthread/src/lib/spinel/radio_spinel.hpp:1015,
                 from ../src/imx_rt/rt1060/radio.cpp:39:
../openthread/src/lib/spinel/radio_spinel_impl.hpp:420:19: note:   couldn't deduce template parameter 'Type'
  420 |     Instance::Get().template Get<SettingsDriver>().Init();
      |                   ^
../openthread/src/lib/spinel/radio_spinel_impl.hpp:420:48: error: expected primary-expression before '>' token
  420 |     Instance::Get().template Get<SettingsDriver>().Init();
      |                                                ^
../openthread/src/lib/spinel/radio_spinel_impl.hpp:420:50: error: expected primary-expression before ')' token
  420 |     Instance::Get().template Get<SettingsDriver>().Init();
      |                                                  ^
../openthread/src/lib/spinel/radio_spinel_impl.hpp:429:19: error: no matching function for call to 'ot::Instance::Get()'
  429 |     Instance::Get().template Get<SettingsDriver>().Deinit();
      |                   ^
In file included from ../openthread/src/lib/spinel/radio_spinel_impl.hpp:45,
                 from ../openthread/src/lib/spinel/radio_spinel.hpp:1015,
                 from ../src/imx_rt/rt1060/radio.cpp:39:
../openthread/src/core/common/instance.hpp:298:43: note: candidate: 'template<class Type> Type& ot::Instance::Get()'
  298 |     template <typename Type> inline Type &Get(void);
      |                                           ^~~
../openthread/src/core/common/instance.hpp:298:43: note:   template argument deduction/substitution failed:
In file included from ../openthread/src/lib/spinel/radio_spinel.hpp:1015,

jwhui, 2021-03-02:
@doru91 , thanks for raising this issue. Can you provide more details on your toolchain? This may be due to a toolchain issue not supporting this use of `template` key word.

As FYI - our continuous integration currently includes build checks against arm-gcc versions (4, 5, 6, 7, 9), gcc (5, 6, 7, 8, 9, 10), and clang (6, 7, 8, 9).

doru91, 2021-03-02:
doru@ubuntu:~/Desktop/private_repos/ot-nxp$ /usr/local/bin/arm-none-eabi-g++ -v
Using built-in specs.
COLLECT_GCC=/usr/local/bin/arm-none-eabi-g++
COLLECT_LTO_WRAPPER=/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: /mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/src/gcc/configure --target=arm-none-eabi --prefix=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native --libexecdir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/lib --infodir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/arm-none-eabi --build=x86_64-linux-gnu --host=x86_64-linux-gnu --with-gmp=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-native/host-libs/usr --with-mpfr=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-native/host-libs/usr --with-mpc=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-native/host-libs/usr --with-isl=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-native/host-libs/usr --with-libelf=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Arm Embedded Toolchain 9-2020-q2-update' --with-multilib-list=rmprofile,aprofile
Thread model: single
gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)

jwhui, 2021-03-02:
@doru91 , is it possible for you to provide the build command and any additional details so that we can reproduce?

The latest build checks passed using arm-gcc-9 - https://github.com/openthread/openthread/actions/runs/614628402. Are you aware of any differences that may trigger this build issue?

abtink, 2021-03-02:
@doru91 , one theory/guess is that somehow the `radio_spinel_impl.hpp` is being compiled in a build config without either `OPENTHRED_FTD` or `OPENTHREAD_MTD` being defined. The `Instance::Get<SettingsDriver>()` (which seems to be causing the compile error) is under the check `#if OPENTHREAD_MTD || OPENTHREAD_FTD`.

doru91, 2021-03-04:
This is the compilation command: 

2/8] Building CXX object src/imx_rt/rt1060/CMakeFiles/openthread-rt1060.dir/radio.cpp.obj
FAILED: src/imx_rt/rt1060/CMakeFiles/openthread-rt1060.dir/radio.cpp.obj 
/usr/local/bin/arm-none-eabi-g++  -DOPENTHREAD_CONFIG_ASSERT_ENABLE=1 -DOPENTHREAD_CONFIG_BACKBONE_ROUTER_DUA_NDPROXYING_ENABLE=0 -DOPENTHREAD_CONFIG_BACKBONE_ROUTER_MULTICAST_ROUTING_ENABLE=0 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=1 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS_MANAGEMENT=1 -DOPENTHREAD_CONFIG_FILE=\"openthread-core-rt1060-config.h\" -DOPENTHREAD_CONFIG_IP6_SLAAC_ENABLE=1 -DOPENTHREAD_CONFIG_NCP_UART_ENABLE=1 -DOPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_1 -DOPENTHREAD_CORE_CONFIG_PLATFORM_CHECK_FILE=\"openthread-core-rt1060-config-check.h\" -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"openthread-core-rt1060-config.h\" -DOPENTHREAD_SPINEL_CONFIG_OPENTHREAD_MESSAGE_ENABLE=0 -DOPENTHREAD_SPINEL_CONFIG_RCP_RESTORATION_MAX_COUNT=0 -DPACKAGE_NAME=\"OPENTHREAD\" -DPACKAGE_VERSION=\"0.01.00\" -I../openthread/examples/platforms -I../openthread/third_party/mbedtls/repo/include -I../openthread/include -I../openthread/src -I../openthread/src/core -Iopenthread/third_party/mbedtls -I../src/imx_rt/rt1060 -Iopenthread/etc/cmake -I../openthread/etc/cmake -mthumb -fdata-sections -ffunction-sections -mcpu=cortex-m0plus -mfloat-abi=soft -fno-exceptions -fno-rtti -Os -DNDEBUG   -Wno-unknown-pragmas -Wno-sign-compare -Wno-unused-function -Wno-unused-parameter -Wno-empty-body -std=gnu++11 -MD -MT src/imx_rt/rt1060/CMakeFiles/openthread-rt1060.dir/radio.cpp.obj -MF src/imx_rt/rt1060/CMakeFiles/openthread-rt1060.dir/radio.cpp.obj.d -o src/imx_rt/rt1060/CMakeFiles/openthread-rt1060.dir/radio.cpp.obj -c ../src/imx_rt/rt1060/radio.cpp

I tried to manually add -DOPENTHREAD_FTD=1 in the compilation command but I still get the error.

doru91, 2021-03-11:
How to reproduce the issue:

```
doru@ubuntu:~/Desktop/public_repos/$ git clone https://github.com/doru91/ot-nxp.git
doru@ubuntu:~/Desktop/public_repos/$ cd ot-nxp
doru@ubuntu:~/Desktop/public_repos/ot-nxp$ git submodule update --init
doru@ubuntu:~/Desktop/public_repos/ot-nxp$ export NXP_RT1060_SDK_ROOT=/home/doru/Desktop/private_repos/ot-nxp/third_party/rt1060_sdk/RT1060_SDK/
doru@ubuntu:~/Desktop/public_repos/ot-nxp$ touch /home/doru/Desktop/private_repos/ot-nxp/third_party/rt1060_sdk/RT1060_SDK/SW-Content-Register.txt
doru@ubuntu:~/Desktop/public_repos/ot-nxp$ touch /home/doru/Desktop/private_repos/ot-nxp/third_party/rt1060_sdk/RT1060_SDK/TODO.c
doru@ubuntu:~/Desktop/public_repos/ot-nxp$ ./script/build_rt1060
```

jwhui, 2021-03-11:
@doru91 , I get an error "Could not found a valid SDK package!" Is `doru91/ot-nxp/third_party` supposed to include some additional submodules? Or what are the steps necessary to obtain a valid SDK package?

doru91, 2021-03-12:
@jwhui, NXP_RT1060_SDK_ROOT is supposed to reference a valid SDK. This is still work in progress so for the moment we are simulating an SDK inside third_party/rt1060_sdk/RT1060_SDK/ by creating TODO.c file and SW-Content-Register.txt. I copy-pasted the above instructions in my Ubuntu 20 terminal and the compilation process starts:

doru@ubuntu:~/Desktop/public_repos/ot-nxp$ ./script/build_rt1060 
+ OT_CMAKE_NINJA_TARGET=
++ pwd
+ readonly OT_SRCDIR=/home/doru/Desktop/public_repos/ot-nxp
+ OT_SRCDIR=/home/doru/Desktop/public_repos/ot-nxp
+ OT_OPTIONS=("-DCMAKE_TOOLCHAIN_FILE=src/imx_rt/rt1060/arm-none-eabi.cmake" "-DCMAKE_BUILD_TYPE=MinSizeRel" "-DOT_PLATFORM=external" "-DOT_SLAAC=ON")
+ readonly OT_OPTIONS
+ main
+ '[' -z /home/doru/Desktop/private_repos/ot-nxp/third_party/rt1060_sdk/RT1060_SDK/ ']'
+ '[' -f /home/doru/Desktop/private_repos/ot-nxp/third_party/rt1060_sdk/RT1060_SDK//SW-Content-Register.txt ']'
+ SDK_RELEASE=ON
+ echo 'NXP_RT1060_SDK_ROOT set to ' /home/doru/Desktop/private_repos/ot-nxp/third_party/rt1060_sdk/RT1060_SDK/
NXP_RT1060_SDK_ROOT set to  /home/doru/Desktop/private_repos/ot-nxp/third_party/rt1060_sdk/RT1060_SDK/
+ options=("${OT_OPTIONS[@]}")
+ local options
+ options+=("$@")
+ options+=' -DSDK_RELEASE=ON'
+ build '-DCMAKE_TOOLCHAIN_FILE=src/imx_rt/rt1060/arm-none-eabi.cmake -DSDK_RELEASE=ON' -DCMAKE_BUILD_TYPE=MinSizeRel -DOT_PLATFORM=external -DOT_SLAAC=ON
+ local builddir=build
+ mkdir -p build
+ cd build
+ cmake -GNinja -DOT_APP_NCP=OFF -DOT_BUILD_RT=ON -DOT_BUILD_RT1060=ON -DOT_COMPILE_WARNING_AS_ERROR=ON -DCMAKE_TOOLCHAIN_FILE=src/imx_rt/rt1060/arm-none-eabi.cmake -DSDK_RELEASE=ON -DCMAKE_BUILD_TYPE=MinSizeRel -DOT_PLATFORM=external -DOT_SLAAC=ON /home/doru/Desktop/public_repos/ot-nxp
-- The C compiler identification is GNU 9.3.1
-- The CXX compiler identification is GNU 9.3.1
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- CMake build type: MinSizeRel
-- Version: 0.01.00
CMake Warning (dev) at openthread/third_party/mbedtls/repo/CMakeLists.txt:5 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python (found version "2.7.18") 
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/doru/Desktop/public_repos/ot-nxp/build
[... 82 lines cut ...]
      |                                           ^~~
../openthread/src/core/common/instance.hpp:298:43: note:   template argument deduction/substitution failed:
In file included from ../openthread/src/lib/hdlc/hdlc.hpp:42,
                 from ../openthread/src/lib/spinel/spinel_interface.hpp:38,
                 from ../openthread/src/lib/spinel/radio_spinel.hpp:41,
                 from ../src/imx_rt/rt1060/radio.cpp:39:
../openthread/src/lib/spinel/radio_spinel_impl.hpp:734:41: note:   couldn't deduce template parameter 'Type'
  734 |     SuccessOrExit(error = Instance::Get().template Get<SettingsDriver>().Set(
      |                                         ^
../openthread/src/core/common/code_utils.hpp:107:14: note: in definition of macro 'SuccessOrExit'
  107 |         if ((aStatus) != 0)    \
      |              ^~~~~~~
../openthread/src/lib/spinel/radio_spinel_impl.hpp:734:70: error: expected primary-expression before '>' token
  734 |     SuccessOrExit(error = Instance::Get().template Get<SettingsDriver>().Set(
      |                                                                      ^
../openthread/src/core/common/code_utils.hpp:107:14: note: in definition of macro 'SuccessOrExit'
  107 |         if ((aStatus) != 0)    \
      |              ^~~~~~~
../openthread/src/lib/spinel/radio_spinel_impl.hpp:734:72: error: expected primary-expression before ')' token
  734 |     SuccessOrExit(error = Instance::Get().template Get<SettingsDriver>().Set(
      |                                                                        ^
../openthread/src/core/common/code_utils.hpp:107:14: note: in definition of macro 'SuccessOrExit'
  107 |         if ((aStatus) != 0)    \
      |              ^~~~~~~
ninja: build stopped: subcommand failed.

abtink, 2021-03-12:
@doru91 , I think your last comment adds more confidence in the theory/guess from https://github.com/openthread/openthread/issues/6224#issuecomment-789116593, i.e. the `OPENTHREAD_FTD` or `OPENTHREAD_MTD` are not in the list of defines when building some file that is including `"radio_spinel.hpp"` (which itself includes `"radio_spinel_impl.hpp"`).

Here is the flow I see:
- This file that is being compiled: `src/imx_rt/rt1060/radio.cpp`
- Here is the compiler call to `arm-none-eabi-g++` (I wrapped the lines at `100` chars so we can see all lines more easily).
```
/usr/local/bin/arm-none-eabi-g++  -DOPENTHREAD_CONFIG_ASSERT_ENABLE=1
/-DOPENTHREAD_CONFIG_BACKBONE_ROUTER_DUA_NDPROXYING_ENABLE=0
/-DOPENTHREAD_CONFIG_BACKBONE_ROUTER_MULTICAST_ROUTING_ENABLE=0
/-DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=1
/-DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS_MANAGEMENT=1
/-DOPENTHREAD_CONFIG_FILE=\"openthread-core-rt1060-config.h\" -DOPENTHREAD_CONFIG_IP6_SLAAC_ENABLE=1
/-DOPENTHREAD_CONFIG_NCP_UART_ENABLE=1 -DOPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_1
/-DOPENTHREAD_CORE_CONFIG_PLATFORM_CHECK_FILE=\"openthread-core-rt1060-config-check.h\"
/-DOPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"openthread-core-rt1060-config.h\"
/-DOPENTHREAD_SPINEL_CONFIG_OPENTHREAD_MESSAGE_ENABLE=0
/-DOPENTHREAD_SPINEL_CONFIG_RCP_RESTORATION_MAX_COUNT=0 -DPACKAGE_NAME=\"OPENTHREAD\"
/-DPACKAGE_VERSION=\"0.01.00\" -I../openthread/examples/platforms
/-I../openthread/third_party/mbedtls/repo/include -I../openthread/include -I../openthread/src
/-I../openthread/src/core -Iopenthread/third_party/mbedtls -I../src/imx_rt/rt1060
/-Iopenthread/etc/cmake -I../openthread/etc/cmake -mthumb -fdata-sections -ffunction-sections
/-mcpu=cortex-m0plus -mfloat-abi=soft -fno-exceptions -fno-rtti -Os -DNDEBUG   -Wno-unknown-pragmas
/-Wno-sign-compare -Wno-unused-function -Wno-unused-parameter -Wno-empty-body -std=gnu++11 -MD -MT
/src/imx_rt/rt1060/CMakeFiles/openthread-rt1060.dir/radio.cpp.obj -MF
/src/imx_rt/rt1060/CMakeFiles/openthread-rt1060.dir/radio.cpp.obj.d -o
/src/imx_rt/rt1060/CMakeFiles/openthread-rt1060.dir/radio.cpp.obj -c ../src/imx_rt/rt1060/radio.cpp
```
- We don't seem to have `OPENTHREAD_MTD` or `OPENTHREAD_FTD` among the defines listed here.
- The specialization of template `Instance::Get<>` for `Instance::Get<SettingsDriver>()`) is available under `#if OPENTHREAD_MTD || OPENTHREAD_FTD` check.

abtink, 2021-03-12:
I think @jwhui  and I found the root-cause here. 😃 

The error is actually with the non-template `static Instance &Instance::Get(void)` method. This method is available under single OT instance config and the build here seems to be enabling `OPENTHREAD_CONFIG_MULTIPLE_INSTANCE_ENABLE` in its [config](https://github.com/doru91/ot-nxp/blob/dfcc142f59d9220d3cae11c61dfb3d319a84658a/src/imx_rt/rt1060/openthread-core-rt1060-config.h#L168).

In addition to this, I think we still need to have `OPENTHREAD_FTD` (or `OPENTHREAD_MTD`) defined in the config build as well (to ensure that we can get to `Get<SettingsDriver>()` specialization).

@doru91, for a quick fix, you can consider disabling `OPENTHREAD_CONFIG_MULTIPLE_INSTANCE_ENABLE` (do you need/use the multi instance behavior?)

We will also submit a change to make the implementation work under multi-instance config in addition single instance case.

doru91, 2021-03-15:
Thanks a lot for your help. Disabled OPENTHREAD_CONFIG_MULTIPLE_INSTANCE_ENABLE for the moment and the code compiles.

abtink, 2021-03-15:
@doru91 , can you please help provide more info on your use-case and if/why you may need the `MUTILPLE_INSTANCE` feature?

In general, would recommend not using `MULTIPLE_INSTANCE` feature:
- Enabling this, requires all OT core objects to track the `ot::Instance` they belong to.
- This adds a lot of memory and code overhead.

doru91, 2021-03-16:
Agree, we don't need the MULTIPLE_INSTANCE feature. So this issue can be closed from my side.

closed 2021-03-16 by jwhui

anhns26693, 2025-09-10:
> [@doru91](https://github.com/doru91) , can you please help provide more info on your use-case and if/why you may need the `MUTILPLE_INSTANCE` feature?
>
> In general, would recommend not using [...]

I have the case in my project :

The hardware is raspberry pi 4b modle
OS : Debian GNU/Linux 11 (bullseye)
g++ : g++ (Debian 10.2.1-6) 10.2.1 20210110

We build lighting-app from source [homeconnectedip](https://github.com/project-chip/connectedhomeip) which support Thread

We have two raspiberry pi :
- The first one is configed and formed as Thread Border Router, it has ONE nrf52840 usb dongle
- Version Openthred is 1.3.0.0

- The second one is used for running multiple lighting-app thread device, it has TWO nrf52840 usb dongle
- Each thread lighting will config as Thread End device and use a seperate RCP usb dongle

Should I use MULTIPLE_INSTANCE option in this case ?

Actually, I met the error when enable MULTIPLE_INSTANCE ?
raspberry pi 4b model, debian os 11 bulleyes
clone ot-br-posix: https://github.com/openthread/ot-br-posix
cd ot-br-posix
mkdir mybuild && cd mybuild
cmake .. -DOT_MULTIPLE_INSTANCE=ON

#error "OPENTHREAD_CONFIG_LOG_PREPEND_UPTIME is not supported under OPENTHREAD_CONFIG_MULTIPLE_INSTANCE_ENABLE"

Can you give me some advice for this issue ?
Thank you so much !!!