openthread/openthread #6225: ECDSA signature failure: MBEDTLS_ERR_ECP_BAD_INPUT_DATA

nealjack, 2021-03-02:
**Describe the bug**
Attempting an ECDSA signature using `otCryptoEcdsaSign` results in `OT_ERROR_FAILED` . Digging into this a little deeper, the call to `mbedtls_ecdsa_from_keypair` returns MBEDTLS_ERR_ECP_BAD_INPUT_DATA. [My code](https://github.com/lab11/nrf52x-base/blob/up_openthread/lib/google_cloud/gc_iot.c#L163) that uses this function call worked on commit c6a258e.

I have a feeling that recent changes with the heap or mbedtls configuration and api may not be reflected in my codebase. I tried reading through recent commits and I was unable to find anything obvious.

**To Reproduce** Information to reproduce the behavior, including:
On commit 397e94d, nRF52840, with libraries built with the following make command:
```
make -f examples/Makefile-nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COAP=1 COMMISSIONER=1 DISABLE_BUILTIN_MBEDTLS=1 DNS_CLIENT=1 DIAGNOSTIC=1 EXTERNAL_HEAP=1 JOINER=1 LINK_RAW=1 MAC_FILTER=1 MTD_NETDIAG=1 SERVICE=1 UDP_FORWARD=1 ECDSA=1 SNTP_CLIENT=1 DHCP6_SERVER=1 DHCP6_CLIENT=1
```

**Expected behavior**
I would expect that given valid parameters, the `otCryptoEcdsaSign` would function and return successfully.

**Console/log output**
Not sure how helpful this is, but I tried to dissect the ECDSA [sign process](https://github.com/openthread/openthread/blob/master/src/core/crypto/ecdsa.cpp#L197) and call the mbedtls functions directly and print out parameter pointers (to check for nullptr) and function results:
```
<info> app: PK parse result: 0
<info> app: PK type compare: 2, 2
<info> app: PK pointer: 0x2000A414
<info> app: PK ecdsa from keypair result: 0
<info> app: grp pointer: 0x2003E8A4, grp id: 3, N.p value: 0x0005DE58
<info> app: r pointer: 0x2003E82C
<info> app: s pointer: 0x2003E838
<info> app: d pointer: 0x2003E920
<info> app: input buf pointer, length: 0x2003E844, 32
<info> app: ecdsa sign result: -4F80
```

jwhui, 2021-03-04:
@LuDuda , can you assign this to someone that can take a look?

LuDuda, 2021-03-04:
Yes will do. Thanks Jonathan.

edmont, 2021-03-10:
Hi @nealjack. Did you enable deterministic ECDSA? Please take a look to this PR, see if it's somehow related: https://github.com/openthread/openthread/pull/6261

nealjack, 2021-03-11:
I did not enable deterministic ECDSA to my knowledge. Should I be?

I updated my openthread libraries to include that PR, however I am still having the same issue - the ECDSA signature is still failing.

edmont, 2021-03-12:
OpenThread added deterministic ECDSA signature recently (https://github.com/openthread/openthread/commit/c00cb8a4c4f6e2a619770ffbb4abfd0f36a7fdc2) but it seems this does not affect to your code which is still using `Ecdsa:Sign`, not `P256::KeyPair::Sign`.

Just to confirm, when you say _`mbedtls_ecdsa_from_keypair` returns MBEDTLS_ERR_ECP_BAD_INPUT_DATA_ you mean `mbedtls_ecdsa_sign`, right? According to your logging information:

`<info> app: PK ecdsa from keypair result: 0`

nealjack, 2021-03-12:
Yes, I meant `mbedtls_ecdsa_sign` returns MBEDTLS_ERR_ECP_BAD_INPUT_DATA. The call to `mbedtls_ecdsa_from_keypair` is successful.

edmont, 2021-03-15:
@nealjack could you check if you properly updated OpenThread and nrf_security libraries in your project? I hooked your code using `mbedtls_ecdsa_sign` within the `joiner stop` CLI command in this branch: https://github.com/edmont/openthread/commit/a338dc62518b92b8be97621520af2386f88c348a. It works well on a recent OT snapshot.

nealjack, 2021-03-17:
I double checked that I've copied in the most recent nrf_security libraries from `third_party/NordicSemiconductor/libraries/nrf_security`. 

Thanks for taking the time to hook in my code to an example. I'll try it out and report back.

What make flags are you using to build the cli application?

nealjack, 2021-03-17:
I cloned and built your CLI example and verified that the `joiner stop` command executes successfully. I used the following to build for an nRF52840 dongle:
```
make -f examples/Makefile-nrf52840 USB=1 BOOTLOADER=USB JOINER=1 ECDSA=1
```

I still can't get it to work with my own application however.

edmont, 2021-03-18:
Any luck with disabling the external heap? https://github.com/lab11/nrf52x-base/commit/7d7f7a5aa64ab9135024746947e514ce41365ccf I didn't try with external heap enabled myself.

BTW, nRF5 SDK project is now discontinued by Nordic in favor of [nRF Connect SDK](https://www.nordicsemi.com/Software-and-tools/Software/nRF-Connect-SDK). FYI, nrf_security libraries are now generated based on the latest.

nealjack, 2021-03-19:
I disabled the external heap and still had the same issue.

I can look into migrating my codebase to the new nRF SDK, but that will take some time and isn't my highest priority at the moment.

I link in the nrf_security included in openthread, not the one included in the old SDK. So it shouldn't be an issue.

edmont, 2021-03-24:
@nealjack could you give a try increasing the Memory Manager buffers? https://github.com/lab11/nrf52x-base/blob/master/boards/nrf52840dk_thread/app_config.h#L75-L85

nealjack, 2021-03-26:
Okay, after a lot of digging I believe that I've identified what causes this issue. I'm not sure why though.

I was able to build a working application that uses my code that performs an ECDSA sign. However, when I build that application to work with nordic's background dfu bootloader, I run into the ECDSA signature failure.

The difference between the two applications are primarily a different linker script to make room for the MBR, bootloader, and bootloader settings. I altered the linker scripts to be nearly identical besides start and stop addresses for ram/flash, with no success.

Has something changed with Openthread's flash configuration that could be conflicting with the bootloader or bootloader settings? Does it require more flash pages than previous?

edmont, 2021-04-06:
Hi, not sure how flash distribution could affect this issue, specially when the application is programmed after the bootloader. In any case, could you confirm that there are no overlapping flash regions for your layout?

By the way, did you give a try to the increased Memory Manager buffers?

nealjack, 2021-06-02:
I have also been trying to enable joiner functionality in some of my applications, and I am running into a similar issue where the joiner process fails with OT_ERROR_SECURITY (8) if the application is built for use with the background bootloader and flashed together. The application works as expected if built without the bootloader included.

I am not knowledgeable with the inner workings of the background bootloader crypto vs that used in openthread, but are they both utilizing the hardware accelerator? If so, could there be some initialization conflict between the two pieces of software?

nealjack, 2021-09-22:
@edmont Do you have any other ideas on what could be causing this issue? Does the background bootloader work for you with more recent openthread versions?

jwhui, 2022-08-23:
Closing stale issue.

closed 2022-08-23 by jwhui