openthread/openthread #6226: Support for Spinel Interface ID

mnp222, 2021-03-03:
Silabs would like to add support for using the Spinel Interface ID with the RCP.
The Spinel header contains a 2-bit IID field which allows the protocol to
support multiple networks over a single connection. Currently OpenThread sets
the Spinel IID value to 0 and it remains unused.

There are many possible use cases. The one we are initially targeting is
a multi-PAN RCP, in which multiple host processes are connected to a single
RCP via a multiplexer. Each process runs a network on a separate PAN, but on
a common channel. The Spinel messages are multiplexed using the Spinel IID to
distinguish between networks.

On the host side, the Spinel IID for each process must be obtained from the
application and stored in the RadioSpinel class. It is then used in the IID
field of all outgoing Spinel headers, and used to filter Spinel responses from
the RCP.

On the RCP side, the IID field of incoming Spinel messages must be cached.
The cached value is echoed in the header of Spinel responses, and used by
the radio platform code when setting parameters.

Two modifications are required to allow multiple independent processes to use
a single RCP.  First, in the case of an unexpected RCP reset, the RCP state must
be restored without an additional reset of the RCP, otherwise multiple processes
would alternate resetting the RCP.  Second, if the RCP is busy transmitting a
packet for one process when a second process requests a transmit, the second
process must wait and retry the transmission.

We've implemented a solution and can create a pull request. Let me know how
you'd like to proceed. Thanks.

romacdon, 2021-03-03:
@jwhui can you comment on the process here.  I asked @mnp222 to add this issue to get some feedback before proceeding with a PR.  Are there specific people we should wait on for approval?

jwhui, 2021-03-03:
> @jwhui can you comment on the process here. I asked @mnp222 to add this issue to get some feedback before proceeding with a PR. Are there specific people we should wait on for approval?

@romacdon , raising an issue is good for larger changes/enhancement to get some initial feedback on the approach. If you already have a PR ready, feel free to submit that as actual code can provide more visibility into the proposal.

jwhui, 2021-03-03:
@mnp222 , thanks for proposing the enhancement. This all seems reasonable to me.

@abtink , thoughts?

abtink, 2021-03-03:
The description of using Spinel IID sounds reasonable to me as well.

mnp222, 2021-03-03:
Thank you @jwhui and @abtink, I'll submit the PR.  Currently it is written with a configuration switch OPENTHREAD_MULTIPAN_RCP_ENABLE that turns the whole feature on and off.  That may not be really necessary in many places, so we can discuss.

jwhui, 2024-01-12:
Resolved by https://github.com/openthread/openthread/pull/9360

closed 2024-01-12 by jwhui