openthread/openthread #6250: DNS help

purvisb, 2021-03-09:
I'm burning FTD SOC firmware on an NRF52840-MDK. This node is joined to a partition with a verified working border router attached.

When I make firmware from OpenThread repository with parameters `USB=1 COMMISSIONER=1 DNS_CLIENT=1` and run the command `dns resolve ...` it returns:
```
> dns resolve ipv4.google.com fd11:22::383d:3ad7:2891:31cd
Error 35: InvalidCommand
```

When I burn FTD SOC firmware from the nrf SDK for Thread and Zigbee v4.1, the `dns resolve ...` command returns:
```
> dns resolve ipv4.google.com fd11:22::383d:3ad7:2891:31cd
> DNS response for ipv4.google.com - 64:ff9b:0:0:0:0:d83a:c0ce TTL: 59
Done

```

I presume the SDK is including some libraries. Am I missing some switches when I make my own firmware from the OpenThread repo? Or am I missing something else?

jwhui, 2021-03-09:
@purvisb , does the `dns` command appear in the list when you enter the `help` command? If not, it is likely because you initially did a build without the `DNS_CLIENT` feature enabled and did not reconfigure the build system.

One thing you can try is: `make -f examples/Makefile-nrf52840 clean` and rebuild.

Or you can blow away the existing build and output directories `rm -rf build output` and try building again.

purvisb, 2021-03-09:
That worked. Guess I wasn't being diligent with cleaning the output directory.

Thanks!

closed 2021-03-09 by purvisb