openthread/openthread #6260: OT-Daemon kills itself while handling RCP timeout at random time.

Hard1kPrajapat1, 2021-03-10:
Hello OT Team,

I am working on the custom product where Openthread would be used. I am facing the issue while commissioning the other device to join the network.

**The environment setup:**
- A custom product that has CC2538 connected on UART and RCP binary(without any change) is flashed on the CC2538.
- A nRF52840 DK is flashed with a simple client binary to act as an end device

**Problem:**
- I have started and run the ot-daemon on CC2538, created it as a Leader using ot-ctl binary.
- I need to connect the nRF52840 to the leader to act as an end device. So commissioned the nRF52840 to allow it to join the thread network.
- Sometimes, it allows to join the network, and sometimes, it shows the error while in the commissioning process.

**Error 1:** This comes while applying the commissioning commands to join the other devices to the network.
ot-daemon: /home/openthread/src/posix/../../src/posix/platform/udp.cpp:271: otError otPlatUdpBind(otUdpSocket*): Assertion `aUdpSocket->mHandle != nullptr' failed.
line 1: 1897 Aborted ./ot-daemon 'spinel+hdlc+uart:///dev/ttyS1?uart-baudrate=115200'
[1]+ Exit 134

**Error 2:** This comes once the device is joined to the network and at a random time it disconnects as following.
[1]+ Exit 6         ./ot-daemon 'spinel+hdlc+uart:///dev/ttyS1?uart-baudrate=115200'

I have debugged into the source, and it seems that it fails in handling the RCP timeout, so is there any reason why it does that also I am looking for the solution.
This error is totally random, any help would be highly appreciated.

Thanks in advance,
Hardik

jwhui, 2021-03-10:
@Hard1kPrajapat1 , thanks for raising this issue.

When you say "RCP binary(without any change)" do you mean:
1. A binary you built without any changes to the source? If so, what git commit are you using?
2. A binary that you downloaded?

> Error 1: This comes while applying the commissioning commands to join the other devices to the network.
> ot-daemon: /home/openthread/src/posix/../../src/posix/platform/udp.cpp:271: otError [...]

@bukepo , thoughts on this error?

> Error 2: This comes once the device is joined to the network and at a random time it disconnects as following.
> [1]+ Exit 6 ./ot-daemon 'spinel+hdlc+uart:///dev/ttyS1?uart-baudrate=115200'

This has been previously reported in #5662 and #6063. It seems that HW flow control can help. Given that you are also using nRF52840 DK, can you try using the nRF52840 DK as an RCP and see if the issue persists?

Hard1kPrajapat1, 2021-03-10:
@jwhui, thanks for the quick response.

The RCP binary without change means that I have cloned the openthread source from GitHub and just compiled it like that only. The last commit from my source is:

```
commit babb66d9c7cde0f3b8902a830fcb7de8294ccaac 
Author: Abtin Keshavarzian <abtink@google.com>
Date:   Tue Aug 25 07:59:21 2020 -0700
    [neighbor-table] add 'StateFilter' parameter in 'Find' methods (#5435)
```

I will check for the HW flow control. Also, If I flash the RCP binary on nRF52840 DK, it will need the host processor right? or it will work as an RCP directly just like an end device without the need for ot-daemon and ot-ctl? 

Thanks,
Hardik

bukepo, 2021-03-10:
What's the host os for ot-daemon? And what's the git commit of ot-daemon?

jwhui, 2021-03-10:
> The RCP binary without change means that I have cloned the openthread source from GitHub and just compiled it like that only. The last commit from my source is:
>
> ```
> commit [...]

@Hard1kPrajapat1 , you could try moving forward to a newer commit, given that we have increased buffer sizes to help reduce the likelihood of dropped RCP->host frames (see #6043).

> I will check for the HW flow control. Also, If I flash the RCP binary on nRF52840 DK, it will need the host processor right? or it will work as an RCP directly just like an end device without the need for ot-daemon and ot-ctl?

Can you use the same host and just replace CC2538 with nRF52840? The RCP always needs a host.

Hard1kPrajapat1, 2021-03-10:
> What's the host os for ot-daemon? And what's the git commit of ot-daemon?
The host OS is Linux, tested with kernel 2.6 as well as 4.9 and faced the same problem. The last git commit for ot-daemon is below:

```
commit ad91ac3b9c697357d1f89b806589e0533e2ed87a
Author: Yakun Xu <xyk@google.com>
Date:   Wed Feb 3 13:50:18 2021 +0800
    [posix] enable expect tests on macOS (#5384)
    This commit enables expect tests on macOS, also includes the following
    required changes:

    * Fixed bound to interface issue on macOS.
    * Added ability to configure scan period, because macOS environment on
      GitHub Actions is much slower.
```

I will test with new commits by cloning new openthread source and will try the RCP binary on nRF52840 as well.

Hard1kPrajapat1, 2021-03-11:
Just to inform you that I have tried the ot-daemon by enabling the hardware flow control as following, but still that issue persists. 
`'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200&uart-flow-control'`

It was running for long duration and then it failed with error "HandleRcpTimeout". May be the buffer size is the issue.

jwhui, 2021-03-11:
Thanks for the update.

> Just to inform you that I have tried the ot-daemon by enabling the hardware flow control as following, but still that issue persists.
> `'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200&uart-flow-control'`

Did you make any other changes? For example, did you try changing out the CC2538 with an nRF52840?

> It was running for long duration and then it failed with error "HandleRcpTimeout". May be the buffer size is the issue.

Let us know if you have any more findings.

Hard1kPrajapat1, 2021-03-11:
I have cloned the fresh Openthread source today and compiled the RCP binary as well as ot-daemon and tested the thread network. Still, it is failed with the same error "HandleRcpTimeout".

I will test the same RCP binary with nRF52840 and get back to you.

JFYR, below is the last commit ID of new openthread source:
```
commit 82cca6294cb271c2949e044cdeefe3bb4b6e3d6c
Author: Jonathan Hui <jonhui@google.com>
Date:   Wed Mar 10 11:45:45 2021 -0800
    [dataset-updater] set `mDataset` to `nullptr` in `Cancel()` (#6262)
```

jwhui, 2021-03-30:
@Hard1kPrajapat1 , any update on this issue?

Hard1kPrajapat1, 2021-03-31:
@jwhui, I was testing with the new cloned source, with the latest RCP and ot-daemon binary from the latest Openthread source, I am not getting that error anymore. I have tested for a week on multiple boards but have not seen after.

Could you please help me to understand what would be the difference between the older source and the newer Openthread source? What could resolve this? Transmit buffer size? where can I look into it?

Thanks,
Hardik

jwhui, 2021-04-02:
@Hard1kPrajapat1 , yes increasing the buffer sizes helps avoid dropped frames. When a dropped frame is a response that the host was expecting, the `ot-daemon` process will die.

For more background see the following comments:

1. https://github.com/openthread/openthread/issues/5678#issuecomment-714735882
2. https://github.com/openthread/ot-br-posix/issues/559#issuecomment-721982478

If you are no longer experiencing this issue, can we call this closed?

jwhui, 2021-04-10:
Closing stale issue.

closed 2021-04-10 by jwhui